zephyrproject-rtos/zephyr #86847: The request_ctx misses data in http server when resource has wildcard in its path

phb98, 2025-03-10:
**Describe the bug**

In the latest commit on main, when the http server resource has wildcard in its path(for ex: /foo/* ), when there is new http request, the request_ctx data will miss the path right at the wildcard.
For example: If the resource has path: /foo/*, and a client make a request to resource /foo/bar , in callback the `request_ctx->data` will not include "bar"

**To Reproduce**

Add a http resource with wildcard in the end.
Make http request to that resource.
Observe the `request_ctx->data` in http server callback
**Expected behavior**

Should include the data at wildcard.
**Impact**

In my project i have an OAuth2 module, since there will be multiple service using OAuth2, the redirection link will be something like "zephyr.local/OAuth2/svc-name", so i need data at wildcard to extract service name.
**Logs and console output**

**Environment (please complete the following information):**
 Latest commit on main (df5318479b9)
 EPS32 platform
**Additional context**

I believe the commit 33cf7dc78a9 is root cause. It will skip the request url until it hit the '?', when in fact it should stop at the start of resource path string

rlubos, 2025-03-10:
Hi @phb98, thanks for the report.

I'm not sure the current behavior should be considered as a bug, the `data` pointer in the request context was not intended to include the URL path, only the parameters, the commit you've mentioned fixed that.

It seems that for your use case, you could simply access the URL directly from the client struct (`client->url_buffer`), it contains the full  URL for the current request.

phb98, 2025-03-11:
Hi @rlubos . The behavior before the update include the `bar` in resource path, just off-by-1. So I supposed this new commit will fix the off-by-1 instead removing it completely

I could use the full URL, but it makes me wonder, does that mean whenever a resource contain wildcard in its path, we always have to parse the full URL by ourselves ?

rlubos, 2025-03-11:
> I could use the full URL, but it makes me wonder, does that mean whenever a resource contain wildcard in its path, we always have to parse the full URL by ourselves ?

That would be the case yes, the data pointer was only intended to contain parameters, not path.

fabiobaltieri, 2025-03-11:
Tentatively closing, sounds like this is working as intended.

closed 2025-03-11 by fabiobaltieri