zephyrproject-rtos/zephyr #86848: tests: tests/drivers/flash/common: frdm_mcxw71/mcxw716c: failed: Failed harness:'Testsuite failed'

hakehuang, 2025-03-10:
**Describe the bug**
below case are failed
drivers.flash.common.default.flash_driver.read_unaligned_address
drivers.flash.common.default.flash_driver.flash_erase
drivers.flash.common.default.flash_driver.flash_copy
drivers.flash.common.default.flash_driver.flash_fill
drivers.flash.common.default.flash_driver.flash_flatten
drivers.flash.common.default.flash_driver.get_size
drivers.flash.common.default.flash_driver.flash_page_layout
drivers.flash.common.default.arm_rfft_f32_real_noisy_32

test is failed on v4.1.0-269-gdf5318479b91 on frdm_mcxw71/mcxw716c
testcase path is tests/drivers/flash/common

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p frdm_mcxw71/mcxw716c  -T tests/drivers/flash/common -s drivers.flash.common.default
```
or
```
# west twister -p frdm_mcxw71/mcxw716c -T tests/drivers/flash/common -s drivers.flash.common.default
```
2. See error

```
Failed harness:'Testsuite failed'
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_read_unaligned_address
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_read_unaligned_address in 0.016 seconds

START - test_flash_erase
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_flash_erase in 0.016 seconds

START - test_flash_copy
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_flash_copy in 0.016 seconds

START - test_flash_fill
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_flash_fill in 0.016 seconds

START - test_flash_flatten
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_flash_flatten in 0.016 seconds

START - test_get_size
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_get_size in 0.016 seconds

START - test_flash_page_layout
Test will run on device flash@0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false
FAIL - test_flash_page_layout in 0.016 seconds

    
```

```
phyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_flatten in 0.016 seconds
===================================================================
START - test_flash_page_layout
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_page_layout in 0.016 seconds
===================================================================
START - test_get_size
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_get_size in 0.016 seconds
===================================================================
START - test_read_unaligned_address
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_read_unaligned_address in 0.016 seconds
===================================================================
TESTSUITE flash_driver failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 7, skip = 0, total = 7 duration = 0.112 seconds
 - FAIL - [flash_driver.test_flash_copy] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_erase] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_fill] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_flatten] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_page_layout] duration = 0.016 seconds
 - FAIL - [flash_driver.test_get_size] duration = 0.016 seconds
 - FAIL - [flash_driver.test_read_unaligned_address] duration = 0.016 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 4f28068ed63d501e7be7104ef68d015e
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.1.0-269-gdf5318479b91

hakehuang, 2025-03-11:
@decsny strange, I still get this
```
E: Device id 00 00 00 does not match config c2 28 17
*** Booting Zephyr OS build v4.1.0-363-g895d91169e8b ***
Running TESTSUITE flash_driver
===================================================================
START - test_flash_copy
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_copy in 0.016 seconds
===================================================================
START - test_flash_erase
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_erase in 0.016 seconds
===================================================================
START - test_flash_fill
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_fill in 0.016 seconds
===================================================================
START - test_flash_flatten
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_flatten in 0.016 seconds
===================================================================
START - test_flash_page_layout
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_flash_page_layout in 0.016 seconds
===================================================================
START - test_get_size
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_get_size in 0.016 seconds
===================================================================
START - test_read_unaligned_address
Test will run on device flash@0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:67: flash_driver_before: device_is_ready(flash_dev) is false

 FAIL - test_read_unaligned_address in 0.016 seconds
===================================================================
TESTSUITE flash_driver failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 7, skip = 0, total = 7 duration = 0.112 seconds
 - FAIL - [flash_driver.test_flash_copy] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_erase] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_fill] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_flatten] duration = 0.016 seconds
 - FAIL - [flash_driver.test_flash_page_layout] duration = 0.016 seconds
 - FAIL - [flash_driver.test_get_size] duration = 0.016 seconds
 - FAIL - [flash_driver.test_read_unaligned_address] duration = 0.016 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: f1ea0dbfbd6d9f2f6a0775a6db76fee0
PROJECT EXECUTION FAILED

```

hakehuang, 2025-03-11:
> is it possibly something on your setup ?

Thanks a lot for checking this. @decsny  I find this is because the flash is conflicted with spi_loopback

closed 2025-03-11 by hakehuang

decsny, 2025-03-11:
> > is it possibly something on your setup ?
> 
> Thanks a lot for checking this. [@decsny](https://github.com/decsny) I find this is because the flash is conflicted with spi_loopback

that's still interesting because this test should not be using the spi? I added the spi nor flash to the DTS but still common flash test default should test the on chip flash... ?