zephyrproject-rtos/zephyr #86853: buildsystem: Kconfig: Add a way to configure warnings as errors

clamattia, 2025-03-10:
**Is your enhancement proposal related to a problem? Please describe.**
Sometimes `Kconfig` warnings can be pretty critical. For example if a security relevant feature is selected but gets clobbered by missing dependencies. To avoid this for CI-builds of applications. As an application developer I would like to configure the option to treat Kconfig warnings as errors and abort build with them.

**Describe the solution you'd like**
I would like to have an option to treat Kconfig-warnings as errors.

**Describe alternatives you've considered**
Using `grep` to search the build output with a regex. This seems unstable and not ideal.

**Additional context**
TBD

tpambor, 2025-05-20:
For reference, see https://github.com/zephyrproject-rtos/zephyr/pull/77529

clamattia, 2025-05-21:
Similar issue: https://github.com/zephyrproject-rtos/zephyr/issues/77530

@tejlmand 
Proposal to enpower CI-builds without Kconfig-warnings :
- Add an flag to the python script to select, that all warnings should be treated as errors (even those which are by default treated as warning)
- Introduce a way to select this flag from cmake. I do not yet know how? Any idea/preference @tejlmand ?
- 
Or:
- Change behavior and treat all Kconfig warnings as errors. Personally I do not see the benefit of allowing builds with Kconfig "warnings".

clamattia, 2025-05-30:
> [@clamattia](https://github.com/clamattia) I have a comment on this in [#77530](https://github.com/zephyrproject-rtos/zephyr/issues/77530)

I support your suggestion. As per above description, I am mainly worried about `assigned the value x but got y` "warnings" which we both agree (I think) should be "errors" instead.

closed 2026-09-30 by nordicjm