zephyrproject-rtos/zephyr #86908: Extra carrige return when using printf and newlib

runsiv, 2025-03-11:
**Describe the bug**

When using printf together with newlib.c the z_impl_zephyr_write_stdout from lib-hooks.c add an extra carriage return from https://github.com/zephyrproject-rtos/zephyr/blob/f9601948cafa238a7c369c77d91b259ec6f80961/lib/libc/newlib/libc-hooks.c#L202 in addition to the uart_console.c which  prepend every line feed (\n)

**To Reproduce**

1. Open a hello world sample
2. Add CONFIG_NEWLIB_LIBC=y in prj.conf
3. add printk("printk 0d0a:\n);
4. add printf("0d0d0a:\n);
5. compile
6. observe print in hex format

**Expected behavior**
Both print should have the same number of carriage return
**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (Ubuntu 24.04 and windows 11)
 - Toolchain (NCS 2.9, but the functions are the same in zephyr main)
 - Commit SHA or Version used

[image]

**Additional context**

fabiobaltieri, 2025-03-11:
Yeah happens with the blink sample as well, with picolibc

```
00000410  61 74 65 3a 20 4f 46 46  0d 0a 4c 45 44 20 73 74  |ate: OFF..LED st|
00000420  61 74 65 3a 20 4f 4e 0d  0a 4c 45 44 20 73 74 61  |ate: ON..LED sta|
00000430  74 65 3a 20 4f 46 46 0d  0a 4c 45 44 20 73 74 61  |te: OFF..LED sta|
```

with newlib

```
000005a0  74 65 3a 20 4f 46 46 0d  0d 0a 4c 45 44 20 73 74  |te: OFF...LED st|
000005b0  61 74 65 3a 20 4f 4e 0d  0d 0a 4c 45 44 20 73 74  |ate: ON...LED st|
000005c0  61 74 65 3a 20 4f 46 46  0d 0d 0a 4c 45 44 20 73  |ate: OFF...LED s|
```

fabiobaltieri, 2025-03-11:
@runsiv any chance you know if this is a regression or if it's always been the case?

runsiv, 2025-03-11:
I can't see the same behavior in your picolib  output. The libc-hooks.c in the picolibc do not have the https://github.com/zephyrproject-rtos/zephyr/blob/f9601948cafa238a7c369c77d91b259ec6f80961/lib/libc/newlib/libc-hooks.c#L202 which adds the extra \r 
I can see the function has been there at least since Zephyr v2.0 however i'm not familiar enough the the lib to know what will break if the appending \r is added.

closed 2025-05-28 by github-actions[bot]