zephyrproject-rtos/zephyr #86924: ESP32: DMA is enabled only when using UART0 and no reset of the rx fifo when the modem goes to suspend and resume power mode

dmkuzmin, 2025-03-11:
**Describe the bug**

Some of the information is described here #82370

zephyr version: Zephyr 3.7.1
modem: Quectel BG95
platform: ESP32-C3 and ESP32-S3
config options: for ISR CONFIG_UART_INTERRUPT_DRIVEN=y or for async CONFIG_UART_INTERRUPT_DRIVEN=n,
CONFIG_UART_ASYNC_API=y, CONFIG_MODEM_BACKEND_UART_ASYNC=y

Unfortunately, the problem with entering and exiting modem sleep mode has not been solved, where the controller takes garbage upon waking up and needs to reset the RX FIFO, as it seems to me, but I cannot guess where to insert it.

Secondly, there is a problem with asynchronous mode operation. DMA is enabled only when using UART0

fabiobaltieri, 2025-03-11:
Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

bjarki-andreasen, 2025-03-11:
This is not a modem subsystem issue, should be assigned to someone from ESP given it seems to be a UART driver bug

fabiobaltieri, 2025-03-11:
Thanks @bjarki-andreasen.

rftafas, 2025-03-11:
This was a known issue, solved sometime ago. Please, check if it is still there on v4.1

wmrsouza, 2025-03-11:
@dmkuzmin I could not find any evidence DMA is enabled only when using UART0. Going through uart_esp32.c, UART0 and UART1 are both treated equally with regards to DMA on esp32s3 and esp32c3 when `CONFIG_UART_ASYNC_API=y`, `CONFIG_DMA=y`, and dma status is `okay` in .overlay file. Would you mind telling us what led you to conclude that DMA is only enabled when using UART0?

dmkuzmin, 2025-03-11:
@wmrsouza , https://github.com/zephyrproject-rtos/zephyr/issues/82370#issuecomment-2517267144

That's my conclusion. I was convinced of this when I changed the number of the UART controller for the modem or for modbus devices in the dts file.

Examples of unsuccessful communication with devices are also given in #82370. There is a waiting for a response from the modem after issuing the AT command, timeouts in the drivers do not work. The exception is only for UART0

rftafas, 2025-03-12:
> Examples of unsuccessful communication with devices are also given in [#82370](https://github.com/zephyrproject-rtos/zephyr/issues/82370). There is a waiting for a response from the modem after issuing the AT command, timeouts in the drivers do not work. The exception is only for UART0

Ok, #82370 is also based on 3.7.1. Does that happens on 4.1? Do you mind testing it on most recent release? Otherwise, we have no plans to go back to 3.7 and fix things.