zephyrproject-rtos/zephyr #86954: qemu_x86_64/atom:sample.smf.hsm_psicc2 is flaky

fabiobaltieri, 2025-03-11:
**Describe the bug**
Hi, I noticed few `qemu_x86_64/atom` tests being a bit flaky recently, we see them failing on the main CI and then after a retry they pass, this is a sample run:

bad: https://github.com/zephyrproject-rtos/zephyr/actions/runs/13785932905/attempts/1
retry: https://github.com/zephyrproject-rtos/zephyr/actions/runs/13785932905/attempts/2

See:

```
INFO    - 1181/1181 qemu_x86_64/atom          sample.smf.hsm_psicc2                              FAILED Timeout (qemu 119.885s <zephyr>)
```

Is this something you Intel folks could look into?

**To Reproduce**
No idea, `west twister -p qemu_x86_64/atom -s sample.smf.hsm_psicc2` seems to work locally.

**Expected behavior**
Pass

**Impact**
Annoyance in CI, if it's failing for a good reason maybe we can exclude the platform.

aescolar, 2025-03-12:
The same(?) issue on 
twister -p qemu_x86_64/atom -T tests/lib/c_lib/thrd -s libraries.libc.c11_threads.picolibc.notls
(The test tends to hung and timeout quite often both in CI and locally)
https://github.com/zephyrproject-rtos/zephyr/actions/runs/13807339089/job/38620827891#step:11:2121
I'd say this is worse than a low. We cannot have a platform tests failing this often. It may be an indication of something being rotten.

kwd-doodling, 2025-03-12:
It has a high reproduce rate on my WSL environment and I debugged a while today.
It seems that the case runs well but somehow its logs cannot be printed out.
With CONFIG_DEBUG=y and CONFIG_NO_OPTIMIZATIONS=y set, the FW sticks during early booting.
On my ISH Simics environment, it shows that Shell log backend causes assertion inside k_spin_lock().

[image]

kwd-doodling, 2025-03-14:
Ignore the assertion, it's caused by stack overflow after set CONFIG_NO_OPTIMIZATION=y.
latest finding is that the logs are somehow being filtered away by logging API LOG_INF(). Need to more time to track the logging filter settings.

nashif, 2025-03-14:
when the sample fails, which is easy to reproduce under load, I get:

```
ESCcSeaBIOS (version rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org)
Booting from ROM...

uart:~$ *** Booting Zephyr OS build v4.1.0-544-g4bb5ffd7867d ***
uart:~$ State Machine Framework Demo
uart:~$ See PSiCC2 Fig 2.11 for the statechart
uart:~$ https://www.state-machine.com/psicc2
```

and not the expected logging:

```
    regex:
      - ".*<inf> hsm_psicc2_thread: initial_entry.*"
      - ".*<inf> hsm_psicc2_thread: s_entry.*"
      - ".*<inf> hsm_psicc2_thread: s2_entry.*"
      - ".*<inf> hsm_psicc2_thread: s21_entry.*"
      - ".*<inf> hsm_psicc2_thread: s211_entry.*"
      - 
      - 
   ```

nashif, 2025-03-14:
#87099 does not fix the issue for me. still getting the same output above and no logging output.

nashif, 2025-03-14:
btw, to reproduce easy, I use `stress --cpu 22` and launch twister on this sample in another terminal.

nashif, 2025-03-14:
@peter-mitsis  one of thos flaky CI SMP issues

nashif, 2025-03-14:
also fails on `qemu_cortex_a53/qemu_cortex_a53/smp samples/subsys/smf/hsm_psicc2/sample.smf.hsm_psicc2 FAILED Timeout (qemu 59.176s <zephyr>)`

nashif, 2025-03-14:
my suggestion is not to run this on SMP qemus systems in CI

kwd-doodling, 2025-03-14:
@nashif two patches needed. https://github.com/zephyrproject-rtos/zephyr/pull/87097 makes the reproduce rate lower to 1/20 on my WSL, and with https://github.com/zephyrproject-rtos/zephyr/pull/87099 I run the case over 100 times.
I tried your stress and qemu_cortex_a53/qemu_cortex_a53/smp and yes there's still chance to fail. But this seems to be another failure case I've never seen in my previous debugging days. This time even printk cannot print out.
I think the two things I captured are really problem and suggest to go on merging. I will take a look on the new issue next week.

kwd-doodling, 2025-03-14:
> [@nashif](https://github.com/nashif) two patches needed. [#87097](https://github.com/zephyrproject-rtos/zephyr/pull/87097) makes the reproduce rate lower to 1/20 on my WSL, and with [#87099](https://github.com/zephyrproject-rtos/zephyr/pull/87099) I run the case over 100 times. I tried your stress and qemu_cortex_a53/qemu_cortex_a53/smp and yes there's still chance to fail. But this seems to be another failure case I've never seen in my previous debugging days. This time even printk cannot print out. I think the two things I captured are real problems and suggest to go on merging. I will take a look on the new issue next week.

and more, from memory dump, the case has finished running, it seems to be still a logging lost issue.

nashif, 2025-03-15:
@kwd-doodling  I tried your patches, they definietly improve things and I do not see failures anymore, need to stress this even more I think and identify other spots. Thanks for the debugging and fixes

@nordic-krch  please take a look at the proposed fixes for logging.

nashif, 2025-03-15:
btw, looking at kibana, this shows that it has been failing like this on multiple platforms for a while:

[image]

[Kibana link](https://kibana.zephyrproject.io/s/public/app/dashboards#/view/f2ad2dda-2d26-4e73-854c-bc0b97edf1f1?_g=(filters:!())&_a=(controlGroupInput:(chainingSystem:HIERARCHICAL,controlStyle:twoLine,defaultControlGrow:!t,defaultControlWidth:medium,ignoreParentSettings:(ignoreFilters:!f,ignoreQuery:!f,ignoreTimerange:!f,ignoreValidations:!f),panels:('049cfa5b-44cf-4360-a72f-eeb283d23c8b':(explicitInput:(dataViewId:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',enhancements:(),fieldName:component.keyword,id:'049cfa5b-44cf-4360-a72f-eeb283d23c8b',parentFieldName:component,selectedOptions:!(),title:Area),grow:!t,order:3,type:optionsListControl,width:medium),'61ebc7fe-1175-4848-a8e3-9c4e9584b8e6':(explicitInput:(dataViewId:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',enhancements:(),fieldName:environment.zephyr_version.keyword,id:'61ebc7fe-1175-4848-a8e3-9c4e9584b8e6',parentFieldName:environment.zephyr_version,selectedOptions:!(),title:Commit),grow:!t,order:0,type:optionsListControl,width:medium),aad2dd65-2441-46a6-8e39-291b676beb73:(explicitInput:(dataViewId:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',enhancements:(),fieldName:status.keyword,id:aad2dd65-2441-46a6-8e39-291b676beb73,parentFieldName:status,selectedOptions:!(),title:Status),grow:!t,order:2,type:optionsListControl,width:medium),f41746b0-e865-4d59-b3fe-c42c4e9adc9a:(explicitInput:(dataViewId:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',enhancements:(),fieldName:platform.keyword,id:f41746b0-e865-4d59-b3fe-c42c4e9adc9a,parentFieldName:platform,selectedOptions:!(),title:Platform),grow:!t,order:1,type:optionsListControl,width:medium))),description:'Results%20from%20main%20branch.%20Data%20uploaded%20on%20every%20push%20and%20completed%20test%20run%20of%20main%20branch.',filters:!(('$state':(store:appState),meta:(alias:!n,disabled:!f,index:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',key:environment.run_attempt,negate:!f,params:(query:'1'),type:phrase),query:(match_phrase:(environment.run_attempt:'1'))),('$state':(store:appState),meta:(alias:!n,disabled:!f,index:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',key:name.keyword,negate:!f,params:(query:sample.smf.hsm_psicc2),type:phrase),query:(match_phrase:(name.keyword:sample.smf.hsm_psicc2)))),fullScreenMode:!f,options:(hidePanelTitles:!f,syncColors:!f,syncTooltips:!f,useMargins:!t),panels:!((embeddableConfig:(attributes:(references:!((id:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',name:indexpattern-datasource-layer-869b8e49-7c8b-4bed-a59b-1f74b7b62778,type:index-pattern)),state:(adHocDataViews:(),datasourceStates:(indexpattern:(layers:('869b8e49-7c8b-4bed-a59b-1f74b7b62778':(columnOrder:!('45cf1020-eb80-47f5-9de2-22684e086cdf'),columns:('45cf1020-eb80-47f5-9de2-22684e086cdf':(customLabel:!t,dataType:number,isBucketed:!f,label:'Test%20Runs',operationType:unique_count,params:(emptyAsNull:!t),scale:ratio,sourceField:environment.zephyr_version.keyword)),incompleteColumns:())))),filters:!(),internalReferences:!(),query:(language:kuery,query:''),visualization:(layerId:'869b8e49-7c8b-4bed-a59b-1f74b7b62778',layerType:data,metricAccessor:'45cf1020-eb80-47f5-9de2-22684e086cdf')),title:'',type:lens,visualizationType:lnsMetric),enhancements:()),gridData:(h:4,i:efa2a889-0de4-4f0d-9e7d-17cdaf7af84f,w:5,x:7,y:0),panelIndex:efa2a889-0de4-4f0d-9e7d-17cdaf7af84f,title:'',type:lens,version:'8.5.3'),(embeddableConfig:(attributes:(references:!((id:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',name:indexpattern-datasource-layer-869b8e49-7c8b-4bed-a59b-1f74b7b62778,type:index-pattern)),state:(adHocDataViews:(),datasourceStates:(indexpattern:(layers:('869b8e49-7c8b-4bed-a59b-1f74b7b62778':(columnOrder:!('45cf1020-eb80-47f5-9de2-22684e086cdf'),columns:('45cf1020-eb80-47f5-9de2-22684e086cdf':(customLabel:!t,dataType:number,isBucketed:!f,label:Architetures,operationType:unique_count,params:(emptyAsNull:!t),scale:ratio,sourceField:arch.keyword)),incompleteColumns:())))),filters:!(),internalReferences:!(),query:(language:kuery,query:''),visualization:(layerId:'869b8e49-7c8b-4bed-a59b-1f74b7b62778',layerType:data,metricAccessor:'45cf1020-eb80-47f5-9de2-22684e086cdf')),title:'',type:lens,visualizationType:lnsMetric),enhancements:()),gridData:(h:4,i:b348dd2e-757b-418a-80dd-17588c4cefc1,w:4,x:12,y:0),panelIndex:b348dd2e-757b-418a-80dd-17588c4cefc1,type:lens,version:'8.5.3'),(embeddableConfig:(attributes:(references:!((id:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',name:indexpattern-datasource-layer-869b8e49-7c8b-4bed-a59b-1f74b7b62778,type:index-pattern)),state:(adHocDataViews:(),datasourceStates:(indexpattern:(layers:('869b8e49-7c8b-4bed-a59b-1f74b7b62778':(columnOrder:!('45cf1020-eb80-47f5-9de2-22684e086cdf'),columns:('45cf1020-eb80-47f5-9de2-22684e086cdf':(customLabel:!t,dataType:number,isBucketed:!f,label:Platforms,operationType:unique_count,params:(emptyAsNull:!t),scale:ratio,sourceField:platform.keyword)),incompleteColumns:())))),filters:!(),internalReferences:!(),query:(language:kuery,query:''),visualization:(layerId:'869b8e49-7c8b-4bed-a59b-1f74b7b62778',layerType:data,metricAccessor:'45cf1020-eb80-47f5-9de2-22684e086cdf')),title:'',type:lens,visualizationType:lnsMetric),enhancements:()),gridData:(h:4,i:'8ac08070-9d7d-4604-a76f-2a6b9731434b',w:4,x:16,y:0),panelIndex:'8ac08070-9d7d-4604-a76f-2a6b9731434b',title:'',type:lens,version:'8.5.3'),(embeddableConfig:(attributes:(references:!((id:'0742d09e-4209-4cad-bba7-c4aa4b0f1b9e',name:indexpattern-datasource-laye [... cut ...] 72a1-4ef5-8172-589448cf76d8',isTransposed:!f)),layerId:a24e6b5a-3ea8-492e-8c53-289e0c31a08b,layerType:data)),title:'',type:lens,visualizationType:lnsDatatable),enhancements:()),gridData:(h:30,i:'969c08b4-623a-47d9-8ee4-0ac48aabc9e4',w:48,x:0,y:136),panelIndex:'969c08b4-623a-47d9-8ee4-0ac48aabc9e4',type:lens,version:'8.5.3')),query:(language:kuery,query:''),tags:!('465c2e18-ef30-4e7e-981d-2f8bcd899a05'),timeRange:(from:now-7d%2Fd,to:now),timeRestore:!t,title:'Zephyr%20Main%20(CI)',viewMode:view))