zephyrproject-rtos/zephyr #86975: X-NUCLEO-WB05KN1 shield not seems to work on different nucleo boards

cgbdevelco, 2025-03-12:
**Describe the bug**
A nucleo wb05kn1 shield is placed in Arduino header from Nucleo H743ZI2 and the peripheral_hr sample is build with the shield argument "x_nucleo_wb05kn1_spi". The sample does not work, it is stuck in bt_spi_open when trying to receive header. 

I have tried to add an overlay to the STM32H7 like the ones in the shield folder without any luck. 

I also tried to place shield on Nucleo F401RE with same result.
I also tried on nRF52840 DK and here it work as expected.

**To Reproduce**

1. Place shield on one of the following Nucleo boards
2. Build Bluetooth sample such as peripheral_hr with the shield argument
3. See error

**Expected behavior**
I would expect that it works just like on the nRF52840 DK

**Impact**
Showstopper

**Logs and console output**

**Environment (please complete the following information):**
OS: Linux
Toolchain: Zephyr SDK
Zephyr version: Tried both latest and Zephyr 3.7

**Additional context**
The overlay file I tried to add the STM32H7
```
 &spi1_sck_pa5 {
	/delete-property/ bias-pull-down;
	bias-pull-up;
};

&spi1_miso_pa6 {
	slew-rate = "high-speed";
};

&spi1_mosi_pb5 {
	slew-rate = "high-speed";
};

/ {
	chosen {
		zephyr,bt-c2h-uart = &usart3;
	};
};
```

HoZHel, 2025-04-09:
I used the latest main branch of Zephyr on `nucleo_f401re` and the `x_nucleo_wb05kn1_spi` shield with `samples/bluetooth/beacon`, and it worked fine after increasing the main stack size (e.g. `CONFIG_MAIN_STACK_SIZE=1600`). Since `nucleo_f401re` doesn't have TRNG peripheral, fake entropy source is used by Zephyr and it requires more stack size to run BLE samples on this board.

Unfortunately, you can't use `samples/bluetooth/peripheral_hr` by default on this board as the `PA5` is common between `arduino_spi_sck` and `led0 alias` (i.e. LD2) according to the [schematic](https://www.st.com/resource/en/schematic_pack/mb1136-default-c04_schematic.pdf). Hence, there will be a glitch on the SPI clock line.
An easy solution could be changing the pin for LED on the sample, or not using `led0 alias`.

I used the following overlay file for `nucleo_f401re`:
```
&spi1_sck_pa5 {
	/delete-property/ bias-pull-down;
	bias-pull-up;
};

&spi1_miso_pa6 {
	slew-rate = "high-speed";
};

&spi1_mosi_pa7 {
	slew-rate = "high-speed";
};

/ {
	chosen {
		zephyr,bt-c2h-uart = &usart2;
	};
};
```

Below is the screenshot of `samples/bluetooth/beacon`:

[image]

cgbdevelco, 2025-04-09:
Hi @HoZHel thanks for your reply. I have tried both beacon and peripheral_hr on `nucleo_f401re` and can confirm that it works when using the overlay file and disable the use of led on the peripheral_hr example. 

Sadly the `nucleo_f401re` is not the target MCU but just a devkit that I tried with the shield since the Nucleo H743ZI2 didn't work. 

I have tried Nucleo H743ZI and Nucleo H753ZI and latest Nucleo H755ZI (target board).

For the Nucleo H755ZI i'm using the following overlay file but it still fails:

```
&spi1_sck_pa5 {
	/delete-property/ bias-pull-down;
	bias-pull-up;
};

&spi1_miso_pa6 {
	slew-rate = "high-speed";
};

&spi1_mosi_pb5 {
	slew-rate = "high-speed";
};

/ {
	chosen {
		zephyr,bt-c2h-uart = &usart3;
	};
};

&spi1 {
	status = "okay";
	pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pb5>;
	pinctrl-names = "default";
	cs-gpios = <&gpiod 14 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
};

arduino_spi: &spi1 {};
```
I'm stuck at spi_context_wait_for_completion when waiting for k_sem_take(&ctx->sync, timeout)

Strange thing is that spi_context_complete is called all the time giving k_sem_give(&ctx->sync);

Maybe it is blocking somehow?

I'm not sure if it is related somehow to this? [https://github.com/zephyrproject-rtos/zephyr/issues/66326](https://github.com/zephyrproject-rtos/zephyr/issues/66326)

cgbdevelco, 2025-04-11:
@HoZHel Do you have any Nucleo based on the STM32H7 series that you can test this shield on?

HoZHel, 2025-04-11:
Hi @cgbdevelco, at the moment I don't have a Nucleo-H755ZI (your target board); however, yesterday I received Nucleo-H723ZG. I'll try to check the sample and investigate possible issues.

cgbdevelco, 2025-04-22:
@HoZHel any updates? I created a ticket on ST's forum and they were able to replicate the issue and suspect it is related to H7 SPI CLK management in Zephyr.

cgbdevelco, 2025-05-28:
@FRASTM Any updates? HozHel provided a temporary fix to the low level spi driver but it would be nice with a proper fix in upstream Zephyr

erwango, 2025-05-28:
> I recently add the https://github.com/zephyrproject-rtos/zephyr/pull/90263

Note that it is only a temp PR for verifying the fix.
Before integrating it in the driver, we need to better understand the underlying issue (what is specific with this shield, with this board/series) to be sure the fix the root cause.

cgbdevelco, 2025-06-25:
@FRASTM with #92117 the BLE shield and SPI comms on other buses works. Thanks!

HoZHel, 2025-06-26:
I tested with SPI1 which is connected to Arduino pins, and still it doesn't work.

cgbdevelco, 2025-06-27:
@HoZHel 
I assume you have used overlay file to change the sck from bias-pull-down to bias-pull-up?

I use the following overlay file for the nucleo h755

```
/ {
	 chosen {
		 zephyr,bt-hci = &hci_spi;
		 zephyr,bt-c2h-uart = &usart3;
	 };
};

&spi1_sck_pa5 {
	/delete-property/ bias-pull-down;
	bias-pull-up;
};

&spi1_miso_pa6 {
	slew-rate = "high-speed";
};

&spi1_mosi_pb5 {
	slew-rate = "high-speed";
};

&spi1 {
	status = "okay";
	pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pb5>;
	pinctrl-names = "default";

	cs-gpios = <&arduino_header 16 GPIO_ACTIVE_LOW>;	      /* D10 */

	hci_spi: wb05n@0 {
		compatible = "st,hci-spi-v2";
		reg = <0>;
		reset-gpios = <&arduino_header 13 GPIO_ACTIVE_LOW>; /* D7 */
		irq-gpios = <&arduino_header 0 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>;  /* A0 */
		spi-cpol; /* CPOL=1 */
		spi-cpha; /* CPHA=1 */
		spi-hold-cs;
		spi-max-frequency = <DT_FREQ_M(8)>; /* the maximum supported SPI speed */
		reset-assert-duration-ms = <6>;
		status = "okay";
	};
};	
```

HoZHel, 2025-06-27:
@cgbdevelco, Yes, I used the same overlay that you put. Yesterday, I tried multiple times, but it didn't pass this part:
`if (k_sem_take(&sem_initialised, K_SECONDS(CONFIG_BT_SPI_BOOT_TIMEOUT_SEC)) < 0) {
		return -EIO;
}`
It reached timeout and returned negative value. However, surprisingly it works fine today, and the only reason that comes to my mind is the hardware connection issue (loose connection between the shield and Arduino pins). I also tested `peripheral_hr` sample and it works fine after defining `green_led`.

closed 2025-06-27 by danieldegrasse