zephyrproject-rtos/zephyr #86977: Support `IP_MULTICAST_LOOP` socket option

pdgendt, 2025-03-12:
**Is your feature request related to a problem? Please describe.**

I'd like to have the option to enable/disable socket multicast loopback, determining whether or not transmitted messages are delivered to the sending host.

**Describe the solution you'd like**

Similar to how this is provided in [linux](https://man7.org/linux/man-pages/man7/ip.7.html), a Zephyr implementation.

**Describe alternatives you've considered**

None

**Additional context**

sayooj-aerlync, 2025-03-15:
I am looking into this. Starting with a linux multicast socket!

sayooj-aerlync, 2025-03-15:
Alright @Abd002 👍

jukkar, 2025-03-17:
@sayooj-aerlync As you were also interested implementing this, may I suggest you implement support for IPv6 variant of this option `IPV6_MULTICAST_LOOP`?

sayooj-aerlync, 2025-03-17:
@jukkar Sure! I am working on it! I had a chat with @pdgendt in Discord on the same :)

sayooj-aerlync, 2025-03-18:
@jukkar @pdgendt I am done with the coding for IPV6_MULTICAST_LOOP. I will rasie the pull request after I complete the tests.

sayooj-aerlync, 2025-03-19:
Please find the pull request for IPV6_MULTICAST_LOOP here: https://github.com/zephyrproject-rtos/zephyr/pull/87338
Test logs: https://github.com/sayooj-aerlync/zephyr-unit-test/tree/main/ipv6_mcast_loop_sample_and_log

PR #87167 (2025-03-24): Add support for IP_MULTICAST_LOOP socket option
net: sockets: Add support for IP_MULTICAST_LOOP socket option

This PR adds support for the IP_MULTICAST_LOOP socket option,
which controls whether multicast packets are looped back to the
sending host.

Fixes: #86977
closed 2025-03-24 by kartben