zephyrproject-rtos/zephyr #86990: nanopb: header and source are not generated before build the target.

iabdalkader, 2025-03-12:
**Describe the bug**

I'm working on a driver that needs `nanopb`, following instructions here https://docs.zephyrproject.org/latest/services/serialization/nanopb.html and samples/examples, I made the following CMakeLists.txt:
```
if (CONFIG_MY_DRIVER)

list(APPEND CMAKE_MODULE_PATH ${ZEPHYR_BASE}/modules/nanopb)
include(nanopb)

zephyr_nanopb_sources(app
  my_driver_proto.pb
)

zephyr_library_sources(
  my_driver.c
  ${ZEPHYR_BASE}/build/zephyr/drivers/my_driver/my_driver_proto.pb.c
)

zephyr_include_directories(
  ./
  ${ZEPHYR_BASE}/build/zephyr/drivers/my_driver/
)

endif()
```

When I build a sample that uses this driver, the auto-generated `.c` and `.h` files are Not generated first, so `my_driver_proto.pb.c` doesn't exist. If I comment out this source file, the build proceeds and the files get generated.

**To Reproduce**

It will take some doing to reproduce, but you need a driver that uses nanopb, and a sample that uses/enables this driver.

**Expected behavior**
Expected files to be generated first before building the target.

**Impact**

Showstopper, can't finish this driver without this working, but as a workaround, I've added the auto-generated files manually.

**Logs and console output**
```bash
zephyr/cmake/modules/extensions.cmake:515 (target_sources):
  Cannot find source file:
    /home/zephyr/drivers/my_driver/my_driver_proto.pb.c
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK

pdgendt, 2025-03-12:
I see multiple issues:

- You don't need to add the generated c files to a target, that's what `zephyr_nanopb_sources` does for you.
- You're mixing targets, the `nanopb` sources are added to `app` while `zephyr_library_sources` adds sources to the current library target (the name depends on where you call `zephyr_library()` or `zephyr_library_named(<some name>)`)
- Is the protobuf used only in the library or also in the app? I'm confused

I'd expect something like

```cmake
if (CONFIG_MY_DRIVER)

list(APPEND CMAKE_MODULE_PATH ${ZEPHYR_BASE}/modules/nanopb)
include(nanopb)

# Create a named target
zephyr_library_named(my_driver)

# The following will make sure the source file is added to the target
# and `my_driver_proto.pb.h` can be included.
zephyr_nanopb_sources(my_driver
  my_driver_proto.pb
)

# Other source files, not generated ones
zephyr_library_sources(
  my_driver.c
)

endif()
```

iabdalkader, 2025-03-12:
> You don't need to add the generated c files to a target, that's what zephyr_nanopb_sources does for you.

Ah I see. I thought we needed to add the generated file manually.

> * Is the protobuf used only in the library or also in the app? I'm confused

It's only used by the driver, sorry for the confusion, `app` was supposed to be the library.

After fixing these issues, it's now working as expected. Thanks @pdgendt 🙏🏼

closed 2025-03-12 by iabdalkader