zephyrproject-rtos/zephyr #87013: Overlay Parse Error in Zephyr 4.1.0-rc2 on NUCLEO-WB55RG

GuboDupy, 2025-03-13:
Building with an overlay file (`nucleo_wb55rg.overlay`) fails with a devicetree parse error on Zephyr 4.1.0-rc2 for NUCLEO-WB55RG. Moving changes to `nucleo_wb55rg.dts` works.

**Steps to Reproduce**:
1. Set up Zephyr 4.1.0-rc2 in `C:\Users\g4hol\zephyrproject`.
2. Create `samples\My_blink\boards\nucleo_wb55rg.overlay`:
  
    /dts-v1/;
    &usart1 {
    status = "okay";
    };

4. Error: `devicetree error: parse error: expected '/' or label reference (&foo).
    Checked file with hex editor to ensure no bad characters at beginning of file. 

**Expected Behavior**:
Overlay applies without parse errors, enabling `usart1`.

**Actual Behavior**:
Build fails with parse error. Works if changes are in `boards\nucleo_wb55rg.dts`.

**Environment**:
- OS: Windows Version	10.0.26100 Build 26100
- Zephyr: 4.1.0-rc2
- West: [e.g., v1.3.0]
- Python: [e.g., 3.11.2]
- CMake: [e.g., 3.31.5]
- DTC: [e.g., 1.6.1]
- Compiler: [e.g., arm-zephyr-eabi-gcc (Zephyr SDK 0.16.8)]
- Path: `C:\Users\g4hol\zephyrproject`

**CMD window compile with error**
______________________________________________________________________
(.venv) C:\Users\g4hol\zephyrproject\zephyr\samples\My_blink>west -v build -b nucleo_wb55rg .
-- west build: generating a build system
cmake version 3.31.5 is OK; minimum version is 3.13.1
Running CMake: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Users/g4hol/zephyrproject/.venv/Scripts/python.exe '-BC:\Users\g4hol\zephyrproject\zephyr\samples\My_blink\build' -GNinja '-SC:\Users\g4hol\zephyrproject\zephyr\samples\My_blink'
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: C:/Users/g4hol/zephyrproject/zephyr/samples/My_blink
-- CMake version: 3.31.5
-- Cache files will be written to: C:/Users/g4hol/zephyrproject/zephyr/.cache
-- Zephyr version: 4.1.0-rc2 (C:/Users/g4hol/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nucleo_wb55rg, qualifiers: stm32wb55xx
-- Found host-tools: zephyr 0.17.0 (C:/Users/g4hol/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (C:/Users/g4hol/zephyr-sdk-0.17.0)
-- Found BOARD.dts: C:/Users/g4hol/zephyrproject/zephyr/boards/st/nucleo_wb55rg/nucleo_wb55rg.dts
-- Found devicetree overlay: C:/Users/g4hol/zephyrproject/zephyr/samples/My_blink/boards/nucleo_wb55rg.overlay
devicetree error: C:/Users/g4hol/zephyrproject/zephyr/samples/My_blink/boards/nucleo_wb55rg.overlay:1 (column 1): parse error: expected '/' or label reference (&foo)
CMake Error at C:/Users/g4hol/zephyrproject/zephyr/cmake/modules/dts.cmake:303 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  C:/Users/g4hol/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  C:/Users/g4hol/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Users/g4hol/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:2 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Users/g4hol/zephyrproject/.venv/Scripts/python.exe '-BC:\Users\g4hol\zephyrproject\zephyr\samples\My_blink\build' -GNinja '-SC:\Users\g4hol\zephyrproject\zephyr\samples\My_blink'
_________________________________________________________________________________

nordicjm, 2025-03-13:
`/dts-v1/;` is invalid for overlay files, remove it and it works e.g.
```
&usart1 {
        status = "okay";
};`
```

closed 2025-03-13 by nordicjm

FRASTM, 2025-03-13:
@GuboDupy 
 
yes, The `/dts-v1/;`  is alrdeay in the board DTS, it should not be elsewhere: 
"parse error: expected '/dts-v1/;' at start of file"