zephyrproject-rtos/zephyr #87018: Compilation Error with ARM Cortex-M55 MVE Function in Zephyr SDK v0.17

ZhiyuanTang17, 2025-03-13:
Describe the bug
I encountered a compilation error related to an ARM Cortex-M55 MVE function while using the Zephyr SDK toolchain version 0.17. The error message is: error: argument 3 must be a constant immediate. This issue does not occur when using the GNU Arm Embedded toolchain, suggesting that the Zephyr SDK toolchain might not support the function correctly.

Target platform: ARM Cortex-M55
Workaround attempts: Switched to GNU Arm Embedded toolchain, which resolved the issue
Regression: Occurs with Zephyr SDK version 0.17
To Reproduce
Steps to reproduce the behavior:

1. Set up the environment with Zephyr SDK v0.17.
2. choose a CM55 board with MVE enabled.
3. Use the following test code:```
const int d = 5;
int main(void)
{
    uint16x8_t a = vdupq_n_u16(0xff);
    uint16x8_t b = vdupq_n_u16(0xff);
    uint16x8_t c;
    c = vsriq_n_u16(a, b, d);
}
```
Compile the code.
You will see the error regarding the third argument needing to be a constant immediate.
Expected behavior
The code should compile without error, as it does when using the GNU Arm Embedded toolchain.

Impact
This issue prevents the successful compilation of applications using the ARM Cortex-M55 MVE functions with Zephyr SDK v0.17, obstructing progress on development.

Logs and console output
This is the relevant part of the error message:

```
In file included from C:/ACODE/bb2u-zephyr/zephyrproject/samples/hello_world/src/main.c:8:
In function '__arm_vsriq_n_u16',
    inlined from 'main' at C:/ACODE/bb2u-zephyr/zephyrproject/samples/hello_world/src/main.c:15:7:
c:\users\zhiyuan_tang\zephyr-sdk-0.17.0\arm-zephyr-eabi\lib\gcc\arm-zephyr-eabi\12.2.0\include\arm_mve.h:7678:10: error: argument 3 must be a constant immediate 7678 |   return __builtin_mve_vsriq_n_uv8hi (__a, __b, __imm);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
Environment (please complete the following information):

OS: Windows
Toolchain: Zephyr SDK v0.17
Commit SHA or Version used: Zephyr v4.0.0

Additional context
The issue is circumvented by using the GNU Arm Embedded toolchain, which suggests an incompatibility or lack of support in the Zephyr SDK toolchain for this specific function.

wearyzen, 2025-03-13:
@ZhiyuanTang17, AFAIS, arm_mve.h defines vsriq_n_u16 as below:
#define vsriq_n_u16(__a, __b,  __imm) __arm_vsriq_n_u16(__a, __b,  __imm)

and according to this the error is correct and in the latest [Arm GNU toolchain](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads), vsriq_n_u16 has been removed but the definition doesn't seem to be changed in between so I am bit curious which toolchain worked for you.
I might be wrong so would like to confirm what you version of toolchain has and for that could you share the version of the toolchain that worked for you?

ZhiyuanTang17, 2025-03-14:
Hi @stephanosio,

I’m currently using the latest Arm GNU Toolchain (version 14.2 Re1 for Windows) [arm-gnu-toolchain-14.2.rel1-mingw-w64-i686-arm-none-eabi.zip]. After verifying with this toolchain, I noticed something odd. The build completes successfully; however, upon examining the disassembly, it appears that the vsriq_n_u16 instruction is bypassed by the compiler:

Disassembly of section .text:

```
700a19a8 <main>:
#include <stdio.h>
#include <arm_mve.h>
const int d = 5;
#define D 6
int main(void)
{
700a19a8:	b508      	push	{r3, lr}
	uint16x8_t a = vdupq_n_u16(0xff);
    uint16x8_t b = vdupq_n_u16(0xff);
    uint16x8_t c;
    c = vsriq_n_u16(a, b, D);
	
	printf("Hello World! %s\n", CONFIG_BOARD_TARGET);
700a19aa:	4903      	ldr	r1, [pc, #12]	@ (700a19b8 <main+0x10>)
700a19ac:	4803      	ldr	r0, [pc, #12]	@ (700a19bc <main+0x14>)
700a19ae:	f002 fd07 	bl	700a43c0 <printf>

	return 0;
}
700a19b2:	2000      	movs	r0, #0
700a19b4:	bd08      	pop	{r3, pc}
700a19b6:	bf00      	nop
700a19b8:	700a939c 	.word	0x700a939c
700a19bc:	700a93b3 	.word	0x700a93b3
```

This behavior is really puzzling.

Let me know if there’s anything else you need or if you have further questions!

wearyzen, 2025-03-14:
Hi @ZhiyuanTang17, if you look at the `arm_mve.h` file in the arm-gnu-toolchain-14.2.rel1-mingw-w64-i686-arm-none-eabi.zip `lib/gcc/arm-none-eabi/14.2.1/include/arm_mve.h` you'll notice that it doesn't have the function `vsriq_n_u16`. I am a bit confused why the compiler didn't warn you about this but in any case this doesn't seem to be an issue with the Zephyr SDK and IIUC you should probably find an alternative to this function since it is not available in the latest release of the other toolchain as well.

ZhiyuanTang17, 2025-03-17:
@wearyzen  👍 
Thanks for your suggestions. I will consider to find an alternative way.

tejlmand, 2025-04-28:
@ZhiyuanTang17 afaict then this isn't a Zephyr nor Zephyr SDK issue.

Please comment if you still believe this to be a Zephyr issue. or should be fixed by Zephyr or close the issue report.

closed 2025-05-12 by tejlmand