zephyrproject-rtos/zephyr #87021: drivers: flash: flash_mcux_flexspi_nor incorrectly handles border case

Laczen, 2025-03-13:
**Describe the bug**

#84899 introduced input parameter checking for flash read operations.

The introduced check incorrectly handles the corner case where a read is performed of size 0 into a NULL array. The read fails with error code `-EINVAL`.

This introduces a difference between platforms when e.g. `nvs_read` is used to retrieve the size of an already stored element as reported in #86891.

**To Reproduce**

See #86891.

**Expected behavior**

No difference between platforms.

**Logs and console output**

See #86981.

**Environment (please complete the following information):**

See #86981.

JordanYates, 2025-03-13:
I think you transposed some numbers in those issues, I doubt the reproduction case is a LoRa PR :)

Laczen, 2025-03-13:
> I think you transposed some numbers in those issues, I doubt the reproduction case is a LoRa PR :)

Updated, thanks :)

dleach02, 2025-03-14:
@Laczen do you have a recommended fix?

Laczen, 2025-03-14:
> [@Laczen](https://github.com/Laczen) do you have a recommended fix?

Return 0 when the read/write length is 0 before checking the read buffer.

commit c0c1b0af (2025-05-30): drivers: flash: Align NXP flash drivers on parameter checking

Parameter checking of flash read API expects checking for len == 0
before checking dest buffer validation.

Fixes #87021
closed 2025-05-30 by kartben