zephyrproject-rtos/zephyr #87024: Build failure when attempting to use external Bluetooth module with nRF53

skordal, 2025-03-13:
**Describe the bug**
I am attempting to use an nRF53 as host for an external IW612-based Bluetooth module. However, I am getting a build failure due to multiple definitions of `bt_hci_transport_setup()`.

This is caused by Zephyr always including `drivers/bluetooth/hci/nrf53_support.c` in the build for nRF53, while also including `drivers/bluetooth/hci/hci_nxp_setup.c` due to my build settings (see below). Both drivers implement `bt_hci_transport_setup()`. The nRF53 support file is always included when `CONFIG_SOC_NRF5340_CPUAPP` is defined, which makes it impossible to disable through project configuration. Although you would usually want to use the nRF53 Bluetooth support, it would be nice to have the option of disabling this driver and use an external module instead.

Looking at `drivers/bluetooth/hci/CMakeLists.txt`, this is the only HCI driver which is not controlled by its own Kconfig setting. A possible solution could be to add a `CONFIG_BT_NRF53` Kconfig option to control whether to include this driver in the build.

**To Reproduce**
* Build for the `nrf5340dk/nrf5340/cpuapp` platform
* Add `CONFIG_BT_H4_NXP_CTLR=y` to prj.conf
* Add the external HCI controller to the device tree, using something like:
```
/ {
    chosen {
        zephyr,bt-hci = &bt_hci_uart;
    };
};

&uart1 {
    status = "okay";

    bt_hci_uart: bt_hci_uart {
        compatible = "zephyr,bt-hci-uart";
        status = "okay";

        bt_module: bt_module {
            compatible = "nxp,bt-hci-uart";
            status = "okay";

            hw-flow-control;
            hci-operation-speed = <1000000>;

            sdio-reset-gpios = <&gpio1 6 GPIO_ACTIVE_HIGH>;
            w-disable-gpios  = <&gpio1 7 GPIO_ACTIVE_HIGH>;

            fw-download-primary-speed = <115200>;
            fw-download-secondary-speed = <1000000>;
        };
    };
};
```

**Expected behavior**
Only one HCI driver should be included in the build.

**Impact**
It is currently not possible to use an nRF53 as host for an external Bluetooth controller over HCI.

jhedberg, 2025-03-18:
> A possible solution could be to add a `CONFIG_BT_NRF53` Kconfig option to control whether to include this driver in the build

This seems like a reasonable solution to me. The submodule could even have its own DT compat, like NXP and be a sub-node of the `zephyr,bt-hci-uart` node.

dleach02, 2025-03-19:
General question but why can't a Zephyr platform/application support multiple BT controllers?

jhedberg, 2025-03-19:
> General question but why can't a Zephyr platform/application support multiple BT controllers?

It can, i.e. it's completely fine to have multiple device tree node instances representing multiple controllers. The only limitation is that the Zephyr host stack is designed to use just one of them for any given build configuration, based on a device tree chosen property.

alwa-nordic, 2025-06-10:
> It can, i.e. it's completely fine to have multiple device tree node instances representing multiple controllers. The only limitation is that the Zephyr host stack is designed to use just one of them [...]

I'm not sure we can claim that right now. As this issue points out, `bt_hci_transport_setup` gets in the way of this since it can only be defined globally, not per-device.

Does `bt_hci_driver_api.setup` fulfill the same role as `bt_hci_transport_setup`? If so then let's deprecate `bt_hci_transport_setup`. Likewise with `bt_hci_driver_api.close` and `bt_hci_transport_teardown`?

dleach02, 2025-06-10:
Unassigning me from this since this doesn't seem to be directly related the the NXP external module integration... and it looks like @skordal is working on a fix.

jhedberg, 2025-06-11:
> I'm not sure we can claim that right now. As this issue points out, `bt_hci_transport_setup` gets in the way of this since it can only be defined globally, not per-device.

IIRC that was originally an H:4 internal thing, which was generalized since Nordic wanted to use the same implementation with other drivers. It's not the same as the `setup()` callback of HCI drivers. `drv.setup()` runs in a host context and expects a functional HCI transport, whereas `transport_setup()` is a purely driver-internal thing before the HCI transport is functional. Both of these APIs are problematic, but in different ways. `drv.setup()` implementations typically depend on host APIs (command allocation, encoding & sending) which creates a circular dependency and assumes a specific user of the driver (the Zephyr host stack).

alwa-nordic, 2025-06-11:
Btw, I remember we discussed something that may be a bit relevant here in one of our WG meetings: I pointed out that we might want to have separate driver nodes for the HCI transport (bus-like) and the Controller to avoid putting Controller-specific code in HCI transport code. Here is some rambling elaborating on that point:

The transport setup could be responsible for everything up to and including the HCI reset. I think it could make sense to include the HCI reset as a transport responsibility because of H4 using HCI reset as a synchronization keyword.

The Controller driver would only exist to have its setup function invoked, right after the HCI reset, before the Host sends anything but when sending commands is possible already. This could help keep a clean separation between Controller-specific support code and HCI transport code, and also have the effect that a Controller firmware upload could be activated based on the Controller device node on that transport, not tied to the particular transport itself.

jhedberg, 2025-06-11:
>  I pointed out that we might want to have separate driver nodes for the HCI transport (bus-like) and the Controller to avoid putting Controller-specific code in HCI transport code

We partially of have that already, don't we? E.g. H:4 controller nodes are child nodes of UART nodes.

closed 2025-08-26 by github-actions[bot]