zephyrproject-rtos/zephyr #87039: Help with modem driver writing

Captainfl4me, 2025-03-13:
Hello,

I'm trying to write a driver for a custom new NB-IoT chip (as part of a university project). I've write some of the required file for this to work: dts bindings, shields support (for easier overlay), Kconfig modem configuration file and I'm writing the C skeleton file to try having compiling driver without the logic at first. I would like to have basic working modem_shell capability before doing proper net offloading.

However, I'm stuck on a linking error when building my test app `Undefined initialization levels used`. This seems to be related to wrong device_tree or init function registering but cannot find the real root cause of this. Any ideas or documentation that could help me ?

Here is the files I used:
## Fork Zephyr
- `drivers/modem/Kconfig.st87m01`
```
config MODEM_ST87M01
	bool "ST87M01 modem driver"
	select MODEM_RECEIVER
	select NET_OFFLOAD
	imply GPIO
	help
	  Choose this setting to enable ST87M01 NB-IoT modem driver.

if MODEM_ST87M01
	config MODEM_ST87M01_INIT_PRIORITY
	int "ST87M01 driver init priority"
	default 80
	help
	  ST87M01 device driver initialization priority.
	  Do not mess with it unless you know what you are doing.
	  Note that the priority needs to be lower than the net stack
	  so that it can start before the networking sub-system.
```
- `drivers/modem/st87m01.c`
```c
#include <zephyr/logging/log.h>
#include <zephyr/net/net_if.h>
#include "st87m01.h"

LOG_MODULE_REGISTER(modem_st87m01, CONFIG_MODEM_LOG_LEVEL);

/*
 * Initializes modem handlers and context.
 * After successful init this function calls
 * modem_setup.
 */
static int modem_init(const struct device *dev)
{
	int ret = 0;
	ARG_UNUSED(dev);
	LOG_DBG("ST87M01 Init");

	mctx.data_manufacturer = mdata.mdm_manufacturer;
	mctx.data_model = mdata.mdm_model;
	mctx.data_revision = mdata.mdm_revision;
#if defined(CONFIG_MODEM_SIM_NUMBERS)
	mctx.data_imei = mdata.mdm_imei;
	mctx.data_imsi = mdata.mdm_imsi;
#endif
	mctx.data_rssi = &mdata.mdm_rssi;

	ret = mdm_receiver_register(&mctx, MDM_UART_DEV, mdm_recv_buf, sizeof(mdm_recv_buf));
	if (ret < 0) {
		LOG_ERR("Error registering modem context: %d", ret);
		goto error;
	}

	modem_reset();

error:
	return ret;
}

static void modem_reset(void)
{
[... 158 lines cut ...]
Merged configuration '/home/nicoth/projects/ST-project-workspace/applications/at-shell-app/prj.conf'
Configuration saved to '/home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build/zephyr/.config'
Kconfig header saved to '/home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/nicoth/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nicoth/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done (7.7s)
-- Generating done (0.2s)
-- Build files have been written to: /home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build
-- west build: building application
[1/201] Preparing syscall dependency handling

[2/201] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.0-rc2 (/home/nicoth/projects/ST-project-workspace/zephyr), build: 1d169fb16731
[196/201] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build/zephyr/zephyr_pre0.map
: && ccache /home/nicoth/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/cortex_m/cmse/libarch__arm__core__cortex_m__cmse.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/options/liblib__posix__options.a  zephyr/lib/net_buf/liblib__net_buf.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/net/l2/offloaded_netdev/libsubsys__net__l2__offloaded_netdev.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/modem/libdrivers__modem.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF -L"/home/nicoth/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v8-m.main/nofp" -lc -lgcc && cd /home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build/zephyr && /usr/bin/cmake -E true
/home/nicoth/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: Undefined initialization levels used.
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/nicoth/projects/ST-project-workspace/applications/at-shell-app/build
```

pdgendt, 2025-03-13:
Kconfig options in C have the `CONFIG_` prefix, so:

```c
/* Register device with the networking stack. */
NET_DEVICE_DT_INST_OFFLOAD_DEFINE(0, modem_init, NULL, &mdata, NULL, CONFIG_MODEM_ST87M01_INIT_PRIORITY,
				  &api_funcs, MDM_MAX_DATA_LENGTH);
```

Captainfl4me, 2025-03-13:
Oh thank you so much! I already knew that, but despite several rereads I hadn't noticed the mistake.

pdgendt, 2025-03-13:
> Oh thank you so much! I already knew that, but despite several rereads I hadn't noticed the mistake.

If this resolves your issue, please close it, or I would convert it to a discussion instead.

Captainfl4me, 2025-03-13:
Maybe it would be better to convert it to a discussion as I may have other questions regarding this implementation. This would avoid creating multiple issues in the future.