zephyrproject-rtos/zephyr #87047: net: lib: sockets: Heap mem pool  for SOCKETPAIR is too small

IVandeVeire, 2025-03-13:
**Describe the bug**

The default value of `HEAP_MEM_POOL_ADD_SIZE_SOCKETPAIR` (255) seems to be insufficient to create a `socket_pair`.

To create a socket pair using `z_impl_zsock_socketpair` from `socketpair.c`, `spair_new()` is executed twice. `spair_new()` does a `malloc()` of `sizeof(struct spair)` (148) bytes. This results in a total of 296 bytes of heap usage.

Since the heap is only 256 bytes, this will fail.

Is this derised behaviour? Or should the default size be increased?
Also, `sizeof(struct spair)`  depends on `CONFIG_NET_SOCKETPAIR_BUFFER_SIZE`. Should this somehow be taken into account?

**To Reproduce**
There is no direct way to reproduce using the zephyr tree. I stumbled apon the issue while developing a `coap_server` for `native_sim`. The issue was introduced in 671f0f6cfae1a78cda1d07720aaaae1e675af16f.

**Impact**

Coap server thread stops running, so my device is not reachable over coap.

**Environment (please complete the following information):**

 - OS: Linux
 - Commit SHA or Version used: main since 671f0f6cfae1a78cda1d07720aaaae1e675af16f

jukkar, 2025-03-14:
> The default value of `HEAP_MEM_POOL_ADD_SIZE_SOCKETPAIR` (255) seems to be insufficient to create a `socket_pair`.
>
> To create a socket pair using `z_impl_zsock_socketpair` from `socketpair.c`, [...]

If there is only one entity like socketpair wanting heap, then this is indeed problem. If there are other entities adding their requirements to the heap size, then the issue is probably not seen so often.

> 
> Since the heap is only 256 bytes, this will fail.
> 
> Is this derised behaviour? Or should the default size be increased? Also, `sizeof(struct spair)` depends on `CONFIG_NET_SOCKETPAIR_BUFFER_SIZE`. Should this somehow be taken into account?

Yes, we should allocate the enough space for the two spair struct.

Would you be able to send a fix for this, so increase the default value from 255 to what ever is needed?
Also it might make sense to add build asserts in the code so that if there is not enough space allocated, then the build will fail so we can detect this issue earlier. With the build assert, we do not need to specifically check the value of `CONFIG_NET_SOCKETPAIR_BUFFER_SIZE`.

carlescufi, 2025-04-23:
@IVandeVeire will you open a Pull Request to fix this?

IVandeVeire, 2025-04-23:
> [@IVandeVeire](https://github.com/IVandeVeire) will you open a Pull Request to fix this?

I created a PR https://github.com/zephyrproject-rtos/zephyr/pull/88962

closed 2025-04-25 by kartben