zephyrproject-rtos/zephyr #87054: flexcomm spi asynchroneous mode not working

badevos, 2025-03-13:
I've tested spi on the NXP RW612 in both sync and async mode, which did not give the expected results.

When I use [spi_transceive_cb](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/spi.h#L924) I see that the the given callback is executed prior to the return of the function `spi_transceive_cb` itself. 

The board uses this driver: [spi_mcux_flexcomm.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/spi/spi_mcux_flexcomm.c)

I've used Percepio TraceAlyzer to trace this:
I have added manual trace points, visible in 'yellow'
[app] spi-tx-beg: trace prior to call of function `spi_transceive_cb`
[app] spi-tx-end: trace right after the function `spi_transceive_cb` has returned
[app] spi-tx-cb: trace in the callback provided as argument to `spi_transceive_cb`

Synchronous:
[image]

Asynchronous: callback executed prior to the return point of spi_transceive_cb
[image]

In the screenshots I've also tried to measure the absolute timings, however due to the overhead of tracing itself this can be misleading. The actual data being sent out is the string `"abcdefghijklmnopqrstuvwxyz"` on 5Mhz, which takes about 41us. With the tracing overhead it's hard to find this particular timings.
On the other hand, +/- 300us seems quiete a lot, given the sitution that there is a trace less, and no interrupt context switch.

This has been tested on 4.0.0 as well as the current main, i.e. commit 4bb5ffd7867d9b497fa8f03d73973ab2e32c69ba, both with zephyr sdk 0.17.0 on Linux/Ubuntu.

This is my test code:
```
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
#include <zephyr/sys/printk.h>
#include <zephyr/shell/shell.h>
#include <zephyr/drivers/spi.h>

#ifdef CONFIG_PERCEPIO_TRACERECORDER
#include "trcRecorder.h"
#include "trcTypes.h"
traceString app_trace_channel;
#endif

LOG_MODULE_REGISTER(spi, LOG_LEVEL_DBG);

/* spi configuration */
static const struct device *spi_dev = DEVICE_DT_GET(DT_ALIAS(spi1));

static struct spi_config spi_cfg = {
    .frequency = DT_PROP(DT_ALIAS(spi1), clock_frequency),
    .operation = SPI_OP_MODE_MASTER | SPI_WORD_SET(8) | SPI_TRANSFER_MSB,
    .slave = 0,
    .cs = {
        .gpio = {NULL}, // No GPIO for CS
        .delay = 0,     // No delay
    },
};

static void spi_cb_done(const struct device * dev, int result, void * data);

/* spi shell commands */
static int cmd_spi_send(const struct shell *sh, size_t argc, char **argv)
{
    if (!device_is_ready(spi_dev))
    {
        LOG_ERR("Device SPI not ready, aborting test");
        return -ENODEV;
    }

    char *tx_buffer = argv[1];

    struct spi_buf tx_buf = {
        .buf = tx_buffer,
        .len = strlen(tx_buffer),
    };

    // LOG_DBG("SPI tx_buf.len = %d", tx_buf.len);

    struct spi_buf_set tx_bufs = { .buffers = &tx_buf, .count = 1 };

    // Send the string over SPI
    vTracePrint(app_trace_channel, "spi-tx-beg");
    int ret_val = spi_write(spi_dev, &spi_cfg, &tx_bufs);
    vTracePrint(app_trace_channel, "spi-tx-end");
    if (ret_val < 0)
    {
        shell_error(sh, "SPI write failed: %d", ret_val);
    }

    return ret_val;
}

static int cmd_spi_async(const struct shell *sh, size_t argc, char **argv)
{
    if (!device_is_ready(spi_dev))
    {
        LOG_ERR("Device SPI not ready, aborting test");
        return -ENODEV;
    }

    char *tx_buffer = argv[1];

    struct spi_buf tx_buf = {
        .buf = tx_buffer,
        .len = strlen(tx_buffer),
    };

    struct spi_buf_set tx_bufs = { .buffers = &tx_buf, .count = 1 };

    // Send the string over SPI
    vTracePrint(app_trace_channel, "spi-tx-beg");
    int ret_val = spi_transceive_cb(spi_dev, &spi_cfg, &tx_bufs, NULL, spi_cb_done, NULL);
    vTracePrint(app_trace_channel, "spi-tx-end");
    if (ret_val < 0)
    {
        shell_error(sh, "SPI write failed: %d", ret_val);
    }

    return ret_val;
}

void spi_cb_done(const struct device * dev, int result, void * data)
{
    vTracePrint(app_trace_channel, "spi-tx-cb");
}

SHELL_STATIC_SUBCMD_SET_CREATE(sub_spi_string,
    SHELL_CMD_ARG(send,  NULL, "SPI send",  cmd_spi_send,  2, 0),
    SHELL_CMD_ARG(async, NULL, "SPI async", cmd_spi_async, 2, 0),
SHELL_SUBCMD_SET_END);

SHELL_CMD_REGISTER(spi_string, &sub_spi_string, "SPI commands", NULL);

int main(void)
{
    app_trace_channel = xTraceRegisterString("app");

    while (1)
    {
        k_sleep(K_MSEC(1000));
    }
    return 0;
}
```

With the following `prj.conf`
```
CONFIG_SPI=y
CONFIG_SPI_ASYNC=y
CONFIG_CONSOLE=y
CONFIG_PRINTK=y
CONFIG_STDOUT_CONSOLE=y
CONFIG_LOG=y
CONFIG_SHELL=y
CONFIG_LOG_MODE_DEFERRED=y
CONFIG_DEBUG=y
CONFIG_SPI=y
CONFIG_DMA=y
CONFIG_SPI_MCUX_FLEXCOMM_DMA=y

CONFIG_TRACING=y
CONFIG_PERCEPIO_TRACERECORDER=y
CONFIG_PERCEPIO_TRC_START_MODE_START_AWAIT_HOST=y
CONFIG_PERCEPIO_TRC_CFG_ENTRY_SLOTS=1024
CONFIG_PERCEPIO_TRC_CFG_STACK_MONITOR_MAX_TASKS=32
```

Is this indeed a bug, or do I have the wrong expectation of `spi_transceive_cb`? In the latter case, how do I setup an asynchronous SPI transfer correctly?

badevos, 2025-03-13:
I've analyzed some code myself and found this part in the driver:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/spi/spi_mcux_flexcomm.c#L691

```
		/* wait until TX FIFO is really empty */
		while (0U == (base->FIFOSTAT & SPI_FIFOSTAT_TXEMPTY_MASK)) {
		}
```
I tested my code without this check, and this has no effect.

badevos, 2025-03-17:
Hi @dleach02, 

I see you gave this a low priority. 

Can you confirm that my reasoning is correct, and `spi_transceive_cb` should return earlier than execution of the completion callback?

dleach02, 2025-03-25:
Sorry, this is a Zephyr Project rating and not an internal NXP rating. Zephyr Project tends to put issues that only affect one platform as low priority. 

We have pulled the details of your ticket into our internal tracking system and reviewing it.

DerekSnell, 2025-03-26:
Hi @badevos ,
I can confirm `spi_transceive_cb()` functions as expected, and may not return before the callback executes.

To test this, I ran the [spi_loopback](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/spi/spi_loopback) test on the FRDM-RW612 board using Zephyr v4.1.  For this test to pass, the MOSI and MISO signals must be shorted externally, by shorting header J1 pins 2 and 4.  This test also tests Async mode, and I enabled DMA to match your use-case.

The spi_loopback test calls `spi_transceive_signal()` for the async test, instead of `spi_transceive_cb()`.  And that signal is used to trigger a thread function called `spi_async_call_cb()`.  Similar to what you found, I see that the thread `spi_async_call_cb()` is triggered and executes before `spi_transceive_signal()` completes.

I modified the spi_loopback test with the attached patch [spi_loopback-spi_transceive_cb.txt](https://github.com/user-attachments/files/19466315/spi_loopback-spi_transceive_cb.txt) to use `spi_transceive_cb()` instead of `spi_transceive_signal()`. Apply this patch to v4.1, and this test still passes.  I ran in a debugger with breakpoints instead of trace, but I see the same results: the callback `spi_cb_done()` is called before `spi_transceive_cb()` returns.  

Below is a screenshot from the callstack when the callback `spi_cb_done()` breakpoint is triggered.  You can see this callback is called within the interrupt context, in this case triggered by the DMA interrupt.  With interrupts (or DMA) enabled, starting a SPI transfer will immediately trigger the TX interrupt to start filling the FIFOs from the TX buffers.  Much of the driver is managed in an ISR, which preempts `spi_transceive_cb()`.  The callback is also called within the ISR in the interrupt context.  So depending on the length of the SPI transfer and what else is executing in the app, the callback could be called before the transceive API completes.  To me, there is no requirement for `spi_transceive_cb()` that it must complete and return before the callback executes.

[image]

Since this does not appear to be a bug or an issue with the driver, I am converting this to a Discussion.  If you have further questions, we can discuss further in that Discussion.  Best regards