zephyrproject-rtos/zephyr #87059: drivers: pwm: pwm_gpio_loopback: drivers.pwm.gpio_loopback.nrf fails

katgiadla, 2025-03-13:
**Describe the bug**
The test `drivers/pwm/pwm_gpio_loopback/drivers.pwm.gpio_loopback.nrf` fails.

Observed for:
- `nrf54l15dk/nrf54l15/cpuapp`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -s drivers.pwm.gpio_loopback.nrf -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase" --fixture gpio_loopback`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-383-g92138f7ae718 ***
Running TESTSUITE pwm_gpio_loopback
===================================================================
START - test_pwm
Test case: [Channel: 0] [Period: 160000] [Pulse: 40000] [Inverted: No]
Sampling done: true
Measured period: 176 cycles, high: 41, low: 135 [unit: systimer ticks]
Measured period: 176000 ns, deviation: 10.0%
Measured duty: 23.29%, deviation: 6.81%
PWM output does NOT match the programmed values

    Assertion failed at zephyr/tests/drivers/pwm/pwm_gpio_loopback/src/main.c:213: test_run: (result not equal to TC_PASS)
Test case failed
 FAIL - test_pwm in 0.092 seconds
===================================================================
START - test_pwm_cross
Test case: [Channel: 0] [Period: 160000] [Pulse: 64000] [Inverted: No]
Sampling done: true
Measured period: 175 cycles, high: 79, low: 96 [unit: systimer ticks]
Measured period: 175000 ns, deviation: 9.37%
Measured duty: 45.14%, deviation: 12.85%
PWM output does NOT match the programmed values

    Assertion failed at zephyr/tests/drivers/pwm/pwm_gpio_loopback/src/main.c:213: test_run: (result not equal to TC_PASS)
Test case failed
 FAIL - test_pwm_cross in 0.092 seconds
===================================================================
TESTSUITE pwm_gpio_loopback failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [pwm_gpio_loopback]: pass = 0, fail = 2, skip = 0, total = 2 duration = 0.184 seconds
 - FAIL - [pwm_gpio_loopback.test_pwm] duration = 0.092 seconds
 - FAIL - [pwm_gpio_loopback.test_pwm_cross] duration = 0.092 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA or Version used:  `v4.1.0-383-g92138f7ae718`
 
**Addtional information**
Probably introduced by f44a30109cfb3eab5992b09f34bc61d2ccc047b1

nordic-piks, 2025-03-14:
@nordic-pikr FYI

nordic-piks, 2025-03-14:
@gmarull It was found that from this commit issue starts to occur:
https://github.com/zephyrproject-rtos/zephyr/commit/f44a30109cfb3eab5992b09f34bc61d2ccc047b1
Can you check?

closed 2025-05-12 by katgiadla