zephyrproject-rtos/zephyr #87085: [STM32WB55] power consumption about STOP2

LucienKUANG-vip, 2025-03-14:
**Describe the bug**
When developing a BLE application with the STM32WB, I found that the power consumption in STOP2 low-power mode is higher than what is stated in the official documentation. In STOP2 mode, the average current is around 75 µA.
However, the official documentation STOP2  status that the average current can reach approximately 2.1 µA.

**To Reproduce**
1. using the demo :C:\Users\user name\zephyrproject\zephyr\samples\boards\st\power_mgmt\stm32wb_ble  and using NUCLEO-WB55.
2. change the main.c
```
int main(void)
{
	int err;

	printk("Starting Beacon Demo\n");
	/* Initialize the Bluetooth Subsystem */
	err = bt_enable(bt_ready);
	if (err) {
		printk("Bluetooth init failed (err %d)\n", err);
	}

	/* Give time to bt_ready sequence */
	k_sleep(K_SECONDS(6));

	printk("BLE disable\n");
	err = bt_disable();
	if (err) {
		printk("Bluetooth disable failed (err %d)\n", err);
	}

	k_sleep(K_SECONDS(2));

	printk("BLE restart\n");
	/* Initialize the Bluetooth Subsystem */
	err = bt_enable(bt_ready);
	if (err) {
		printk("Bluetooth init failed (err %d)\n", err);
	}

	/* Give time to bt_ready sequence */
	k_sleep(K_SECONDS(6));

	printk("BLE disable\n");
	err = bt_disable();
	if (err) {
		printk("Bluetooth disable failed (err %d)\n", err);
	}

	//printk("Powering off\n");
	//sys_poweroff();
	while(1){
		k_sleep(K_SECONDS(2));
	}
	

	return 0;
}
```
3. Compile, download, and then run to observe the Realtime current.
[image]
5. When debugging the code, I found that the program does enter STOP2 mode.
[image]

6. Can you help to check why the stop2 real time power consumption is abnormal?

FRASTM, 2025-03-14:
Hi @LucienKUANG-vip 

what is the current  you are measuring ?
If this is the Vdd of the whole nucleo board, all the components are sinking current ( stlink part, rf,  leds etc,  not only the stm2wb55 mcu.
Can you measure current on the JP3 connector ?

[image]

LucienKUANG-vip, 2025-03-17:
Hello @FRASTM 
1. I removed all jumpers on connector JP5 and remove mini-USB.
2. measure current on the JP2 connector 
[image]
3.  In this way, On the demo BLE_HeartRate of CubeWB SDK, I can got the right power consumption.
https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/P-NUCLEO-WB55.Nucleo/Applications/BLE/BLE_HeartRate
but on wb55 with zephyr, I the average current is around 75 µA.

LucienKUANG-vip, 2025-03-17:
measure current on the JP3 connector ,the result is same around 74 µA.

[image]

LucienKUANG-vip, 2025-04-09:
Hello  @asm5878 
  Any update? Can you reproduce this issue/

LucienKUANG-vip, 2025-04-11:
Hello 
    This is not issue on it. I miss add some configuration on it.
    When I close the UART, the power consumption on STOP2 is 3uA.
    As follow is my test prj.conf
```
CONFIG_BT=y
CONFIG_BT_DEVICE_NAME="Test beacon"
CONFIG_POWEROFF=y
CONFIG_PM=y
CONFIG_ENTROPY_GENERATOR=y

CONFIG_LOG=n
CONFIG_UART_CONSOLE=n
CONFIG_SERIAL=n

CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
CONFIG_PM_DEVICE_SYSTEM_MANAGED=y
```

closed 2025-04-11 by LucienKUANG-vip