zephyrproject-rtos/zephyr #87108: tests: tests/drivers/spi/spi_loopback: mimxrt1170_evk@A/mimxrt1176/cm7: failed: Failed harness:'Testsuite failed'

hakehuang, 2025-03-14:
**Describe the bug**
below case are failed
drivers.spi.loopback.rtio.spi_loopback_rtio.spi_loopback_rtio

test is failed on v4.1.0-rc2-15-g6d5dd34860d4 on mimxrt1170_evk@A/mimxrt1176/cm7
testcase path is tests/drivers/spi/spi_loopback

git bisect found the issue is introduced by

```
b5ca5b7b776cda673cc04651d27eb251bb1be4f7 is the first bad commit
commit b5ca5b7b776cda673cc04651d27eb251bb1be4f7
Author: Luis Ubieda <luisf@croxel.com>
Date:   Mon Feb 17 14:14:42 2025 -0500

    spi_rtio: fix transactions for default handler

    This patch fixes transaction op items not performed within a single
    SPI transfer. This is common for Write + Read commands, that depend on
    the CS kept asserted until the end, otherwise the context will be lost.
    A similar fix was applied to i2c_rtio_default on #79890.

    Signed-off-by: Luis Ubieda <luisf@croxel.com>

 drivers/spi/Kconfig    |  18 ++++++++
 drivers/spi/spi_rtio.c | 115 ++++++++++++++++++++++++++++++++-----------------
 2 files changed, 94 insertions(+), 39 deletions(-)
```

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mimxrt1170_evk@A/mimxrt1176/cm7  -T tests/drivers/spi/spi_loopback -s tests/drivers/spi/spi_loopback
```
or
```
# west twister -p mimxrt1170_evk@A/mimxrt1176/cm7 -T tests/drivers/spi/spi_loopback --sub-test tests/drivers/spi/spi_loopback
```
2. See error

```
Failed harness:'Testsuite failed'
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_spi_loopback_rtio
I: SPI test on buffers TX/RX 0x80004160/0x80004140
I: SPI test slow config
I: Start complete multiple
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
I: Passed
I: Start every 4
E: At most CONFIG_SPI_RTIO_FALLBACK_MSGS submissions in a transaction are allowed in the default handler
E: Buffer contents are different: 0x34,0x35,0x36,0x37,
E:                            vs: 0x00,0x00,0x00,0x00,
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi_rtio.c:367: spi_rx_every_4: (1 is true)
Buffer contents are different
FAIL - test_spi_loopback_rtio in 0.056 seconds

    
```

```
 SPI test fast config
I: Start complete multiple
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
I: Passed
I: Start every 4
I: Passed
I: Start rx bigger than tx
I: Passed
I: Start complete large transfers
I: Passed
I: Start complete loop
I: Passed
I: Start complete loop
I: Passed
I: All tx/rx passed
 PASS - test_spi_loopback in 0.207 seconds
===================================================================
TESTSUITE spi_loopback succeeded
Running TESTSUITE spi_loopback_rtio
===================================================================
START - test_spi_loopback_rtio
I: SPI test on buffers TX/RX 0x80004160/0x80004140
I: SPI test slow config
I: Start complete multiple
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
I: Passed
I: Start every 4
E: At most CONFIG_SPI_RTIO_FALLBACK_MSGS submissions in a transaction are allowed in the default handler
E: Buffer contents are different: 0x34,0x35,0x36,0x37,
E:                            vs: 0x00,0x00,0x00,0x00,

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi_rtio.c:367: spi_rx_every_4: (1 is true)
Buffer contents are different
 FAIL - test_spi_loopback_rtio in 0.056 seconds
===================================================================
TESTSUITE spi_loopback_rtio failed.

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [spi_loopback]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.207 seconds
 - PASS - [spi_loopback.test_spi_loopback] duration = 0.207 seconds

SUITE FAIL -   0.00% [spi_loopback_rtio]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.056 seconds
 - FAIL - [spi_loopback_rtio.test_spi_loopback_rtio] duration = 0.056 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 38098632b505991d09c69a94adc6426f
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.1.0-rc2-15-g6d5dd34860d4

bperseghetti, 2025-03-14:
Hey @ubieda can you take a look.

ubieda, 2025-04-07:
I submitted a fix for this @hakehuang please take a look and confirm whether it passes on your end. Thanks!

PR #88271 (2025-04-09): tests: spi_loopback: fix RTIO fallback messages config to 5
Loopback test requires this in order to properly execute half transactions. Otherwise, testsuite fails.

It was overlooked when fixing #85894.

Fixes #87108

Tested on:
- `nrf52840dk/nrf52840`
- `mimxrt1170@B/mimxrt1176/cm7`
PR #88386 (2025-06-03): [Backport v4.1-branch] tests: spi_loopback: fix RTIO fallback messages config to 5
Backport 6feb1c639b48dc4736b11eedad27ee1604521fe0 from #88271.

Fixes #87108
closed 2025-04-09 by kartben