zephyrproject-rtos/zephyr #87161: PTP NXP K64 not working correctly

theadib, 2025-03-15:
**Describe the bug**
I use a FRDM_KL64F board and the ptp sample application that I slightly updated.
The code should be CPU agnostic and runs fine on NUCLEO_H563.
The issue is that there will be no PTP sync. I get :
`<wrn> ptp_clock: Clock offset exceeds 1 second.` 
I also can observe very strange values in PTP time: Current date and time: 2577-03-30 19:35:40 UTC
as of today 2025-03-15 21:24:00 UTC
Those times vary from query to query.

I checked the PTP configuration, which is correct.
The PTP timer runs with the defined clock and accumulates as expected.

I inspected the ptp_clock_delay() function from zephyr/subsys/net/lib/ptp/clock.c
```
void ptp_clock_delay(uint64_t egress, uint64_t ingress)
{
	int64_t delay;

	ptp_clk.timestamp.t3 = egress;
	ptp_clk.timestamp.t4 = ingress;

	delay = ((int64_t)(ptp_clk.timestamp.t2 - ptp_clk.timestamp.t3) +
		 (int64_t)(ptp_clk.timestamp.t4 - ptp_clk.timestamp.t1)) /
		2LL;

	LOG_DBG("Delay %lldns %lldsec", delay, delay/1000000000LL);
	LOG_DBG("t1:%llu t2:%12llu diff:%lld", ptp_clk.timestamp.t1, ptp_clk.timestamp.t2, (int64_t)(ptp_clk.timestamp.t1 - ptp_clk.timestamp.t2));
	LOG_DBG("t4:%llu t3:%12llu diff:%lld", ptp_clk.timestamp.t4, ptp_clk.timestamp.t3, (int64_t)(ptp_clk.timestamp.t4 - ptp_clk.timestamp.t3));
	ptp_clk.current_ds.mean_delay = clock_ns_to_timeinterval(delay);
}
```

and I get the results (shortened):
```
[00:00:04.868,000] <dbg> ptp_clock: ptp_clock_delay: Delay -358514724ns 0sec
[00:00:04.868,000] <dbg> ptp_clock: ptp_clock_delay: t1:1742072877109508472 t2:  3294967295 diff:1742072873814541177
[00:00:04.868,000] <dbg> ptp_clock: ptp_clock_delay: t4:1742072877964511729 t3:  4867000000 diff:1742072873097511729

[00:00:06.868,000] <dbg> ptp_clock: ptp_clock_delay: Delay -1358638241ns -1sec
[00:00:06.868,000] <dbg> ptp_clock: ptp_clock_delay: t1:1742072879109731245 t2:  3294967295 diff:1742072875814763950
[00:00:06.868,000] <dbg> ptp_clock: ptp_clock_delay: t4:1742072879964487468 t3:  6867000000 diff:1742072873097487468

[00:00:08.868,000] <dbg> ptp_clock: ptp_clock_delay: Delay -2358776525ns -2sec
[00:00:08.868,000] <dbg> ptp_clock: ptp_clock_delay: t1:1742072881109978747 t2:  3294967295 diff:1742072877815011452
[00:00:08.868,000] <dbg> ptp_clock: ptp_clock_delay: t4:1742072881964458401 t3:  8867000000 diff:1742072873097458401
```

For me it seems that T2 is not updated.
I looked for the position where that should happen but I am lost for now.

**To Reproduce**
I use an small hardware PTP generator ELPROMA NTS-Pico, that works with the NUCLEO_H563

**Expected behavior**
T2 timestamp should be updated every sync cycle as T1, T3, T4 change

**Impact**
No PTP usable on K64 platform

**Logs and console output**
this is my modified sample main()
```
struct sntp_ctx ctx;
struct sockaddr_in addr;
struct sntp_time sntp_time;

int main(void){
	init_testing();

	//printk("before sntp prepare");

	addr.sin_family = AF_INET;
	addr.sin_port = htons(123);
	inet_pton(AF_INET, "10.2.33.8", &addr.sin_addr);

	k_sleep(K_SECONDS(5));
	printk("2 before sntp init");
	memset(&ctx, 0, sizeof(ctx));
	sntp_init(&ctx, (struct sockaddr *) &addr,
		       sizeof(struct sockaddr_in));
			   k_sleep(K_SECONDS(5));

	uint64_t last_second = 0;
	while(1) {

		get_current_status();

		const struct device *clk = net_eth_get_ptp_clock(net_if_get_by_index(1));
		struct net_ptp_time tm;
		ptp_clock_get(clk, &tm);
		
		printk("### Current PTP time: %llu.%09u,   diff: %llu\n", tm.second, tm.nanosecond, tm.second - last_second);
		last_second = tm.second;
		struct tm *ptm;
		time_t tai_time = tm.second - 37; // Convert TAI to UTC by subtracting leap seconds
		ptm = gmtime(&tai_time);
		printk("### Current date and time: %04d-%02d-%02d %02d:%02d:%02d UTC\n",
			   ptm->tm_year + 1900, ptm->tm_mon + 1, ptm->tm_mday,
			   ptm->tm_hour, ptm->tm_min, ptm->tm_sec);

 	   printk("3 before sntp query");
		if(sntp_query(&ctx, 4 * MSEC_PER_SEC, &sntp_time) >= 0) {
			time_t now = sntp_time.seconds;
			struct tm now_tm;
			gmtime_r(&now, &now_tm);
			printk("### NTP Current date and time: %04d-%02d-%02d %02d:%02d:%02d UTC\n",
				now_tm.tm_year + 1900, now_tm.tm_mon + 1, now_tm.tm_mday,
				now_tm.tm_hour, now_tm.tm_min, now_tm.tm_sec);
		} else {
			printk("### error on SNTP");
		}

		k_sleep(K_SECONDS(1));
	}

	return 0;
}
```

more complete trace log:
```
00:00:11.584,000] <wrn> ptp_clock: Clock offset exceeds 1 second.
[00:00:11.855,000] <dbg> ptp_port: port_delay_req_msg_transmit: Port 1 sends Delay_Req message
[00:00:11.856,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Delay_Resp message
[00:00:11.856,000] <dbg> ptp_clock: ptp_clock_delay: Delay -4144090221ns -4sec
[00:00:11.856,000] <dbg> ptp_clock: ptp_clock_delay: t1:1742073912228773239 t2:  3294967295 diff:1742073908933805944
[00:00:11.856,000] <dbg> ptp_clock: ptp_clock_delay: t4:1742073912500625501 t3: 11855000000 diff:1742073900645625501
[00:00:12.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:00:12.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:00:12.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:00:12.584,000] <dbg> ptp_clock: ptp_clock_synchronize: t1:1742073913228847026 t2:  3294967295
[00:00:12.584,000] <wrn> ptp_clock: Clock offset exceeds 1 second.
TIME RECEIVER
### Current PTP time: 12194517344.143210174,   diff: 12194517344
### Current date and time: 2356-06-06 05:55:07 UTC
3 before sntp query### NTP Current date and time: 2025-03-15 21:24:36 UTC
[00:00:13.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:00:13.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:00:13.585,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:00:13.585,000] <dbg> ptp_clock: ptp_clock_synchronize: t1:1742073914229010418 t2:  3294967295
[00:00:13.585,000] <wrn> ptp_clock: Clock offset exceeds 1 second.
[00:00:13.855,000] <dbg> ptp_port: port_delay_req_msg_transmit: Port 1 sends Delay_Req message
[00:00:13.856,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Delay_Resp message
[00:00:13.856,000] <dbg> ptp_clock: ptp_clock_delay: Delay -5144222700ns -5sec
[00:00:13.856,000] <dbg> ptp_clock: ptp_clock_delay: t1:1742073914229010418 t2:  3294967295 diff:1742073910934043123
[00:00:13.856,000] <dbg> ptp_clock: ptp_clock_delay: t4:1742073914500597723 t3: 13855000000 diff:1742073900645597723
TIME RECEIVER
### Current PTP time: 13936591251.934761636,   diff: 1742073907
### Current date and time: 2411-08-20 03:20:14 UTC
3 before sntp query### NTP Current date and time: 2025-03-15 21:24:37 UTC
[00:00:14.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:00:14.585,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:00:14.585,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:00:14.585,000] <dbg> ptp_clock: ptp_clock_synchronize: t1:1742073915229109896 t2:  3294967295
[00:00:14.585,000] <wrn> ptp_clock: Clock offset exceeds 1 second.
TIME RECEIVER
### Current PTP time: 15678665159.725784957,   diff: 1742073908
### Current date and time: 2466-11-02 00:45:22 UTC
3 before sntp query### NTP Current date and time: 2025-03-15 21:24:38 UTC
[00:00:15.584,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Announce message
[00:00:15.585,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Sync message
[00:00:15.585,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Follow_Up message
[00:00:15.585,000] <dbg> ptp_clock: ptp_clock_synchronize: t1:1742073916229210418 t2:  3294967295
[00:00:15.585,000] <wrn> ptp_clock: Clock offset exceeds 1 second.
[00:00:15.855,000] <dbg> ptp_port: port_delay_req_msg_transmit: Port 1 sends Delay_Req message
[00:00:15.856,000] <dbg> ptp_msg: ptp_msg_post_recv: Port 1 received Delay_Resp message
[00:00:15.856,000] <dbg> ptp_clock: ptp_clock_delay: Delay -6144335939ns -6sec
[00:00:15.856,000] <dbg> ptp_clock: ptp_clock_delay: t1:1742073916229210418 t2:  3294967295 diff:1742073912934243123
[00:00:15.856,000] <dbg> ptp_clock: ptp_clock_delay: t4:1742073916500571245 t3: 15855000000 diff:1742073900645571245
TIME RECEIVER
```

**Environment (please complete the following information):**

 - OS: Linux, 
 - Toolchain : Zephyr 4.1.0 main, SDK 0.17
 - Commit SHA or Version used

**Additional context**

kartben, 2025-03-16:
Yeah, this is a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/75718
cc @decsny

theadib, 2025-04-06:
I am investigated a bit, but it is hard to produce good traces ...

So the thing is that for Tx packet there is no hardware timestamp created. The callback to collect the timestamp is not executed.

I patched that code:
in function zephyr/drivers/ethernet/eth_nxp_enet.c:eth_nxp_enet_tx() 

there is this test:
`	frame_is_timestamped = eth_get_ptp_data(net_pkt_iface(pkt), pkt);
`
but as eth_get_ptp_data() only checks for a PTP VLAN which is gPTP it returns false for 1588
so I made:
`	frame_is_timestamped = eth_get_ptp_data(net_pkt_iface(pkt), pkt) || net_pkt_is_tx_timestamping(pkt);`
This test is now similar as in STM32_h5 code.

But I also have to patch corresponding timestamp consumer:
in function zephyr/drivers/ethernet/eth_nxp_enet.c:ts_register_tx_event() 
`		if(eth_get_ptp_data(net_pkt_iface(pkt), pkt) && frameinfo->isTsAvail) {`
becomes 
`		if((eth_get_ptp_data(net_pkt_iface(pkt), pkt)|| net_pkt_is_tx_timestamping(pkt)) && frameinfo->isTsAvail) {
`

now the callback is executed. 
Somehow the timestamp captured (seconds.nanoseconds) is below 1

@decsny  Do you have any idea how to continue?

Regards, Adib.

theadib, 2025-04-06:
Hi went further and solved the issue with the missing seconds information.
This is maybe a chip implementation.

please see my PR https://github.com/zephyrproject-rtos/zephyr/pull/88214

Regards, Adib.

theadib, 2025-04-06:
I do not know why it closed it auto magically :-O