zephyrproject-rtos/zephyr #87162: Bluetooth sample documentation is outdated and lacking

nashif, 2025-03-15:
**Describe the bug**

Most Bluetooth samples have an old boilerplate README from ~10 years ago, no details how to build and how to use them and a self-reference to the same page in github under 'Building and Running`

[image]

**To Reproduce**

https://docs.zephyrproject.org/latest/samples/bluetooth/central/README.html#ble_central

**Expected behavior**

More details and following our sample guidelines and expectations from samples.

**Impact**

Sample code that is difficult to use.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

jhedberg, 2025-03-18:
@KyraLengfeld is this something you could start looking at, considering that you've been focusing on improving Bluetooth documentation recently?

KyraLengfeld, 2025-03-19:
> [@KyraLengfeld](https://github.com/KyraLengfeld) is this something you could start looking at, considering that you've been focusing on improving Bluetooth documentation recently?

@jhedberg, sample documentation isn't in our plan currently. We will discuss it within our team, but personally I think it is unlikely we will prioritize sample documentation in the near future.