zephyrproject-rtos/zephyr #87200: Use of ld specic function (ADDR) break non-ld linkers

tejlmand, 2025-03-17:
**Describe the bug**

Non-ld linkers was broken with PR#https://github.com/zephyrproject-rtos/zephyr/pull/80768

`zephyr_linker_symbol()` is only intended with simple math operations, as described by the function:
https://github.com/zephyrproject-rtos/zephyr/blob/d6c59f0b4d1553d01297e6fcb7ae1d865a897732/cmake/modules/extensions.cmake#L5366-L5370

The changes introduced in #80678 breaks this rule by introducing `ADDR()` which is a ld specific function not understood by non-ld linkers, such as armlink.

**To Reproduce**
1. Setup Arm Conpiler 6, as described: https://docs.zephyrproject.org/latest/develop/toolchains/arm_compiler.html
2. Build a simple sample, such as hello world:
    `west build -p -b qemu_cortex_m3 samples/hello_world/ -- -DCONFIG_CMAKE_LINKER_GENERATOR=y -DCONFIG_ARMCLANG_STD_LIBC=y
3. See error:
    ```
    Error: L6630E: Invalid token start expected number or ( but found A at position 26 on line 498
    Error: L6226E: Missing base address for region __ramfunc_region_start.
    "/projects/github/ncs/zephyr/build/zephyr/linker_zephyr_pre0.cmd", line 498 (column 26): Error: L6292E: Ignoring unknown attribute 'ADDR' specified for region __ramfunc_region_start.
    Error: L6630E: Invalid token start expected number or ( but found . at position 31 on line 498
    Error: L6629E: Unmatched parentheses expecting ) but found . at position 31 on line 498
    "/projects/github/ncs/zephyr/build/zephyr/linker_zephyr_pre0.cmd", line 498 (column 31): Error: L6228E: Expected '{', found '....'.
    "/projects/github/ncs/zephyr/build/zephyr/linker_zephyr_pre0.cmd", line 498 (column 31): Error: L6228E: Expected '}', found 'EOF'.
    ```

**Expected behavior**
Successful linking:
```
[2/109] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.0-rc2 (/projects/github/ncs/zephyr), build: v4.0.0-rc2-84-g6984237c0638
[109/109] Linking C executable zephyr/zephyr.elf
Generating files from /projects/github/ncs/zephyr/build/zephyr/zephyr.elf for board: qemu_cortex_m3
```

**Impact**
Support for Arm Compiler 6 is broken.

**Logs and console output**
N/A

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Arm compiler 6
 - Commit SHA or Version used: 6023d6a1428d880a988f1be3ab1b5da5a6e86d61 or never

**Additional context**
N/A

tejlmand, 2025-03-17:
@LoveKarlsson FYI, might impact IAR as well.

LoveKarlsson, 2025-03-18:
> [@LoveKarlsson](https://github.com/LoveKarlsson) FYI, might impact IAR as well.

We have added support for ADDR and LOADADDR to our linker

commit b84076a1 (2025-04-24): cmake: arm: linker: remove usage of ld-specific ADDR() function

Remove usage of ld-specific ADDR function within the cmake linker
generator scripting. Since the linker generator scripting doesn't
support MPUs, we can simply set the __ramfunc_region_start symbol to be
equal to __ramfunc_start

Fixes #87200
PR #88599 (2025-04-24): cmake: arm: linker: remove usage of ld-specific ADDR() function
Remove usage of ld-specific ADDR function within the cmake linker generator scripting. Since the linker generator scripting doesn't support MPUs, we can simply set the __ramfunc_region_start symbol to be equal to __ramfunc_start

Fixes #87200
commit 6239b574 (2025-07-20): cmake: arm: linker: remove usage of ld-specific ADDR() function

Remove usage of ld-specific ADDR function within the cmake linker
generator scripting. Since the linker generator scripting doesn't
support MPUs, we can simply set the __ramfunc_region_start symbol to be
equal to __ramfunc_start

Fixes #87200

(cherry picked from commit b84076a192717894b09d6f4b643c727aa754673c)
closed 2025-04-24 by kartben