zephyrproject-rtos/zephyr #87208: The scan delegator receive_state_updated function should give back the semaphore protecting the read_buf buffer as soon as the receive_state_updated function is done accessing the buffer

jilu-ot, 2025-03-17:
**Is your enhancement proposal related to a problem? Please describe.**
Yes.

The problem is that the receive state updated callbacks handlers might change thread by submitting a work item that should be executed immediately after, but is not because some other thread managed to add a work item first which also uses the scan delegator in a way that updated the receive state. This would cause the second update of the receive state to not be sent out since the first is waiting to be executed and thus have taken the semaphore.

Callback handlers would change the thread from the calling thread to the application's own thread in order to protect application data.

**Describe the solution you'd like**

One solution to this problem could be to only protect the read_buf buffer in the scan delegator for as long as it is being accessed, but not the call to callback handlers.

Update the receive_state_updated function as follows:

```c
static void receive_state_updated(struct bt_conn *conn,
				  const struct bass_recv_state_internal *internal_state)
{
	int err;

	/* If something is holding the NOTIFY_PEND flag we should not notify now */
	if (atomic_test_bit(internal_state->flags,
			    BASS_RECV_STATE_INTERNAL_FLAG_NOTIFY_PEND)) {
		return;
	}

	bt_debug_dump_recv_state(internal_state);

	err = k_sem_take(&read_buf_sem, SCAN_DELEGATOR_BUF_SEM_TIMEOUT);
	if (err != 0) {
		LOG_DBG("Failed to take read_buf_sem: %d", err);

		return;
	}

	net_buf_put_recv_state(internal_state);
	bass_notify_receive_state(conn, internal_state);

	k_sem_give(&read_buf_sem);

	if (scan_delegator_cbs != NULL &&
	    scan_delegator_cbs->recv_state_updated != NULL) {
		scan_delegator_cbs->recv_state_updated(conn,
						       &internal_state->state);
	}
}

This will release the read_buf semaphore before calling the application's recv_state_updated callback handler.

```

**Describe alternatives you've considered**
Or the receive_state_updated function could use a type of semaphore that does not block if it is taken from the same thread is already taken by. See C++ stdlib std::recursive_mutex

**Additional context**
None.

Thalley, 2025-03-17:
Sounds like a bug, so changed the type. Will investigate :) 

Do you have any reliable way of reproducing this issue?

jilu-ot, 2025-03-17:
> Sounds like a bug, so changed the type. Will investigate :)
> 
> Do you have any reliable way of reproducing this issue?

Not any easy way unfortunately.

Thalley, 2025-03-17:
> > Sounds like a bug, so changed the type. Will investigate :)
> > Do you have any reliable way of reproducing this issue?
> 
> Not any easy way unfortunately.

I've sent a PR with your suggested fix :) If that fixes this issue, then it's OK (and even if it doesn't fix the issue, it's a good idea to minimize the semaphore usage

closed 2025-04-11 by kartben