zephyrproject-rtos/zephyr #87228: Timezone offset when net_init_clock_via_sntp()

pedrohqr, 2025-03-17:
**Is your enhancement proposal related to a problem? Please describe.**
When the device get the current date time through SNTP server, it gets the UTC+0 offset time. The purpouse is add some feature that allow user calls for a function that sets a global timezone for the device on use `CONFIG_NET_CONFIG_CLOCK_SNTP_INIT=y`.

**Describe the solution you'd like**
Create a new header wich contains enumerated timezones and a variable in `init_clock_sntp.c` that controls the timezone offset when system clock will be update.

**Describe alternatives you've considered**

**Additional context**

pedrohqr, 2025-03-19:
Hello. I have added this feature. I've created a new branch follow the [contribuition guidelines](https://docs.zephyrproject.org/latest/contribute/guidelines.html) and I would like to commit into the repository.

rlubos, 2025-03-20:
Feel free to open a pull request then: https://docs.zephyrproject.org/latest/contribute/guidelines.html#pull-request-guidelines

closed 2025-05-19 by pedrohqr