zephyrproject-rtos/zephyr #87239: NXP DMA Controller Uses Non-Secure Controller

JoeyFreeland-DojoFive, 2025-03-17:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

henrikbrixandersen, 2025-03-17:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

JoeyFreeland-DojoFive, 2025-03-18:
Update on this issue: It does work on flexcomm14, but not on flexcomm0 or flexcomm1. I've tracked it down to the clocks not being configured on flexcomm0. Still trying to figure out why it only works on 14.

JoeyFreeland-DojoFive, 2025-03-18:
@dleach02 @mmahadevan108 I made a change to `zephyr/soc/nxp/rw` yesterday, in the function `clock_init()`. Right after the I2C clock init section (starts with comment `/* Any flexcomm can be I2C */`) I added this:

```
/* Any flexcomm can be I2S */
#if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm0), nxp_lpc_i2s, okay)) && CONFIG_I2S
	CLOCK_SetFRGClock(&(const clock_frg_clk_config_t){0, kCLOCK_FrgPllDiv, 255, 0});
	CLOCK_AttachClk(kFRG_to_FLEXCOMM0);
#endif
#if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm1), nxp_lpc_i2s, okay)) && CONFIG_I2S
	CLOCK_SetFRGClock(&(const clock_frg_clk_config_t){1, kCLOCK_FrgPllDiv, 255, 0});
	CLOCK_AttachClk(kFRG_to_FLEXCOMM1);
#endif
#if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm2), nxp_lpc_i2s, okay)) && CONFIG_I2S
	CLOCK_SetFRGClock(&(const clock_frg_clk_config_t){2, kCLOCK_FrgPllDiv, 255, 0});
	CLOCK_AttachClk(kFRG_to_FLEXCOMM2);
#endif
#if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm3), nxp_lpc_i2s, okay)) && CONFIG_I2S
	CLOCK_SetFRGClock(&(const clock_frg_clk_config_t){3, kCLOCK_FrgPllDiv, 255, 0});
	CLOCK_AttachClk(kFRG_to_FLEXCOMM3);
#endif
#if (DT_NODE_HAS_COMPAT_STATUS(DT_NODELABEL(flexcomm14), nxp_lpc_i2s, okay)) && CONFIG_I2S
	CLOCK_SetFRGClock(&(const clock_frg_clk_config_t){14, kCLOCK_FrgPllDiv, 255, 0});
	CLOCK_AttachClk(kFRG_to_FLEXCOMM14);
#endif
```

After doing this, flexcomm0 was able to output the I2S data. Is it possible that I2S clock init was missed in the original `soc.c` implementation?

DerekSnell, 2025-03-19:
Hi @JoeyFreeland-DojoFive ,
Well done finding the solution.  As you found, these clocks are not configured in soc.c.  Can you submit a Pull Request to add this code?

Thanks for reporting this

closed 2025-04-03 by kartben