zephyrproject-rtos/zephyr #87240: kernel: edf: Change `k_thread_deadline_set` from relative deadline to absolute

alexpaschoaletto, 2025-03-17:
**Describe the solution you'd like**

I'd like to propose that we either change the existing `k_thread_deadline_set` function to accept an absolute deadline instead of relative, or extend the API to feature a `k_thread_deadline_set_absolute` for this effect. The second option might be better as it doesn't break anything and offers more flexibility for the application developer.

In other words, what I'm basically asking is that we change this:
```c
int32_t newdl = k_cycle_get_32() + deadline;
```
to this:
```c
int32_t newdl = deadline;
```
In `z_impl_k_thread_deadline_set`. Or that we simply create the `absolute` counterpart.

***Major benefits***

Allowing the absolute deadline to be passed directly means we can:
 1. Turn the deadline unit-agnostic. Currently, `k_thread_deadline_set` is meant to deal with a deadline in cycle units, whereas most of the kernel is tailored to work with system ticks using the `k_timeout_t` opaque data type. Taht is somewhat inconsistent. By making the change above, the deadline can be passed in whatever unit convention makes more sense to the application.
 2. Easily implement more scheduling algorithms in the kernel. If we look in practical terms, the kernel doesn't seem to make any tight reservations for the meaning of the deadline. This means that if we pass a task's period as the deadline, then suddenly we have a Rate Monotonic scheduler right away. If we pass the task's laxity, we can easily make our way to implement LLF. and so on.

The idea was born out of my need of adding support to Rate Monotonic and other schedulers in Zephyr. Although I've inspected the codebase, so far I haven't figured out how the declaration of `k_thread_deadline_set` programatically relates to `z_impl_k_thread_deadline_set`, which stops me from directly implementing the solution above and posting a PR. If anyone can help me with this information, I'd be glad to do it.

alexpaschoaletto, 2025-03-18:
> > Although I've inspected the codebase, so far I haven't figured out how the declaration of k_thread_deadline_set programatically relates to z_impl_k_thread_deadline_set, which stops me from directly implementing the solution above and posting a PR. If anyone can help me with this information, I'd be glad to do it.
> 
> See https://docs.zephyrproject.org/latest/kernel/usermode/syscalls.html#implementation-function

Thank you! I've applied the proposed changes in PR #87314.

TaiJuWu, 2025-10-22:
fix by https://github.com/zephyrproject-rtos/zephyr/pull/94731

closed 2025-10-22 by TaiJuWu