zephyrproject-rtos/zephyr #87260: Support `uv` as an alternative manager to `pip` for package management

cdwilson, 2025-03-18:
`uv` (https://github.com/astral-sh/uv) is becoming a popular tool for creating and managing Python virtual environments (mainly due to the fact that it's really fast).

It's currently possible to use `uv pip` to install Python dependencies for Zephyr, e.g.:

```sh
west packages pip | xargs uv pip install
```

It would be nice if `uv pip` could be added as a supported "manager" for the `west packages` command so that `uv` would be used under the hood to install packages.

JPHutchins, 2025-08-14:
The real benefit to this is not uv, per se, but having any dependency resolver, period. I think that an appropriate path would be to migrate Zephyr's pip freezes to PEP 751 pylock.toml. Then, the user can choose the dependency resolver of their choice to mix their project dependencies with Zephyr's tooling dependencies, now with locks and conflict resolution.

marc-hb, 2025-08-14:
@JPHutchins please feel free to share any "brain dump" in a new, separate issue. Even if you don't have time to make much progress, this can always be useful to someone else later!

closed 2026-09-26 by github-actions[bot]