zephyrproject-rtos/zephyr #87270: CI compilation issue on board arduino_portenta_h7@1.0.0_stm32h747xx_m7

kbidani, 2025-03-18:
**Describe the bug**

There is a compilation issue occurring in the CI pipeline for the board arduino_portenta_h7@1.0.0_stm32h747xx_m7.
This issue is blocking the progress of my PR and needs to be addressed to ensure the CI pipeline runs smoothly and the PR can be merged.
PR link: https://github.com/zephyrproject-rtos/zephyr/pull/86100
**Logs and console output**

[image]

ifyall, 2025-03-18:
@kbidani, for manually fixing this, before building you need to fetch the firmware blobs for the CY4343W. You do that with this command:

`west blobs fetch hal_infineon`

I am not certain how you make sure the CI does that as a part of its operation. I will do a bit of research to see if I can understand that.

erwango, 2025-03-18:
Thanks @ifyall for the comment. IIC, fetching blobs is not possible in CI.
Hence all boards featuring this module should be prevented to be built in CI BLE tests (`arduino_giga_r1`, `arduino_nicla_vision`, `arduino_portenta_h7`, `cy8cproto_062_4343w`)
Then, the question is: How come this error is seen now only on this specific target ?

Looking at other bluetooth tests, it seems they all restrict targets using:
```
    platform_allow:
      - native_sim
      - native_sim/native/64
```
which seems to be missing on this specific test.

kartben, 2025-03-18:
@kbidani  can you try rebasing your PR? This should be fixed with #87265

kbidani, 2025-03-27:
It is fixed and can be closed.

closed 2025-03-27 by kartben