zephyrproject-rtos/zephyr #87287: Bluetooth: Host: More specific printing of addresses

koffes, 2025-03-18:
**Is your enhancement proposal related to a problem? Please describe.**

No problem, but when calling https://github.com/zephyrproject-rtos/zephyr/blob/3d2059f9b66faf2b6cf4be15b2f4adc2f092f42b/include/zephyr/bluetooth/addr.h#L242-L243
it may only print "random".

**Describe the solution you'd like**

The above is fine, and correct, but it would be nice if this could also account for the address being random static or random private.  It may be as simple as looking at if CONFIG_BT_PRIVACY is set or not, but that needs to be checked.

koffes, 2025-03-19:
FYI @theob-pro @alwa-nordic.

alwa-nordic, 2025-03-19:
I don't think it's worth breaking API for this.

Note that the random/public distinction is crucial to identify the address. It's the 49th bit of the Bluetooth LE address space. Not including that last bit means the address is not complete. OTOH, static and private addresses are sub-spaces of the 49-bit address.

In my opinion, Bluetooth LE addresses should have been canonically displayed with the 49th bit clearly as part of the address, like "1:AA:BB:CC:DD:EE:FF".

Another problem is that the proposed change would add extra failure modes to `bt_addr_from_str`, that's the inverse of `bt_addr_le_to_str`.