zephyrproject-rtos/zephyr #87293: Debugging on Nucleo H755ZI-Q doesn't work properly

mthiede-acn2, 2025-03-18:
**Describe the bug**

Debugging using `west debug` does not work properly.
For example, single stepping is not possible.
`si` will not run the next instruction.
`n` and `s` also don't work as well as a `c` after hitting a breakpoint.

I updated the firmware of the STLink to the latest version.
I tried halting the M4 core using the BOOT_CM4 option without success.
I get the same behavior with Zephyr v4.1.0

**To Reproduce**
1. west build -b nucleo_h755zi_q/stm32h755xx/m7 -p always samples/basic/blinky
2. west debug
3. b main
4. c
5. n

**Expected behavior**
Debugger single steps to next instruction

**Impact**
Debugging hardly possible. It is possible to set and break on a first breakpoint though (like on 'main' above)

**Environment (please complete the following information):**

 - OS: WSL/Ubuntu
 - Toolchain: Zephyr SDK 0.17.0
 - Zephyr: v4.0.0 (release tag), same with v4.1.0
 - Open On-Chip Debugger 0.11.0+dev-00728-gb6f95a16c-dirty (from SDK 0.17.0)
 - GNU gdb (Zephyr SDK 0.17.0) 12.1

erwango, 2025-03-18:
I don't have this board, so I can't test, but I had a quick test on stm32h747i_disco/stm32h747xx/m7 and debug works reliabily.
Though, openocd config file for Nucleo H755ZI-Q isn't aligned on stm32h747i_disco, so it should be aligned.
Would you be able to import the configuration and have a try ?

About M4 debug, IIRC there were some attempts to have it functional with gdb, but from what I could test it doesn't work. The related note in the board documentation still stands: 
```
Debugging a Zephyr application on Cortex M4 side with west is currently not
available. As a workaround, `STM32CubeIDE`_ can be used.
```

lromeraj, 2025-03-19:
Did you build the application without size optimizations (default mode)? Try enabling `CONFIG_DEBUG=y` and ensuring `CONFIG_DEBUG_OPTIMIZATIONS=y`, then reflash the firmware and check if it improves the debugging process.

mthiede-acn2, 2025-03-19:
@erwango I modified `zephyr/boards/st/nucleo_h755zi_q/support/openocd.cfg` to contain line

`source [find board/stm32h745i-disco.cfg]`

instead of

`source [find board/st_nucleo_h745zi.cfg]`

and now single stepping and continue work.

@lromeraj thanks for the hint, this improves debug experience once debugging is working at all

erwango, 2025-03-20:
@mthiede-acn2 Thanks for the feedback. Would you have the possibility to push this change so everyone can benefit ?