zephyrproject-rtos/zephyr #87322: doc: Documentation issue in 'connectivity/networking/api/socket_service'

clamattia, 2025-03-19:
**Describe the bug**
1. I believe, that there is a typo. Should it not be `unregister` here?
```diff
-     /* Update the handler so that client connection is
-      * not monitored any more.
-      */
-      (void)net_socket_service_register(&service_tcp, sockfd_tcp,
-                                        ARRAY_SIZE(sockfd_tcp), NULL);
+     /* Update the handler so that client connection is
+      * not monitored any more.
+      */
+      (void)net_socket_service_unregister(&service_tcp, sockfd_tcp,
+                                        ARRAY_SIZE(sockfd_tcp), NULL);
```
2. It would be kind of nice, to implement a sample without the extra thread for the `accept`-loop (`while(1)...`). In theory it is possible to implement the `accept`-functionality as a callback registered to the socket-service also.

**Environment**

* Page: `connectivity/networking/api/socket_service`
* Version: main
* SHA-1: f13555c93255193f742344bfee3ada180380aa6f

jukkar, 2025-03-19:
No error in the code, we should call register to update listened sockets (so removing the accepted socket). Note that we are still monitoring the listening socket after this. If we call unregister, then the whole service is unregistered which is not the point here.

jukkar, 2025-03-19:
User is suppose to mark non interested sockets to -1, so normal poll behavior.

clamattia, 2025-03-19:
> User is suppose to mark non interested sockets to -1, so normal poll behavior.

```C
struct zsock_pollfd fds1[] = {{.fd=sock1, .event=POLLIN}, {.fd=sock2, .event=POLLIN}};
struct zsock_pollfd fds2[] = {{.fd=sock3, .event=POLLIN}, {.fd=-1}};

net_socket_service_register(&service, fds1, ARRAY_SIZE(fds1), NULL);
net_socket_service_register(&service, fds2, ARRAY_SIZE(fds2), NULL);
```
?

So consequtive calls to `service_register` need to pass the same `len` argument? This seems overly restrictive, the implementation of `net_socket_service_register` could automatically set all `fd`s for index >`len` to `-1`?

clamattia, 2025-03-19:
Alternatively please document, that all calls to `net_socket_service_register` need to pass the same exact `len` argument.

clamattia, 2025-03-20:
Part 1 will be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/87344
Part 2 has a separate issue: https://github.com/zephyrproject-rtos/zephyr/issues/87402
(So this may be closed on merge of Part 1)

clamattia, 2025-03-21:
Part 1 complete https://github.com/zephyrproject-rtos/zephyr/pull/87344#event-16928438068

closed 2025-03-21 by clamattia