zephyrproject-rtos/zephyr #87335: DT_NODE_EXISTS() is true but DEVICE_DT_GET() expands to nothing

5frank, 2025-03-19:
**Describe the bug**

If a device driver `depends on FOO` and `FOO=n`, then `DT_NODE_EXISTS(foo_dev)` is true but `DEVICE_DT_GET(foo_dev)` expands to nothing
(i.e compiler error `undefined reference to __device_dts_ord_NN`)

**To Reproduce**
See attached app. It uses the ws2812-gpio driver for ESP32 but depends on a nordic SOC.

[reproduce_app.zip](https://github.com/user-attachments/files/19338470/reproduce_app.zip)

**Environment (please complete the following information):**

 - OS: Linux/Debian
 - zephyr latest (SHA 20faa0e1142144526475261b550222d2312caf43)

pdgendt, 2025-03-19:
I think this is intended behavior, there's no limitation for `DT_NODE_EXISTS` to "know" about dependencies, it simply verifies the generated device tree. It's up to the user to set the status to `"disabled"` in this case.

The linker error `undefined reference to __device_dts_ord_NN` has been the subject of many discussions and could be improved upon.

pdgendt, 2025-03-19:
It is even stated in the [troubleshooting documentation](https://docs.zephyrproject.org/latest/build/dts/troubleshooting.html#undefined-reference-to-device-dts-ord-n) that Kconfig option need to be enabled.

mbolivar, 2025-03-27:
This is not a bug. In addition to the docs that @pdgendt pointed you to, you can also watch my talk https://www.youtube.com/watch?v=w8GgP3h0M8M for an explanation of why you are seeing this

closed 2025-03-27 by mbolivar