zephyrproject-rtos/zephyr #87351: TI CC1101 Driver

tpitman, 2025-03-19:
## Origin

Zephyr OS driver for TI's cc1101 Sub-Ghz transceiver
https://github.com/studiofuga/cc1101

## Purpose

Zephyr driver for the TI CC1101 RF chip.

## Mode of integration

I would like it integrated into the tree so that it will work with the visual tools that are included in the nRF VSCode extension.

## Maintainership

I assume the current maintainer would continue: studiofuga

## Pull Request

I don't know how to do this

## Description

This driver will facilitate making many different types of products based on the CC1101 RF radio chip. One example is TPMS integration.

There may be other drivers available, but this is the only one I could quickly find that at least had some nRF support.

## Security

As far as I know there would be no need for encryption or other security functionality as that would generally be handled at a higher layer.

## Dependencies

Not that I know of.

## Revision

This driver doesn't seem to be versioned.

Version or SHA you would like to integrate initially

## License

Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``

fabiobaltieri, 2025-04-23:
Hi @tpitman, this sounds to me like a feature request, that's not quite how Zephyr contributions work, someone has to do the work to make a proposal/pull request, the project will help reviewing and approving it. You probably want to get in touch with the author of the external module @studiofuga and ask them if they want to do the integration work, or do it yourself.

tpitman, 2025-04-23:
Ah, didn't understand that. Thanks!

fabiobaltieri, 2025-04-23:
Cool, closing down for now, if you or the author wants to start the process it can probably just be a pull request anyway.

closed 2025-04-23 by fabiobaltieri

studiofuga, 2025-04-23:
Hi, I would be glad to have this driver merged, but frankly, I am not sure if the code has the required quality to be merged. It was mostly an exercise to learn zephyr, and it was part of a side project. Actually, I have not the opportunity to maintain and test it properly. 
To @tpitman, I suggest having a look at my post [here](https://studiofuga.com/blog/2023-10-20-zephyr-workspaces/), where I explain how I come to have it as a driver. I think you can follow the same approach for your case.
Hope that helps.

tpitman, 2025-04-23:
@studiofuga Thank you for weighing in on this. I will have to consider whether I have the expertise to do it myself or not. I am pretty new to both Zephyr and the CC1101.

studiofuga, 2025-04-23:
You probably don't need all this work, likely it is enough to add this driver as an external module. Try following my example in my blog post. Probably "The “Project-centric” star topology" is what you need.