zephyrproject-rtos/zephyr #87363: Font license issue in LVGL repository

jfischer-no, 2025-03-19:
**Describe the bug**

There is at least one problematic font, SimSun, which is included in the LVGL and distributed with Zephyr. The license of this font is unknown, and it does not appear that it can be used and redistributed.

See:
src/font/lv_font_simsun_14_cjk.c
demos/ebike/generated/font_ebike_inter_14.c

In general, the project is not clear about what fonts are used and what their licenses are. The `scripts/built_in_font/font_license/` does not seem to be complete. A step-by-step review of the included fonts is probably needed.

See also https://github.com/lvgl/lvgl/issues/6711

**To Reproduce**

`cd modules/lib/gui/lvgl/; git grep SimSun`
`git grep SIMSUN docs/details/main-components/font.rst`
`git grep SIMSUN`

**Impact**
Bad mood in your legal department.

faxe1008, 2025-03-21:
Opened an upstream issue, if there are no objections will work on a patch to replace the font.
Curious: how did you stumble upon this?

jfischer-no, 2025-03-26:
> Opened an upstream issue, if there are no objections will work on a patch to replace the font. Curious: how did you stumble upon this?

There was a discussion about the LVGL repository in ProcessWG. I just grep for the fonts and then do some more web searches for font names. This is not the first time I have found this kind of problem. You can probably take any open source graphics library for µC and will find the font license issues. What really surprised me was how https://github.com/lvgl/lvgl/issues/6711 is handled.

faxe1008, 2025-03-27:
> You can probably take any open source graphics library for µC and will find the font license issues.

Yeah probably true.

> What really surprised me was how https://github.com/lvgl/lvgl/issues/6711 is handled.

Unfortunate how this was handled, true. Went over the fonts and to me it looks like there are at least two other fonts that need to be removed:
https://github.com/lvgl/lvgl/issues/7968#issuecomment-2754873719

faxe1008, 2025-09-09:
Thanks @AndreCostaaa for carrying this one over the finish line. This issue can finally be closed with the next module sync 💯