zephyrproject-rtos/zephyr #87373: C++ Hello World Sample doesn't work with gnuarmemb toolchain and qemu_cortex_m3

nslowell, 2025-03-19:
**Describe the bug**

Trying to run samples/cpp/hello_world on qemu_cortex_m3 (with expanded flash size)
Using the Zephyr SDK works fine, with and without CONFIG_NEWLIB_LIBC=y.
Trying to use gnuarmemb results in no output and apparent hang (with some really weird stuff going on based on gdb output).

**To Reproduce**

1. Fresh zephyr install using: https://docs.zephyrproject.org/latest/develop/getting_started/index.html
2. Install GNU Arm Embedded toolchain , referring to: https://docs.zephyrproject.org/latest/develop/toolchains/gnu_arm_embedded.html
    a. I chose 11.3 Release: x86_64 Linux hosted cross toolchains, AArch32 bare-metal target (arm-none-eabi)
3. export ZEPHYR_TOOLCHAIN_VARIANT to gnuarmemb and export GNUARMEMB_TOOLCHAIN_PATH to path to toolchain (e.g. ~/arm-gnu-toolchain-11.3/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/ )
4. IMPORTANT: open dts/arm/ti/lm3s6965.dtsi and change flash0 reg to 256\*1024\*1024 (just so the image will "fit" in FLASH)
5. west build -p always -b qemu_cortex_m3 samples/cpp/hello_world
6. west build -t run
Output:
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling

**Expected behavior**
Should see "Hello, C++ world!" just like with Zephyr SDK.

**Impact**

Annoyance...potentially a showstopper.

**Logs and console output**

My GDB session shows something really weird going on.  
1. The program gets into z_bss_zero() in kernel/init.c,
2. I step into z_early_memset(__bss_start, 0, __bss_end - __bss_start);
Here's the disassembly

> 0x00003fbc <+0>:     push    {r3, lr}
>    0x00003fbe <+2>:     ldr     r0, [pc, #12]   @ (0x3fcc <z_bss_zero+16>)
>    0x00003fc0 <+4>:     ldr     r2, [pc, #12]   @ (0x3fd0 <z_bss_zero+20>)
>    0x00003fc2 <+6>:     movs    r1, #0
>    0x00003fc4 <+8>:     subs    r2, r2, r0
>    0x00003fc6 <+10>:    bl      0x47b4e <z_early_memset>
>    0x00003fca <+14>:    pop     {r3, pc}
>    0x00003fcc <+16>:    lsrs    r0, r6, #31
>    0x00003fce <+18>:    movs    r0, #0
>    0x00003fd0 <+20>:    cmp     r5, #76 @ 0x4c
>    0x00003fd2 <+22>:    movs    r0, #0

3. Program Counter will branch to 0x47b4e <z_early_memset>
Here's some more disassembly

> 00047b4e <z_early_memset>:
>         (void) memset(dst, c, n);
>    47b4e:       f7f4 b9ff       b.w     3bf50 <memset>
> 
> 00047b52 <z_early_memcpy>:
>         (void) memcpy(dst, src, n);
>    47b52:       f7b8 bbab       b.w     2ac <memcpy>

This is very similar to assembly to when the Zephyr SDK is used.
5. With the Program Counter at 0x47b4e, on the next stepi instruction, instead of branching to 0x3bf50, it strangely just changes to 0x47b50...huh?!?
6. From that point on, the program counter just continues to increment by 2, as if to give no care to what the actual instructions are.  Super weird.
7. I have verified 0x3bf50 is valid in memory and is the memset code (from newlib/libc.a)

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain: GNU Arm Embedded Toolchain, 11.3 Release: x86_64 Linux hosted cross toolchains, AArch32 bare-metal target (arm-none-eabi)
 - Commit SHA or Version used: At zephyr 62f5386518a09dbc6fccd2dd18e653ca54ce3087

**Additional context**

For what it's worth, removing
`std::cout << "Hello, C++ world!" << std::endl`
does not fix it, but removing it and \#include \<iostream\> does fix the problem.

I am able to do other C++ stuff, but something about iostream and/or cout with GNU Arm Toolchain and possibly memset is wrong.  

I wish I knew the issue, but at this point, I'm stuck.

nashif, 2025-03-25:
seems to work for me:

```
built with 
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
export GNUARMEMB_TOOLCHAIN_PATH=~/bin/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi

arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc

qemu-system-arm: warning: nic lan9118.0 has no peer
*** Booting Zephyr OS build v4.1.0-1214-g1ac19d3e9811 ***
Hello, C++ world! mps2

```

tejlmand, 2025-05-12:
@nashif I can reproduce the issue, do you build for the `qemu_cortex_m3` board ?:
```
$ west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
qemu-system-arm: terminating on signal 2
ninja: build stopped: interrupted by user.
$
```
Note, things are working with Zephyr SDK.

wearyzen, 2025-05-13:
Hi @tejlmand , I am not experienced with toolchains but the version mentioned in the issue is 11.3 which seems old. I tried with 14.2 like Anas did and I too don't see the issue.
@nslowell would it be possible for you to update arm gnu toolchain to what Zephyr has (12.2) or latest?

```
export ZEHYR_TOOLCHAIN_VARIANT=gnuarmemb; export GNUARMEMB_TOOLCHAIN_PATH=~/workspace/toolchain/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/
west build -p -b qemu_cortex_m3 samples/cpp/hello_world -t run
...
[118/119] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build v4.1.0-3725-g08269963370b ***
Hello, C++ world! qemu_cortex_m3
qemu-system-arm: terminating on signal 2

ninja: build stopped: interrupted by user.

zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-arm --version
QEMU emulator version 7.0.0 (v7.0.0-11-gffe03b7e85-dirty)
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers
```

nslowell, 2025-05-23:
@wearyzen I tried again with 11.3, 12.2, and 14.2 at the HEAD of zephyr as of today, all with the same blocked result as originally described.

I noticed your example showed export GNUARMEMB_TOOLCHAIN_PATH=~/workspace/toolchain/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin/, but when I tried to append "/bin", cmake complained about being unable to find the toolchain b/c it was appending "/bin" to the toolchain path itself.

```
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
export GNUARMEMB_TOOLCHAIN_PATH=~/toolchains/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi
west build -p always -b qemu_cortex_m3 samples/cpp/hello_world
...
west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling

```

closed 2025-08-12 by github-actions[bot]