zephyrproject-rtos/zephyr #87437: STM32WB09 fails during Bluetooth connectivity

mdk, 2025-03-20:
**Describe the bug**
It seems all common Bluetooth samples fail during _Connect_ stage on a vanilla **NUCLEO-WB09KE** board — both in peripheral and central roles. Advertising itself works, scanning as well. 

**To Reproduce**
1. Install zephyr-4.1 using the usual steps
2. `west blobs fetch hal_stm32`
3. Compile ie. `peripheral_ht` sample for board `nucleo_wb09ke/stm32wb09` & flash
4. Use helper tool on iOS or Android or Mac (ie. LightBlue) to inspect BLE devices
5. Find the Nucleo board advertising itself as `Zephyr Temperature Sensor`
6. Click connect

**Expected behavior**
Board can be connected to and service/peripherals can be inspected.

**Experienced behavior**
Connection stalls on the helper app and never completes. Nucleo logs show "connected" and "disconnected" events shortly after.

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0 ***
[00:00:00.009,000] <inf> bt_hci_core: Identity: 02:80:E1:00:00:00 (public)
[00:00:00.009,000] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x1064, manufacturer 0x0030
[00:00:00.009,000] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0x0000
Bluetooth initialized
no temperature device; using simulated data
Advertising successfully started
Connected
Disconnected, reason 0x08 
``` 

(logs with DBG below) 

**Environment:**
 - OS: MacOS
 - Toolchain: Zephyr SDK 
 - Version: both `4.1` branch and recent (20Mar) `1dd9308e2f784aa07bdc3cc00387c0a0c8b58dbe`
 - Board: NUCLEO-WB09KE

mdk, 2025-03-20:
Full log with DBG enabled:

[stm32wb09.txt](https://github.com/user-attachments/files/19375500/stm32wb09.txt)

mdk, 2025-03-23:
Noticed some differences after more testing: 

On iOS/MacOS (NRFConnect app, LightBlue app) the connectivity never succeeds. 
On Android (NRFConnect app on Pixel) it does ~75% of the times on first try but never does on secondary try (in examples that resume advertising ie. `peripheral_hr`).

msmttchr, 2025-03-24:
Tested "*** Booting Zephyr OS build v4.1.0-1132-gc05cfbf15eeb ***":

- Android 12 (Samsung S10e) looks good.
- iOS 18.3.2 (iPhone 11) fails.

antonio-quarta, 2025-04-02:
Tested *** Booting Zephyr OS build v4.1.0-1692-g0f852c82f025 ***

- Android 15 Pixel 7 pro looks good, it connects and reconnects without problems.
- Android 14 Lenovo Tab p12 looks good, it connects and reconnects without problems.
- HP EliteBook 840 G10 Notebook  Windows 11 looks good, it connects and reconnects without problems.

msmttchr, 2025-04-03:
@mdk, a possible workaround is to change the file:
https://github.com/zephyrproject-rtos/hal_stm32/blob/def7e1a025877a35d070439e70668f5ec319c32e/lib/stm32wb0/CMakeLists.txt#L34

```suggestion
set(CFG_BLE_CONTROLLER_2M_CODED_PHY_ENABLED "0")
```
This will disable support for 2M and Coded PHY in the controller, preventing the phone to issue the `LL_PHY_REQ`. Could you please check if this fixes the issues you are seeing?
Even if, this is not a proper fix, this will allow to better isolate the problem.

mdk, 2025-04-03:
@msmttchr tested that this does solve the problem, both on iOS and MacOS.

mdk, 2025-04-03:
Attaching sniffs for both the working (`CFG_BLE_CONTROLLER_2M_CODED_PHY_ENABLED "0"`) and failing connectivity scenarios from iOS. 

[WB09-sniff.zip](https://github.com/user-attachments/files/19591971/WB09-sniff.zip)

msmttchr, 2025-04-03:
> [@msmttchr](https://github.com/msmttchr) tested that this does solve the problem, both on iOS and MacOS.

Thanks for the update, really useful and appreciated. Did you check also Pixel? We were unable to replicate it in our office.

salvopit, 2025-04-09:
Thanks a lot for the sniffer trace, @mdk . 
They confirm that the issue you saw on iOS is the same as we have observed.

[image]

As also highlighted by the sniffer, the phone has _initiated a new control procedure before the previous was complete_.
This is something most probably related to the controller on the Phone (Broadcom).

We are curious to verify if the issue you saw on Pixel phone is the same.

mdk, 2025-04-15:
Hello, approaching this again I couldn't not replicate this on Pixel8a.

After some thinking, I think what was originally happening is that I had iOS or MacOS somehow trying to connect in the background which was causing my Pixel problems, I just didn't notice it. I'm attaching a snif from Pixel 8a (which looks like correct behaviour to me). 

Summing up, the issue is only about iOS/MacOS. Sorry for the confusion. 

[WB09-Pixel.zip](https://github.com/user-attachments/files/19763488/WB09-Pixel.zip)

erwango, 2025-06-30:
@mdk , @msmttchr What is the next step for this issue. Should it be closed ?
If needs to be fixed, can we target v4.2.0 ?

msmttchr, 2025-11-22:
I expect that integration of STM32CubeWB0 v1.4.0 (https://github.com/STMicroelectronics/STM32CubeWB0) will solve this problem. I expect this to happen in the next 2-3 weeks.

msmttchr, 2026-03-17:
@mdk, finally new version has been merged, sorry for the huge delay. Could you please confirm that problem is fixed?

closed 2026-04-30 by github-actions[bot]