zephyrproject-rtos/zephyr #87444: U-Blox M10 UBX/NMEA mix issues with M8 driver

XDjackieXD, 2025-03-21:
**Describe the bug**

The way the M8 driver configures the module, it regularly sends UBX messages which are not parsed by the driver (as the ubx pipe is only attached if the driver expects a UBX message as response to some request).

This seems to work fine with the M8 modules but the M10 modules seem to "confuse" the NMEA pipe with the UBX messages, leading to the GNRMC messages to not be read.

Simply disabling the UBX output by adding `payload.out_proto_mask = UBX_CFG_PRT_OUT_PROTO_NMEA;` to `ubx_m8_ubx_cfg_prt_set` seems to fix this without any noticeable downsides.  
I'd also be fine with some interface I could call from user code to disable UBX output.

**To Reproduce**
1. connect U-Blox M10 based module to some board
2. run GNSS sample on this board
3. Observe GNSS data callback never getting called because of missing GNRMC message

**Expected behavior**
The M8 driver should also work with the M10 modules - the protocol is similar enough and this seems to be the only change required for basic functionality.

**Impact**
I need to patch the zephyr codebase to make my project work.  
Mostly an annoyance but should be a trivial fix.

**Logs and console output**
no relevant output. If I add some printk statements I can see that the RMC parser never gets called, and therefore `rmc_utc` never getting set, which prevents `gnss_publish_data` from getting called.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr? (West + git clone of zephyr + out-of-tree project)
 - Commit SHA: 43564e740450f6b5564c3d8709948ae6e5e83b50 (tagged v4.1.0-rc3)

bjarki-andreasen, 2025-03-24:
Could you possibly create a PR which does "payload.out_proto_mask = UBX_CFG_PRT_OUT_PROTO_NMEA; to ubx_m8_ubx_cfg_prt_set", seems like an appropriate fix :)

XDjackieXD, 2025-03-25:
PR added (https://github.com/zephyrproject-rtos/zephyr/pull/87627)

closed 2025-03-26 by kartben