zephyrproject-rtos/zephyr #87454: Renesas HAL generates CMake warning when building for non-Renesas platforms

jhedberg, 2025-03-21:
**Describe the bug**

The Zephyr build recently started outputting a warning related to the Renesas HAL when building anything that's not Renesas.

**To Reproduce**

Build any app for any non-Renesas platform, e.g.:
```
west build -p -b qemu_cortex_m3 samples/hello_world/
```

**Expected behavior**

No build warnings from unrelated HALs/modules

**Impact**

Annoyance & unnecessary noise in the build output.

**Logs and console output**

```
CMake Warning at /Users/johedber/src/zephyr/zephyr/CMakeLists.txt:1011 (message):
  No SOURCES given to Zephyr library: ..__modules__hal__renesas__drivers

  Excluding target from build.
```

**Environment (please complete the following information):**

 - OS: macOS
 - Toolchain: Zephyr SDK 0.17.0
 - Latest `main`, commit d4d81637fbec0a1966c615c0d3ece86e3fb2cb19 in my case

soburi, 2025-03-21:
@KhiemNguyenT 

Can this fix with https://github.com/zephyrproject-rtos/hal_renesas/pull/86 ?

jhedberg, 2025-03-21:
> Can this fix with [zephyrproject-rtos/hal_renesas#86](https://github.com/zephyrproject-rtos/hal_renesas/pull/86) ?

Looks like that should fix it. Is there a Zephyr main tree PR yet to update to that version of the HAL?

quytranpzz, 2025-03-24:
> > Can this fix with [zephyrproject-rtos/hal_renesas#86](https://github.com/zephyrproject-rtos/hal_renesas/pull/86) ?
> 
> Looks like that should fix it. Is there a Zephyr main tree PR yet to update to that version of the HAL?

Hi @jhedberg , the HAL version was updated via PR https://github.com/zephyrproject-rtos/zephyr/pull/87445. Thank you.

jhedberg, 2025-03-24:
> Hi [@jhedberg](https://github.com/jhedberg) , the HAL version was updated via PR [#87445](https://github.com/zephyrproject-rtos/zephyr/pull/87445). Thank you.

So this issue should be closed then? Why wasn't there a "Fixes ..." entry in that PR?

pdgendt, 2025-03-24:
> So this issue should be closed then? Why wasn't there a "Fixes ..." entry in that PR?

I think so, yes. The issue was first reported on discord, so no issue was available.

jhedberg, 2025-03-24:
Fixed by #87445

quytranpzz, 2025-03-24:
> So this issue should be closed then? 

Yes, please.

> Why wasn't there a "Fixes ..." entry in that PR?

Sorry. The issue was first reported on Discord, and we created a PR to fix it before this ticket issue was officially opened, so I wasn’t aware of it. I have updated the PR to mention this issue. Thank you

closed 2025-03-24 by jhedberg