zephyrproject-rtos/zephyr #87497: `west flash --context` fails for Sysbuild projects

brycewilkins, 2025-03-21:
**Describe the bug**
Note: The bug described below is present when a project is built using **Sysbuild** (`west build --sysbuild`). The bug does not occur if the project is built using Zephyr's build system (`west build --no-sysbuild`).

`west flash --context` should list the available runners for a specific board, along with various common and runner-specific options. However, if the project is built with Sysbuild there is a fatal error:
```
$ west flash --context
FATAL ERROR: no runners.yaml found in /home/user/project/app/build/zephyr. Either board None doesn't support west flash/debug/simulate, or a pristine build is needed.
```

**To reproduce**
```
$ west build --board ${MY_BOARD} --build-dir build --pristine --sysbuild .
$ west flash --board ${MY_BOARD} --build-dir build --context
```

**Expected behavior**
Build and runner information is output similar to the following:
```
build configuration:
  build directory: build/abc
  board: abc/nrf52840
  runners.yaml: build/abc/zephyr/runners.yaml
...
```

**Additional information/Troubleshooting**
My project has a single domain (`abc`); no bootloader or other application is built. The `build/domains.yaml`  file looks like:
```
default: abc
build_dir: /home/user/project/app/build
domains:
  - name: abc
    build_dir: /home/user/project/app/build/abc
flash_order:
  - abc
```

When I run `west flash --context`:
- `zephyr/scripts/west_commands/run_common.py` `dump_context()` looks at `build/zephyr/.config` (the Sysbuild Kconfig) to load the build configuration and determine the board (`CONFIG_BOARD_TARGET`), but that returns `{}` (`None`) because it is looking (incorrectly) at the Sysbuild Kconfig file instead of the `abc` domains Kconfig file.
- `runners.yaml` file is not in `build/zephyr`; it is actually in `build/abc/zephyr`.

**Work around**
If I include the domain `abc` as part of the build path:
```
$ west flash --board ${MY_BOARD} --build-dir build/abc --context
```
results in expected output.

Furthermore, if the code has been built with `west build ..... --no-sysbuild`, this problem does not occur.

**Impact**
This is a minor annoyance. I'm not sure how `--context` would apply for Sysbuild projects if there are multiple domains. Would the `--context` information have to be output for all domains or just the default one?

While I could find a work around it may be unexpected and troublesome. Also, I'm not sure if my work around is the appropriate thing to do.

**Environment:**
 - OS: Linux (Fedora 40)
 - Toolchain: Zephyr SDK 0.17.0
 - Zephyr OS: v3.7.99 (SHA ca954a6216c9)
 - nRF Connect SDK v2.9.1 (SHA 60d0d6c8d42d)

henrikbrixandersen, 2025-03-21:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

brycewilkins, 2025-03-21:
> Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

@henrikbrixandersen Thanks for the note! I will check this on Zephyr `main` and report back.

brycewilkins, 2025-03-22:
@henrikbrixandersen This is reproducible on Zephyr `main`. 

Below I use an NXP board and `samples/hello_world`:

```
$ west build --pristine --sysbuild -b frdm_mcxn947/mcxn947/cpu0 samples/hello_world
...
$ west flash --context                                                                                                                         
FATAL ERROR: no runners.yaml found in /home/bryce/zephyr/build/zephyr. Either board None doesn't support west flash/debug/simulate, or a pristine build is needed.
```

If I provide the `--build-dir` option and append the Sysbuild domain `hello_world` as part of the build path the output is as expected:
```
$ west flash --context --build-dir build/hello_world
-- west flash: rebuilding
ninja: no work to do.
build configuration:
  build directory: build/hello_world
  board: frdm_mcxn947/mcxn947/cpu0
  runners.yaml: build/hello_world/zephyr/runners.yaml
...
```

If the build is done using Zephyr's build system there is no problem:
```
$ west build --pristine -b frdm_mcxn947/mcxn947/cpu0 samples/hello_world
...
$ west flash --context                                                                                                                         
-- west flash: rebuilding
ninja: no work to do.
build configuration:
  build directory: build/hello_world
  board: frdm_mcxn947/mcxn947/cpu0
  runners.yaml: build/hello_world/zephyr/runners.yaml
...
```

This issue is not related to nRF Connect SDK (NCS) and I have updated my original post to link to this example.

vignesh-atmosic, 2025-03-24:
I have a fix for this! Pushing it soon.

vignesh-aerlync, 2025-03-24:
I have a fix for this. I will raise pull request soon.

vignesh-aerlync, 2025-05-02:
Hi @nordicjm Could you please reopen this issue?  The domain_file should have been handled more effectively to support multi-image flashing, and I’ve now implemented the necessary changes.

nordicjm, 2025-05-06:
> Hi [@nordicjm](https://github.com/nordicjm) Could you please reopen this issue? The domain_file should have been handled more effectively to support multi-image flashing, and I’ve now implemented the necessary changes.

Done

PR #89535 (2025-05-08): west: fix for `west flash --context` with sysbuild.
Fixes #87497

- show west flash --context with sysbuild.
- flashing multiple images is tested.

Originally merged here: https://github.com/zephyrproject-rtos/zephyr/pull/87910 and then reverted here: https://github.com/zephyrproject-rtos/zephyr/pull/89179
closed 2025-05-08 by kartben