zephyrproject-rtos/zephyr #87516: STM32 I2C DMA Incorrect DMA config

erian747, 2025-03-23:
**Describe the bug**

Testing communication on I2C4 on a STM32H747i-Discovery board with DMA turned on
results in an error message:
```
[00:00:43.223,000] <err> dma_stm32_bdma: source and dest unit size error, 0  
[00:00:43.223,000] <err> dmamux_stm32: cannot configure the dmamux.  
[00:00:43.223,000] <err> i2c_ll_stm32_v2: Problem setting up DMA: -22  
[00:00:43.223,000] <err> i2c_ll_stm32_v2: Problem setting up TX DMA   

Seems like the 'source_data_size' is not setup in configure_dma() in  I2C driver 
```
**To Reproduce**

Use configuration:
```
CONFIG_I2C_STM32_V2_DMA=y
CONFIG_DMA_STM32_BDMA=y
```
And device tree overlay
```
&i2c4 {
	dmas = <&dmamux2 0 14 (STM32_DMA_PERIPH_TX | STM32_DMA_PRIORITY_HIGH)
		&dmamux2 1 13 (STM32_DMA_PERIPH_RX | STM32_DMA_PRIORITY_HIGH)>;
		dma-names = "tx", "rx";
};
&bdma1 {
	status = "okay";
};
&dma2 {
	status = "okay";
};
&dmamux2 {
	status = "okay";
};
```
Test communication on I2C4 (Touch controller on STM32H747i-Discovery)
**Expected behavior**

No error message and communication working
**Impact**

None
**Logs and console output**

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Ubuntu 22
 - Toolchain (e.g Zephyr SDK, ...) NA
 - Commit SHA or Version used 56f7dc4c2ea3390ef8efeed295334afc62ebbea9
**Additional context**

sgilbert182, 2025-03-23:
@JarmouniA thanks for tagging me, I'll take a look at this shortly. 

@erian747 I don't have the same nucleo board but I'll see if I can set something up that's similar enough here and go from there

erian747, 2025-03-23:
I shall do some debugging later today with source_data_size and dest_data_size set to 1 and see what happens

sgilbert182, 2025-03-23:
I appreciate that. I vaguely remember those values being hardcoded in the i2c_ll macrobatics because it's all byte work. But it's clearly no longer the case. I don't have access to my work computer where I tested it working last but I will see if I can get it working in it's current form on a clean set up

erian747, 2025-03-23:
I got DMA to accept configuration by setting
	dma_cfg->source_data_size=1;
	dma_cfg->dest_data_size=1;
and data->dma_cfg.channel_direction to MEMORY_TO_PERIPHERAL or PERIPHERAL_TO_MEMORY depending on Write / Read

BTW i can see that at other places, DMA configuration is set up like in uart_stm32.c below, maybe something similar can be used for i2c?
[image]

sgilbert182, 2025-03-23:
> I got DMA to accept configuration by setting dma_cfg->source_data_size=1; dma_cfg->dest_data_size=1; and data->dma_cfg.channel_direction to MEMORY_TO_PERIPHERAL or PERIPHERAL_TO_MEMORY depending on [...]

It was based off the UART driver, and it was pretty much a copy and paste except for a few fields (like the callback) I'm unsure why I didn't include those fields in my MR. I'll put together an MR with a fix shortly, if you would be happy to test?

erian747, 2025-03-24:
Yes i hopefully can try it out to night

erian747, 2025-03-24:
I got compile errors,  struct stream in i2c_stm32_ll.h seems to miss a few fields
```
 struct stream {
	const struct device *dev_dma;
	int32_t dma_channel;
};
```

sgilbert182, 2025-03-25:
Unsure how I managed it but I was building a stale project. I've set up a clean project and I've made the changes to get it to build now. I'm looking to test with an h735 when I get the chance and I am hoping it is similar enough to your setup. 

Changes have been pushed and the PR marked draft until I can test it

sgilbert182, 2025-03-26:
@erian747 latest changes pushed seem to transmit OK on my h735 board but it is a very basic test program. Could you run up these changes and see if it works for you please? If it works OK I can make the PR ready for review

erian747, 2025-03-26:
Good job, now it compiles and I2C communication seems to be working!
Left a comment in PR about unused variables

sgilbert182, 2025-03-26:
Thanks for testing and your feedback!

sgilbert182, 2025-04-20:
Fix doesn't appear to have  gone in yet

erian747, 2025-04-20:
Sorry, i closed it yesterday without actually checking if fix was merged, reopen if you wish

PR #87534 (2025-04-22): i2c: stm32: add missing DMA configuration fields
Add missing fields for DMA tx and rx configuration macros

fix #87516

Tested on stm32h735-Discovery and STM32H747i-Discovery board
closed 2025-04-22 by kartben