zephyrproject-rtos/zephyr #87531: stm32h573i_dk can't boot from external flash with mcuboot

Jowllel, 2025-03-23:
**Describe the bug:**

I'm trying to boot my application from the external ospi nor flash on the stm32h573i_dk board. My project ([MCUBOOT_EXT_Test.zip](https://github.com/user-attachments/files/19411028/MCUBOOT_EXT_Test.zip)) is based of the mcuboot sample. When I place the slot0 and slot1 partition in the internal flash everything works as expected, however when I place them in the external flash the code stops, when jumping to the application image. (As shown in the log below) At this point the MCU can't be accessed through the debugger (west attach or STM32CubeProgrammer), because it's in a locked up state.

**To Reproduce:**

I uploaded my project files as a zip. Build them with sysbuild and flash them to an stm32h573i_dk. And edit the board file (stm32h573i_dk.dts) in the zephyr folder, so the slot0 and slot1 partitions are placed in the ospi nor flash.

[MCUBOOT_EXT_Test.zip](https://github.com/user-attachments/files/19411028/MCUBOOT_EXT_Test.zip)

**CMake Warnings for mcuboot target:**
```
CMake Warning at C:/Users/xxxx/zephyrproject/zephyr/CMakeLists.txt:1011 (message):
  No SOURCES given to Zephyr library: ..__modules__hal__renesas__drivers

  Excluding target from build.

MCUBoot bootloader key file: C:/Users/xxxx/zephyrproject/bootloader/mcuboot/root-rsa-2048.pem
CMake Warning at CMakeLists.txt:367 (message):
  WARNING: Using default MCUboot signing key file, this file is for debug use
  only and is not secure!

CMake Warning at CMakeLists.txt:589 (message):
  Unable to determine erase size of slot0 or slot1 partition, setting to 1
  (this is probably wrong)

CMake Warning at CMakeLists.txt:594 (message):
  Unable to determine write size of slot0 or slot1 partition, setting to 8
  (this is probably wrong)
```

**CMake Warnings for main app target:**
```
warning: MCUBOOT_UPDATE_FOOTER_SIZE (defined at subsys/dfu/Kconfig:55) was assigned the value
'0x3031' but got the value ''. Check these unsatisfied dependencies: MCUBOOT_IMG_MANAGER (=n),
IMG_MANAGER (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE and/or look up
MCUBOOT_UPDATE_FOOTER_SIZE in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

CMake Warning at C:/Users/xxxx/zephyrproject/zephyr/CMakeLists.txt:1011 (message):
  No SOURCES given to Zephyr library: ..__modules__hal__renesas__drivers

  Excluding target from build.

-- Including signing script: C:/Users/xxxx/zephyrproject/zephyr/cmake/mcuboot.cmake
CMake Warning at C:/Users/xxxx/zephyrproject/zephyr/cmake/mcuboot.cmake:71 (message):
  slot0_partition write block size devicetree parameter is missing, assuming
  write block size is 4
Call Stack (most recent call first):
  C:/Users/xxxx/zephyrproject/zephyr/cmake/mcuboot.cmake:245 (zephyr_mcuboot_tasks)
  C:/Users/xxxx/zephyrproject/zephyr/CMakeLists.txt:2053 (include)

```

**Logs and console output:**
```
D: Flash @0x8000000 initialized. BS: 16
D: Block 0: bs: 8192 count: 256
D: XSPI Init'd
D: Delay Block Init
D: Reset Mem (SPI/STR)
D: Status Match cb
D: Mem Ready (SPI/STR)
D: XSPI configuring Octo SPI mode
D: Status Match cb
D: Status Match cb
D: Status Match cb
D: Status Match cb
[... 151 lines cut ...]
D: Memory-mapped read from 0x90006900, len 256
D: Memory-mapped read from 0x90006a00, len 256
D: Memory-mapped read from 0x90006b00, len 256
D: Memory-mapped read from 0x90006c00, len 256
D: Memory-mapped read from 0x90006d00, len 256
D: Memory-mapped read from 0x90006e00, len 50
D: Memory-mapped read from 0x90006e32, len 4
D: Memory-mapped read from 0x90006e36, len 4
D: Memory-mapped read from 0x90006e3a, len 32
D: Memory-mapped read from 0x90006e5a, len 4
D: Memory-mapped read from 0x90006e5e, len 32
D: Memory-mapped read from 0x90006e7e, len 4
D: Memory-mapped read from 0x90006e82, len 256
I: Bootloader chainload address offset: 0x0
I: Image version: v0.0.0
I: Jumping to the first image slot
D: Memory-mapped read from 0x90000400, len 8

```

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain: Zephyr SDK, CLion
 - Zephyr: 4.1.99

FRASTM, 2025-04-04:
> [@Jowllel](https://github.com/Jowllel) it would be more helpful for people who want to help if you could provide a git branch instead of a zip file, and reproduce the issue with a Zephyr sample (for example samples/sysbuild/with_mcuboot) if possible. FYI there is this PR [mcu-tools/mcuboot#2154](https://github.com/mcu-tools/mcuboot/pull/2154) that hasn't been merged yet in upstream MCUBoot.

@Jowllel, I will go with the stm32h573 disco kit in the coming days to execute from external flash

FRASTM, 2025-04-11:
I finally found the problem : same as for the stm32H7, the stm32H5 should not disable the PLL when booting from the MCU_Boot   since the xspi is clock by the PLL
```
*** Booting MCUboot v2.1.0-rc1-294-gf96de215453d ***
*** Using Zephyr OS build v4.1.0-1737-g496a511978ce ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x0
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.1.0-1737-g496a511978ce ***
Hello World! from external flash  stm32h573i_dk
--> PC at 0x9000092a
```

--> PR to come

closed 2025-07-19 by aescolar