zephyrproject-rtos/zephyr #87560: IRIS w01 support

ouysal-git, 2025-03-24:
I have IRIS w01 board and i wonder is that supported by zephyr.

jakkra, 2025-04-02:
@perssonmagnus

perssonmagnus, 2025-04-03:
Hello @ouysal-git ,

There is as of today no public guidance available for using Zephyr on the IRIS-W1 boards. I am unfortunately not fully up to date on possible ongoing work in this area so I would propose to ask the questions on the [u-blox Support Community]((https://portal.u-blox.com/s/) in the Short Range section, and a colleague more familiar with the matter will get back to you.

ouysal-git, 2025-04-04:
Hi Magnus, thanks for your reply. I have succeeded it with following the instruction of other boards which are based RW612(RD-RW612-BGA). You can close this issue if you want.

closed 2026-09-26 by github-actions[bot]