zephyrproject-rtos/zephyr #87602: Add Support for ArteryTek SoC's HAL driver

Maxjta, 2025-03-25:
## Origin

Q: Name of project hosting the original open source code, Provide a link to the source 
A: The HAL driver link (https://github.com/Maxjta/hal_at32.git)
A: The zephyr device driver PR [86694](https://github.com/zephyrproject-rtos/zephyr/pull/86694).

## Purpose

Q: Brief description of what this software does
A: This software support ArteryTek Soc's HAL driver.

## Mode of integration

Q: Describe whether you'd like to integrate this external component in the main tree
or as a module, and why. If the mode of integration is a module, suggest a
repository name for the module

A: Arterytek  is an SoC manufacturer that supports many models. Currently, many customers want to use Zephyr OS and integrate Arterytek SoC into the Zephyr OS system. We hope to creating a repo named hal_at32 to better support Zephyr OS in the future.

## Maintainership

Q: List the person(s) that will be maintaining the integration of this external code
for the foreseeable future. Please use GitHub IDs to identify them. You can
choose to identify a single maintainer only or add collaborators as well

A: maintainer: [Maxjta](https://github.com/Maxjta)

## Pull Request

Q: Pull request (if any) with the actual implementation of the integration, be it
in the main tree or as a module (pointing to your own fork for now). Make sure
the PR is correctly labeled as "DNM"

A: The zephyr device driver PR [86694](https://github.com/zephyrproject-rtos/zephyr/pull/86694).

## Description

Long description that will help reviewers discuss suitability of the
component to solve the problem at hand (there may be a better options
available.)

Q: What is its primary functionality (e.g., SQLLite is a lightweight
database)?
A: Provides Arterytek SoC hardware driver interface.

Q: What problem are you trying to solve? (e.g., a state store is
required to maintain ...)
A: Add support Arterytek SoC's driver in Zephry OS.

Q: Why is this the right component to solve it (e.g., SQLite is small,
easy to use, and has a very liberal license.)
A:This is a  Standard and efficient hardware driver for Arterytek SoC’s.
## Security

Q: Does this component include any cryptographic functionality?
If so, please describe the cryptographic algorithms and protocols used.
A： ArteryTek HAL driver is not have cryptographic functionality.

## Dependencies

What other components does this package depend on?

Q: Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?
A: The Zephyr project direct dependency on the component hal_at32.

## Revision

Q: Version or SHA you would like to integrate initially
A: The [hal_at32](https://github.com/Maxjta/hal_at32.git) Revision(5b427d057902e13f7036f0f8d9221a0d1cb919cd)

## License

Q: Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``
A: Yes, the license is BSD 3-Clause License

Maxjta, 2025-07-07:
@nashif 
(https://github.com/Maxjta/hal_at32/blob/main/at32f423/drivers/inc/at32f423_adc.h#L6)
at32f423_adc.h:  the driver part is the standard library of Arterytek. This part is open source. Does this part need to be modified?

(https://github.com/Maxjta/hal_at32/blob/main/common_source/at32_hal_udc.c)
Is this part of the apache license acceptable? Or do I need to modify this part of the license to  BSD-3.

nashif, 2025-07-09:
you do not need to modify the licenses, but any licenses included in the HAL need to be documented and all applicable licenses need to be listed as part of the original request.

Please use SPDX to mark all files with the appropriate licenses and avoid using custom text for licenses.

Maxjta, 2025-07-10:
@nashif Thank‘s. 

  I will modify the hal_at32/LICENCE statement and declare this license  as the Apache-2.0 and arterytek custom licenses. Is this OK?

carlescufi, 2025-07-30:
> [@nashif](https://github.com/nashif) Thank‘s.
> 
> I will modify the hal_at32/LICENCE statement and declare this license as the Apache-2.0 and arterytek custom licenses. Is this OK?

Licensing the software under the Apache License is not necessary, but the license needs to be accepted and compatible (i.e. accepted by the OSI and non-copyleft). Also, the license information must be clearly present (if possible as SPDX tags) in each file.

Maxjta, 2025-08-15:
> [@Maxjta](https://github.com/Maxjta) any updates? Have you made the required changes?

It is already being modified and it is expected that this part of the update can be submitted next week.

Maxjta, 2025-08-15:
> [@Maxjta](https://github.com/Maxjta) any updates? Have you made the required changes?

@carlescufi The copyright statment  has been updated to the driver file.
The Commit: https://github.com/Maxjta/hal_at32/commit/aae6b15ddc492e1621c14b1c57024d0ad9ddf807

Maxjta, 2025-09-01:
@carlescufi Is there anything else that needs to be modified?

carlescufi, 2025-09-10:
> [@carlescufi](https://github.com/carlescufi) Is there anything else that needs to be modified?

No, I will bring this to the attention of the TSC chair, @nashif

Maxjta, 2025-09-18:
> It looks like the HAL contains several copies of CMSIS-Core and CMSIS-DSP headers. In Zephyr, the shared CMSIS_6 module should be used, the HAL shouldn't contain its own copy.
> 
> Also, it looks like the drivers still have a proprietary license:
> 
> ```
>   * Copyright (c) 2025, Artery Technology, All rights reserved.
>   * 
>   * The software Board Support Package (BSP) that is made available to 
>   * download from Artery official website is the copyrighted work of Artery. 
>   * Artery authorizes customers to use, copy, and distribute the BSP 
>   * software and its related documentation for the purpose of design and 
>   * development in conjunction with Artery microcontrollers. Use of the 
>   * software is governed by this copyright notice and the following disclaimer.
> ```

OK, I will delete CMSIS-core and CMSID-DSP.

Maxjta, 2025-09-24:
> [@Maxjta](https://github.com/Maxjta) also, looks like the linked PR has been closed ([#86694](https://github.com/zephyrproject-rtos/zephyr/pull/86694)). If there's a new one (I suspect [#91273](https://github.com/zephyrproject-rtos/zephyr/pull/91273)), please update the Issue description to include it.

OK, Thanks, the issue description is include PR https://github.com/zephyrproject-rtos/zephyr/pull/91273

asmellby, 2025-09-24:
> @asmellby Is it accurate that this has been addressed as well?

Yeah, looks good to me too. License is consistent, and duplicated content has been removed.

LuoZhongYao, 2025-11-10:
@Maxjta https://github.com/Maxjta/hal_at32/blob/2c8de655d3d6774e6412411a6f29bab0304fcdf5/include/dt-bindings/pinctrl/at32f405r(c-g-m)xx-pinctrl.h#L428C9-L428C23
The definition here is inconsistent with the one in the specification. Which one is correct?

Maxjta, 2025-11-11:
> [@Maxjta](https://github.com/Maxjta) https://github.com/Maxjta/hal_at32/blob/2c8de655d3d6774e6412411a6f29bab0304fcdf5/include/dt-bindings/pinctrl/at32f405r(c-g-m)xx-pinctrl.h#L428C9-L428C23 The definition here is inconsistent with the one in the specification. Which one is correct?

Refer to the specification

Maxjta, 2026-01-04:
@ubieda 
 Is there anything else that needs to be changed here?

Maxjta, 2026-01-22:
> [@Maxjta](https://github.com/Maxjta) great to see AT32F405 to get upstream support. I tried to run the `hello_world` sample on a board very similar to the `at_start_f405`, but it does not print anything on the output pin. The Zephyr support from Artery however works: https://github.com/ArteryTek/zephyr/tree/artery-v1.1-branch. The USART driver seems to be there, is there something missing in the PR [#101409](https://github.com/zephyrproject-rtos/zephyr/pull/101409).

@mlasch This issue has been modified.

Maxjta, 2026-03-02:
@nashif We would like to ask when we can add AT32 HAL driver to the Zephyr organization. Could you please advise on the procedure and the estimated timeline?

Thank you.

jalyli2027, 2026-09-04:
Hi @maintainers,

I’m reaching out from the ArteryTek/AT32 official team regarding this issue.

We would like to ask if this issue could be reopened for further discussion.

We are now actively interested in upstreaming AT32 support to Zephyr from the vendor side. As the AT32 vendor, we are willing to provide engineering resources for the upstream development and commit to long-term maintenance of the AT32 support.

We understand that the original proposal focused primarily on integrating the AT32 HAL as a Zephyr module. We would like to revisit the overall approach based on the current Zephyr architecture and Hardware Model, rather than assuming that the previous implementation is the final approach.

We would like to work with the Zephyr maintainers to determine the appropriate upstream architecture for:

AT32 SoC support
AT32 HAL integration
Zephyr-native peripheral drivers
DeviceTree and Kconfig
Reference board support
Testing and CI
Long-term maintenance

We are happy to update the previous proposal or prepare a new RFC based on the maintainers’ recommendations.

If reopening this issue is not appropriate, could you please advise whether we should create a new RFC/issue for the current proposal?

We would appreciate the opportunity to continue the discussion and work with the Zephyr community toward upstream AT32 support.

Thank you!

martinjaeger, 2026-09-04:
@jalyli2027 and @Maxjta It makes sense to continue discussion here. I removed the Stale label and moved it back to "In Progress" for TSC attention.

jalyli2027, 2026-09-07:
> [@jalyli2027](https://github.com/jalyli2027) and [@Maxjta](https://github.com/Maxjta) It makes sense to continue discussion here. I removed the Stale label and moved it back to "In Progress" for TSC attention.

Thank you for your attention and for removing the Stale label and moving

the issue back to "In Progress" for TSC attention.
 
We really appreciate your continued support on this topic. Our team is

taking this matter seriously, and multiple members of our team will be

actively involved in this project and in maintaining the Artery SOC support

going forward.
 
The following members of our team will be involved in maintaining the

AT32 support:
 
- [@Maxjta](https://github.com/Maxjta)

- [@jalyli2027](https://github.com/jalyli2027)

- [@sheltonyu](https://github.com/sheltonyu)

- [@thomastao2026](https://github.com/thomastao2026)

- [@dannydeng-arterytek](https://github.com/dannydeng-arterytek)

- [@ScottZhangArtery](https://github.com/ScottZhangArtery)

- [@bobyang21](https://github.com/bobyang21)
 
Based on the discussion so far, we are planning to make the following

changes. We would appreciate any feedback or suggestions from the

Zephyr maintainers before we proceed.
 
1. **HAL repository name**
 
   We would like to rename the HAL repository from `hal_at32` to

   `hal_artery`, which is consistent with our company name and provides

   a better foundation for supporting Artery Technology SoCs in the

   future.
 
2. **HAL license**
 
   We plan to change the HAL license to the BSD-3-Clause license.
 
Once these changes are completed, we will update the issue and related

PR accordingly. We would appreciate it if the Zephyr maintainers could

review the changes and provide any feedback or guidance on the next

steps.
 
Thank you again for your time and support.

Maxjta, 2026-09-15:
@martinjaeger 

Thanks for reopening this issue.

We have updated the HAL integration information:

- Official ArteryTek HAL: https://github.com/ArteryTek/hal_artery
- License: BSD-3-Clause
- Additional maintainers have been added

Could you please take a look and let us know if any further changes are required for the TSC review?

Thank you!

jalyli2027, 2026-09-17:
Hi @martinjaeger,

Just following up on the updated HAL integration information above. When you have a chance, could you please take a look and let us know if any further changes are needed for the TSC review?

We’d appreciate your feedback so we can proceed with the next steps.

Thank you!

martinjaeger, 2026-09-17:
Hi @Maxjta and @jalyli2027. We briefly discussed this topic in the TSC meeting last week. BSD license is fine and the repo name should also not be an issue.

We need to give people a bit more time to check the repo again after your recent changes.

I had a brief look and noticed that not all files are consistently licensed under BSD, see here for example:
https://github.com/ArteryTek/hal_artery/blob/32ffb504b71db2a371dd7f5d62921ef634dd6527/at32f423/cmsis/cm4/device_support/at32f423.h#L8

Also I would suggest to add a README.md to your repo.

BTW: There is also a new APAC-friendly TSC meeting now. Feel free to join them if you like. They are open for anyone. See [here](https://lists.zephyrproject.org/g/tsc) for instructions to get a calendar invite.

Maxjta, 2026-09-17:
Hi @martinjaeger :

Thank you for the feedback and for discussing this in the TSC meeting.

Our team has updated the `hal_artery` repository accordingly:
- Added `NOTICE.md` and `OSS_MANIFEST.md` to clarify license and copyright information

We will continue to review the repository and address any additional feedback from the maintainers.

Thank you also for sharing the new APAC-friendly TSC meeting. This is very helpful for our team，and we will look into joining the meetings
and participating in the TSC discussions.

Thanks again for your guidance and support.

 `hal_artery`  Repository maintainers:
@jalyli2027 @sheltonyu @thomastao2026

jalyli2027, 2026-09-28:
Hi @martinjaeger,

Following up on this issue — we have completed the requested license changes, and Maxjta has replied with the updated information.

Could you please help review it when you have a chance? If everything looks good, we would appreciate it if the issue could move forward to the next step.

Please let us know if there are any additional changes or information needed from our side.

Thank you!

Maxjta, 2026-11-12:
Hi @martinjaeger and @maintainers,

Just following up on this topic. We would like to check if there are any updates or further feedback regarding the `hal_artery` repository review.

Our team remains committed to contributing AT32 support to Zephyr and maintaining the repository.

Please let us know if any further changes or actions are needed from our side.

Thank you for your time and support!

Best regards,
ArteryTek HAL Repository Maintainers