zephyrproject-rtos/zephyr #87621: [ESP32] Instable BLE connection

NunoBarix, 2025-03-25:
Trying to connect two BLE devices using the "Central" application on the "samples/bluetooth" directory. I have only commented out the "bt_conn_disconnect" call inside the connected callback just to keep the connection alive.

The Central application runs on an ESP32-WROOM-32E chip and the other LE device is a beacon kind. The connection between the two devices has many instabilities. Sometimes the connection doesn't happen, other times it takes a few seconds to establish and, the majority of the times, the connection is established but it automatically disconnects after a few seconds (reason 0x08).

I have a RF sniffer where I can see the data exchanged between the two devices. I can see that, after connection, sometimes there are a lot of MTU exchange packets. Like if the peers don't recognize the packet sent by the other end. I can also see a lot of L2CAP fragment packets with 135 bytes size in between these packet exchanges. I have the Wireshark log of the RF sniffer capture, but github doesn't allow me to upload it. However, below is a screenshot of what I am talking about. 

[image]

I also tried to adjust the interval (min and max), latency, and supervision timeout to match the ones on the beacon, but it didn't help. What could be causing this issue? A similar application running on Arduino framework works without any problem. So, it doesn't seem to be related to the beacon or to the ESP32 hardware.

Below are the logs taken from the console:
Device found: D3:C6:D4:C4:B9:24 (public) (RSSI -11)
Connected: D3:C6:D4:C4:B9:24 (public)
Disconnected: D3:C6:D4:C4:B9:24 (public), reason 0x08 
Scanning successfully started

Device found: D3:C6:D4:C4:B9:24 (public) (RSSI -11)
Connected: D3:C6:D4:C4:B9:24 (public)
Disconnected: D3:C6:D4:C4:B9:24 (public), reason 0x3e 
Scanning successfully started

Device found: D3:C6:D4:C4:B9:24 (public) (RSSI -12)
Connected: D3:C6:D4:C4:B9:24 (public)
Disconnected: D3:C6:D4:C4:B9:24 (public), reason 0x3e 
Scanning successfully started

Device found: D3:C6:D4:C4:B9:24 (public) (RSSI -12)
Connected: D3:C6:D4:C4:B9:24 (public)
Disconnected: D3:C6:D4:C4:B9:24 (public), reason 0x08 
Scanning successfully started

Device found: D3:C6:D4:C4:B9:24 (public) (RSSI -11)
Connected: D3:C6:D4:C4:B9:24 (public)
Disconnected: D3:C6:D4:C4:B9:24 (public), reason 0x08 
Scanning successfully started

fabiobaltieri, 2025-03-25:
Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

NunoBarix, 2025-03-25:
Changed the issue description to match the bug report template and added some more useful information. I am glad to provide more if needed. 
I also discovered this issue was introduced on version 4.0.0. BLE connection works great on version 3.7.0.

NunoBarix, 2025-03-26:
> [@NunoBarix](https://github.com/NunoBarix) can you confirm whether your ESP32 is rev3.0 or rev3.1?

@sylvioalves, it's rev3.1.

raffarost, 2025-04-03:
I reproduced the problem locally and it seems the unstable connection happens with the chip's rising temperature.
Would you mind confirming or debunking?

PR #92130 (2025-06-26): soc: espressif: sync hal to latest updates
1) Bring latest hal_espressif updates/sync to latest v5.1 branch. 
2) Update RF libraries for bug fixes and improvements
3) Add necessary BLE Kconfig entries to support latest changes.

Fix #86554
Fix #57980
Fix #87621