zephyrproject-rtos/zephyr #87625: subsys: logging: log_core_additional: test_log_domain_id Twister reports fails even the test passed

arikgreen, 2025-03-25:
**Describe the bug**

The fail was observed on platforms:
- intel_adsp/ace15_mtpm
- intel_adsp/cavs25
- up_squared/apollo_lake

The twister reporting fail in twister.json file for this test:
```
{
    "name":"logging.thread",
    "arch":"xtensa",
    "platform":"intel_adsp/cavs25",
    "path":"tests/subsys/logging/log_core_additional",
    "run_id":"18da6c8e0a8773ff3bd612de2bb0ac71",
    "runnable":true,
    "retries":0,
    "toolchain":"zephyr",
    "status":"passed",
    "execution_time":"10.25",
    "build_time":"40.24",
    "testcases":[
        {
            "identifier":"test_log_core_additional.log_msg_create",
            "execution_time":"0.20",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_generic",
            "execution_time":"0.20",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_thread_trigger",
            "execution_time":"0.00",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_thread",
            "execution_time":"2.00",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.multiple_backends",
            "execution_time":"0.00",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_timestamping",
            "execution_time":"0.20",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_severity",
            "execution_time":"0.20",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_early_logging",
            "execution_time":"0.20",
            "status":"passed"
        },
        {
            "identifier":"test_log_core_additional.log_sync",
            "execution_time":"0.00",
            "status":"skipped",
            "reason":"ztest skip"
        },
        {
            "identifier":"test_log_core_additional.log_domain_id",
[... 91 lines cut ...]
2025-03-25 03:25:25,903 - twister - DEBUG - DEVICE: .201 seconds
```
The correct output should seems that:
```
2025-03-25 03:25:25,805 - twister - DEBUG - DEVICE: [00:00:00.701,000] <inf> log_test: info message for domain id test
2025-03-25 03:25:25,903 - twister - DEBUG - DEVICE: PASS - test_log_domain_id in 0.201 seconds
```

**To Reproduce**

have one of above platform connected
go to your zephyr directory and setup env
call ./scripts/twister -T tests/subsys/logging/log_core_additional -p intel_adsp/cavs25 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
See console output with test suite passed and all test cases passed but in the twister.json file the test log_domain_id is failed.

**Expected behavior**

**Impact**

twister.json file - reporting feature in twister

**Environment (please complete the following information):**

 - OS: Linux 24
 - Zephyr version: v4.1.0-1193-gbc5a60c812ba

arikgreen, 2025-03-25:
> > call ./scripts/twister -T tests/subsys/logging/log_core_additional -p intel_adsp/cavs25 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
> 
> that does not make sense. cavs25 does not have ttyACM0

sorry, I forgot change this parameter. I changed in description.

arikgreen, 2025-03-26:
@nashif I appended to the description the logs from up2 and ace15_mtpm

golowanow, 2025-03-26:
I think there is at least 2 problems here to investigate and resolve

1)  `log_domain_id` test `LOG_INF` output messed with Ztest result and to me it looks actually as a failure for this test case of logging subsystem.

https://github.com/zephyrproject-rtos/zephyr/blob/ca0f5f6924cdc7046ce8dbb6d32e39125b5ded7f/tests/subsys/logging/log_core_additional/src/log_test.c#L219-L222

```
START - test_log_domain_id
 PASS - test_log_domain_id in 0[00:00:00.701,000] <inf> log_test: info message for domain id test
.201 seconds
```

2) Twister Ztest harness in this situation has no match on the test case result (pattern not match on the corrupted line), but the suite summary entry should also update test case status as passed. For some reason it wasn't done, or changed to failed.
It would be helpful to repeat this situation with additional trace logs running twister with `-vvv`

arikgreen, 2025-03-28:
in this case, always are mixed the log message from line 219 and the summary for this test case
https://github.com/zephyrproject-rtos/zephyr/blob/ca0f5f6924cdc7046ce8dbb6d32e39125b5ded7f/tests/subsys/logging/log_core_additional/src/log_test.c#L212-L226

There is the simple workaround for it:
add 250ms sleep between LOG_INF() and zassert_equal()
After that LOG_INF return right message before the test case return summary message:
```
2025-03-28 14:09:15,613 - twister - DEBUG - DEVICE: ===================================================================
2025-03-28 14:09:15,712 - twister - DEBUG - DEVICE: START - test_log_domain_id
2025-03-28 14:09:15,712 - twister - DEBUG - TC_START: Ztest case 'log_domain_id' matched to 'logging.thread.test_log_core_additional.log_domain_id
2025-03-28 14:09:15,835 - twister - DEBUG - DEVICE: [00:00:00.701,000] <inf> log_test: info message for domain id test
2025-03-28 14:09:15,936 - twister - DEBUG - DEVICE: PASS - test_log_domain_id in 0.251 seconds
2025-03-28 14:09:15,936 - twister - DEBUG - TC_END: Ztest case 'log_domain_id' matched to 'logging.thread.test_log_core_additional.log_domain_id
2025-03-28 14:09:15,936 - twister - DEBUG - TC_END: END case 'logging.thread.test_log_core_additional.log_domain_id':{'count': 1}
2025-03-28 14:09:15,937 - twister - DEBUG - DEVICE: ===================================================================
```

@golowanow 
> Twister Ztest harness in this situation has no match on the test case result (pattern not match on the corrupted line), but the suite summary entry should also update test case status as passed. For some reason it wasn't done, or changed to failed.
It would be helpful to repeat this situation with additional trace logs running twister with -vvv

In the harness Test a status of test case is changed to Failed because regex pattern test_case_end_pattern not match line in a log.
https://github.com/zephyrproject-rtos/zephyr/blob/ca0f5f6924cdc7046ce8dbb6d32e39125b5ded7f/scripts/pylib/twister/twisterlib/harness.py#L909-L926

nashif, 2025-03-30:
@golowanow assigning to you since you are looking at this already.

golowanow, 2025-04-01:
> 1. `log_domain_id` test `LOG_INF` output messed with Ztest result and to me it looks actually as a failure for this test case of logging subsystem.

https://github.com/zephyrproject-rtos/zephyr/pull/87952

> 2. Twister Ztest harness in this situation has no match on the test case result (pattern not match on the corrupted line), but the suite summary entry should also update test case status as passed. For some reason it wasn't done, or changed to failed.

https://github.com/zephyrproject-rtos/zephyr/pull/87971

golowanow, 2025-04-04:
not yet completed waiting for #87971

PR #87952 (2025-04-03): tests: logging: logging.thread: fix Ztest log corruption
Fix race condition and Ztest log corruptions in `tests/subsys/logging/log_core_additional` test suite's `logging.thread` test configuration.

Fixes #87625 (partially, for this test suite).

For example, before:
```
twister - DEBUG - DEVICE: PASS - test_log_domain_id in 0[00:00:00.701,000] <inf> log_test: info message for domain id test
twister - DEBUG - DEVICE: .201 seconds
...
twister - DEBUG - DEVICE: Activate backend with context PASS - test_log_early_logging in 0.001 seconds
```
 after
```
twister - DEBUG - DEVICE: START - test_log_d [... cut]