zephyrproject-rtos/zephyr #87671: spi_nxp_lpspi: Regression in nxp_spi_lpspi causing MCP2518FD CAN Device Not Ready

asingh-GiN, 2025-03-26:
**Describe the bug**
After a recent change in the LPSPI driver, the MCP2518FD CAN device (connected to lpspi1) is no longer detected as ready. The CAN test sample worked perfectly in previous versions but is now broken.
Nothing was changed in [can_mcp251xfd](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/can/can_mcp251xfd.c) driver as well.

 - What target platform are you using?
 NXP IMXRT1170 EVK 
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 [d67e705](https://github.com/zephyrproject-rtos/zephyr/commit/d67e705d0b6181948f8f8dc9c0d31bc38a02f638)

git bisect result: 
d67e705d0b6181948f8f8dc9c0d31bc38a02f638 is the first bad commit
commit d67e705d0b6181948f8f8dc9c0d31bc38a02f638
Author: Declan Snyder <declan.snyder@nxp.com>
Date:   Thu Feb 6 10:05:15 2025 -0600

    spi_nxp_lpspi: Fix DMA Async API

    Fix the ASYNC DMA API on the lpspi driver and actually make the entire
    driver go through that path. Rather than having an orthogonal
    internally synchronous path we can just have both APIs go through the
    same asynchronous path and just use wait_for_completion from spi context
    to implement either sync or async.

    Also make DMA driver default y if dependency (an lpspi having dmas
    property) is met.

    And lpspi_wait_tx_fifo_empty can be shared between drivers.

    Signed-off-by: Declan Snyder <declan.snyder@nxp.com>

 drivers/spi/spi_nxp_lpspi/Kconfig                |   1 +
 drivers/spi/spi_nxp_lpspi/spi_nxp_lpspi.c        |   8 -
 drivers/spi/spi_nxp_lpspi/spi_nxp_lpspi_common.c |   8 +
 drivers/spi/spi_nxp_lpspi/spi_nxp_lpspi_dma.c    | 347 ++++++++---------------
 drivers/spi/spi_nxp_lpspi/spi_nxp_lpspi_priv.h   |   2 +

**To Reproduce**
git bisect good df5318479b914d320f52870c6dfd0420779fc5c0
git bisect bad b80526658cc0e6ec5719f09cd8fff65d4e50525c

Steps to reproduce the behavior:
`west build -p -b mimxrt1170_evk@A//cm7 ../zephyr/samples/drivers/can/counter -DEXTRA_DTC_OVERLAY_FILE=custom/mimxrt1170_evk_cm7_A.overlay`

Extra overlay file since I connected MCP2518fd CAN controller to LPSPI:
```
/delete-node/ &flexcan3;
/ {
	chosen {
		zephyr,canbus = &can_controller;
	};
};

&lpspi1 {
	pinctrl-0 = <&pinmux_lpspi1>;
	pinctrl-names = "default";
	status = "okay";
	can_controller: mcp2518fd@0 {
		compatible = "microchip,mcp251xfd";
		reg = <0x0>;
		bitrate = <500000>;
		sample-point = <875>;
		bitrate-data = <500000>;
		sample-point-data = <875>;
		osc-freq = <40000000>;
		spi-max-frequency = <20000000>;
		int-gpios = <&gpio3 1 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
		status = "okay";
	};
};

```
**Expected behavior**

CAN device is ready and we should get:
```
*** Booting Zephyr OS build v4.1.0-269-gdf5318479b91 ***
Change LED filter ID: 0
Finished init.
Counter filter id: 1

uart:~$ Counter received: 0
Counter received: 1
Counter received: 2
```

**Impact**
Very important for our application

**Logs and console output**
Current error after the new updates in the lpspi: 
```
*** Booting Zephyr OS build v4.1.0-675-gb80526658cc0 ***
CAN: Device mcp2518fd@0 not ready.

uart:~$

```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA while testing git bisect: b80526658cc0e6ec5719f09cd8fff65d4e50525c

**Additional context**
During debugging following was found: 
It fails during init and particularly in mcp251xfd_set_mode_internal:
```
reg = mcp251xfd_read_crc(dev, MCP251XFD_REG_CON, MCP251XFD_REG_SIZE)
	if (!reg) {
		ret = -EINVAL;
		goto done;
	}
```
 Error code is -EINVAL

This was not the problem with the last version described above. Nothing was changed in [can_mcp251xfd](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/can/can_mcp251xfd.c) driver as well.

henrikbrixandersen, 2025-03-26:
CC: @decsny

decsny, 2025-03-27:
> During debugging following was found: It fails during init and particularly in mcp251xfd_set_mode_internal:
> 
> ```
> reg = mcp251xfd_read_crc(dev, MCP251XFD_REG_CON, MCP251XFD_REG_SIZE)
> 	if (!reg) {
> 		ret = -EINVAL;
> 		goto done;
> 	}
> ```
> 
> Error code is -EINVAL

Is the error code from mcxp251xd_read_crc / spi_transceive actually -EINVAL or just it gets set to that here. Because if there would be any error code from that commit , I would expect it to cause -EIO

asingh-GiN, 2025-03-27:
> > > > During debugging following was found: It fails during init and particularly in mcp251xfd_set_mode_internal:
> > > > ```
> > > > reg = mcp251xfd_read_crc(dev, MCP251XFD_REG_CON, MCP251XFD_REG_SIZE)
> > > > 	if (!reg) {
> > > > 		ret = -EINVAL;
> > > > 		goto done;
> > > > 	}
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > Error code is -EINVAL
> > > 
> > > 
> > > Is the error code from mcxp251xd_read_crc / spi_transceive actually -EINVAL or just it gets set to that here. Because if there would be any error code from that commit , I would expect it to cause -EIO
> > 
> > 
> > The `mcp251xfd_read_crc` returns `NULL`. So the reg is `0`therefore the `ret` value is set to `-EINVAL`. I get this debug error log: `<err> can_mcp251xfd: Failed to reset the device [-22]`
> 
> what does the spi_transceive return though? can you debug a bit further to see where initial error code is set ?

`spi_transceive_dt` returns 0 but the `crc` is not 0 in `mcp251xfd_read_crc` tahts why it retries a couple of times and if the `crc` is still not zero `mcp251xfd_read_crc` returns `NULL`.

decsny, 2025-03-27:
I am actually finding a lot of regressions right now on the DMA based driver, that were not there when I was working on that PR, I think the recent testing change that I did exposed some issues, it passed the previous version of the spi_loopback test but many platforms are failing the new version of the spi_loopback test only when using the updated DMA driver in the commit you bisected to

decsny, 2025-03-28:
@asingh-GiN  can you try with this patchset in https://github.com/zephyrproject-rtos/zephyr/pull/87789 and see if they fix this issue

decsny, 2025-04-08:
@asingh-GiN can you try with current main, there was some major bug fixes lately fixing some other regressions, might be same root cause

asingh-GiN, 2025-04-08:
Hello @decsny I tested the sample with the main, however, it doesnt work. I still get the same error:
```
uart:~$ *** Booting Zephyr OS build 4925e22a6798 ***
CAN: Device mcp2518fd@0 not ready.
```

decsny, 2025-06-06:
Can you please provide some signal trace to show what is different on the bus, I don't have this CAN device to test with

decsny, 2025-07-01:
@asingh-GiN please try latest main again, there was some fixing of an issue a lot of people were encountering that was actually causing too many bytes sent by the master, could be your issue