zephyrproject-rtos/zephyr #87694: Quarantine usage in Zephyr

PerMac, 2025-03-26:
## Introduction

This RFC proposes the implementation of a quarantine mechanism for the Zephyr project to manage and isolate problematic tests and configurations. The goal is to allow development on the main branch to continue regardless of regressions that would otherwise fail the CI. This approach ensures that development is not blocked by unrelated test failures, maintaining a stable and productive development environment. Additionally, it provides a single source of truth for quarantined tests, making it easier to track, manage, and resolve issues.

### Problem description

Executing all tests on all platforms for every PR is not feasible, hence there is always a risk of regression leaking into the main branch. Such regressions can block development on other PRs due to unrelated failing tests. When such situations occur, prompt action is needed. Current possibilities include:

- Reverting the offending commit (can break developers' work, not always trivial).
- Waiting for a hot fix (can get delayed).
- Adding a workaround to force Twister to skip failing tests.

The last action is the most problematic, as there are various ways to achieve such skipping but no unified way to control those skipped tests. Additionally, due to the pressure of unlocking the main branch, rushed actions can be made, making the issue worse.

**Examples**
**Twister forced to skip**
`find . -name "*.yaml" -exec grep -l "FIXME" {} \;`
18 cases in 14 files

File: [tests/kernel/fatal/exception/testcase.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/fatal/exception/testcase.yaml#L47])
Issue: platform_exclude: qemu_cortex_a9 # FIXME: See issue #39948 [Oct, 2021]
Status: Seems to be passing now.

File: [tests/net/socket/poll/testcase.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/socket/poll/testcase.yaml#L3)
Issue: platform_exclude: mps2/an385 # FIXME: This test fails very frequently on mps2/an385 due to system timer stability issues. [Aug, 2022]
Status: Fixed with PR #63344

File: [samples/drivers/ipm/ipm_esp32/sample.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/drivers/ipm/ipm_esp32/sample.yaml)
Issue: `skip: true`
Reason: Disabled with merging hwmv2 (GitHub cannot even show the change among commit diff)

**Recent Example of "failure of a day"**
Issue: [tests/drivers/wifi/nrf_wifi/nrf70.build.offloaded_raw_tx](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/drivers/wifi/nrf_wifi/testcase.yaml) was failing on nrf7002dk/nrf5340/cpuapp 
https://github.com/zephyrproject-rtos/zephyr/issues/87079
Marked as “priority:high” due to CI breaking even though it it was a "priority: low" type of a bug (very isolated issue, single test valid only for a single platform)
Attempts to Revert: PR #87016
Summary: Benjamins comment from #87016 "When something breaks CI we try to implement the fix ASAP, even if that means short-term it's in the form of a revert. Here it dragged all day with various fixes in multiple PRs, which is really not ideal as many pull requests were stuck meanwhile and impacting many people in the project, while we had an "easy" fix at hand. Something to keep in mind for next time please :)" 
The mentioned "easy fix" was to revert, however, IMO quarantining the test could also work. 

### Proposed change
Implement a structured quarantine process to isolate problematic tests and configurations, allowing for a more controlled and transparent approach to managing test failures. This process will provide a transparent and version controlled (part of zephyr tree) single source of truth for quarantined tests, making it easier to track, manage, and resolve issues. It will be based on twister's [quarantine](https://docs.zephyrproject.org/latest/develop/test/twister.html#quarantine) mechanism. Release managers and affected areas maintainers will be the key stakeholders in the process.

Please note that the goal is not to "sweep failures under the rug" and is applicable to a certain classes of issues where the key is that an issue is isolated and not affecting other components.

## Detailed RFC

The idea is to use `--quarantine-list` flag for in-CI twister calls and use a quarantine file (e.g. located at `scripts/ci/quarantine.yaml`) as an input telling which configurations shall be skipped. Such workflow can be

- Maintainable and reliable: Thanks to a "single source of truth" within a zephyr tree (version controlled). A single quarantine file will be used to manage all quarantined tests and configurations affecting the CI. Each entry in the quarantine file will include comments explaining the reason for quarantine (link to a github issue). A weekly workflow with "--quarantine-verify" flag can be added to find tests that are no longer failing. 
- Scalable and flexible: The quarantine mechanism supports combination of regexes to specify scenarios, platforms, and simulator types (e.g. all scenarios on platformX, or all  drivers.flash.common.* on nrf.* platforms).
- Robust: Quarantine changes will be verified before being merged, automatically for the file format and manually by release managers for validity of the changes.
- Modular: the mechanism allows to use >1 quarantine files. E.g. an extra file "quarantine_hardware.yaml" can be maintained that can be used in addition to store tests that fails on-target tests at vendors site. 

### Proposed change (Detailed)
**Addition of a quarantine file to zephyr tree (e.g. scripts/ci/quarantine.yaml**
This file will provide a "single source of truth" to manage all quarantined tests and configurations. Each entry in the quarantine file must include a link to a reported github issue.

**Quarantine Process**
- When regression on main is detected a quarantine PR can be created (as an alternative to revert or waiting for hotfix), modifying the quarantine.yaml file 
- Each quarantine entry must have a corresponding GitHub issue. 
- Maintainers of the affected area will be notified: gh ping and/or added as reviewers.
- The PR must be accepted by Release Manager(s).

**Fix and Lift Quarantine**
When a fix is ready, the quarantine entry should be removed in the same PR as the fix. This allows for immediate evaluation of the fix within the PR scope.

**Weekly Evaluation**
The quarantine file will be evaluated weekly by twister call with `--quarantine-verify` flag to catch already fixed issues, which were not cleared within a PR with a fix.

**Release Management**
[... 6 lines cut ...]
### Concerns and Unresolved Questions

I copied here an email from @nashif on a testing wg mailing list. My responses are in _italic_:
"As was discussed few years ago when this topic came up, I am fine using quarantine mechanism for a defined class of issues, mostly long term and well-documented issues on certain platforms, toolchains etc. 
Using this mechanism to “unblock” daily development when we have breakage in the tree is a problem IMO. Reverting a change causing breakage should be considered the best option if no immediate fixes are possible. Some examples:
-	If we start quarantining issues as the happen, we will end up with undefined state, i.e. I might quarantine something now, another PR might be submitted that introduces more issues in the same area that the original PR was failing on, you fix the first. Issue, and something else might start failing and so on.

_I agree. The risk you mentioned is real. Deciding if quarantining a given configuration(s) is worth the risk requires some understanding of the context. I proposed a quarantine for cases where it is clear that the impact is isolated (vide nrf_wifi example). Also, as history shown, we already were doing such forced skips of broken configurations in twister, hence the need of such functionality. With proposed quarantine process there will be more control over such skips._

-	The process of doing this and deciding if something should be quarantined or not should not be done on a whim and because some PR starts failing. If a PR fails because of unrelated tests, people need to be patient, and we need to figure out how to localize and target changes so that we do not end up failing on platform X when all I am doing is changing platform Y on completely unrelated changes. 

_This is why I propose that release managers and affected area maintainers should be the key stakeholders in the process. I also agree that improving our dynamic scope selection algorithms should happen whenever their flaws are exposed. Neverth [... cut ...] d" on repeating such failures just adds up. Also, from my experience with extensive on-target runs, it is way easier to maintain such runs, when I only have to look at not yet reported failures instead of having to figure each time what's new and what was already reported._
"

## Alternatives

Keep using skip:true,  #fixme exclude, etc. workarounds to unlock failing CI when prompt hotfix nor revert is possible. I think we all agree that using qquarantine for those cases would only be beneficial.

aescolar, 2025-03-27:
Thanks @permac
Overall I'm ok with having a quarantine file instead of sprinkling provisional skips or excludes in yaml files.

A few thoughts:
* We should be very careful with this feature. And it should be a last resort. I share @nashif's concerns here.
* Please note that many of us run twister locally, including for bisecting. The existance of this file should be very clearly understood by all.
  i.e. I think the file would be better be placed in the top level than under scripts/ci. And maybe twister should default
  to run with the quarantine file that is placed in ${ZEPHYR_BASE}/
* Today a test yaml change is linked to the area it tests and therefore assigned and reviewed by the area maintainer and collaborators. The quarantine file would start decoupled from the areas. You propose some extra scripting to handle the quarantine file in particular and solve this. Fair enough, but that is work to be done.
* We should make sure the quarantine file is only used temporarily for issues to be fixed, and not as a filter of tests that could never be run in some platforms.

About process:
You seem to be mixing the "release managers", "members of the release engineering team", and misunderstanding a bit their roles:
"Release managers" are just the 2 members of the release engineering team who focus on the current release. But all release engineers share most of the responsibilities.
In any case they do not override maintainers. The responsibility to validate that something should be quarantined should be in the maintainers of the affected areas, and over them on the TSC. Release engineers should not decide to quarantine. They should revert if no fix can be found fast. Disabling a test is technical debt or a cover for it, and a release engineer should not decide to do that. Release engineers could, of course, reject an addition to the quarantine file, but so could anybody else.

> Leaving entries in quarantine in a released version of Zephyr requires RM approval..

If anything that would be up to the TSC.
In any case, an entry in the quarantine file would be another view of a bug (which should be already tracked in GitHub and prioritized). So as such, I'm not too sure if adding more process around this subset of bugs would be the best way.

In any case, details about the process around this should probably be discussed in the release engineering meeting after agreeing on the principle of using a quarantine file.

nashif, 2025-03-27:
> I agree. The risk you mentioned is real. Deciding if quarantining a given configuration(s) is worth the risk requires some understanding of the context. I proposed a quarantine for cases where it is clear that the impact is isolated (vide nrf_wifi example).

every other change to zephyr can be considered "isolated", there is nothing special about the nrf_wifi, really. I can submit a change to a board, which breaks CI, it is broken only with this board, so should we just keep it in and quarantine it? Why? At this point of time we KNOW what caused the breakage, i.e. we have the symptom and the cause, if we can fix it quickly, go for it, if not, revert. Keeping the "cause" in the tree just to "unblock development"  (btw, I do not agree that anyone will be blocked) is calling for trouble as you accumulate "causes" and soon enough those will start crossing each other and you will not know anymore how to deal with problem.

>  Also, as history shown, we already were doing such forced skips of broken configurations in twister, hence the need of such functionality. With proposed quarantine process there will be more control over such skips.

You are mixing two things. I agree we could do better managing long term exclusions and workarounds, but many of those workarounds are there for tests and platforms that are known to have issues, not because the code is wrong in the tree, but because of target., i.e. qemu etc.

Others are just bad tests, so the implementation is fine, the test itself is flaky or not perfect or whatever, so we get to skip them.

What you are proposing is not about flaky or long term issues, you want to keep issues in the tree and disable testing using the quarantine option. So, we are keeping the problem instead of fixing it. 

To get to closure and given that the major part of the proposal involves release engineering, I will be happy to put you on the agenda in the releae engineering meeto for you to present your proposal and have a discussion.

nashif, 2025-03-27:
oh btw, there is a PR addressing many of the long term skips and fixmes: https://github.com/zephyrproject-rtos/zephyr/pull/87709

PerMac, 2025-03-27:
Thanks a lot for the feedback!
@aescolar You're right I misunderstood the roles of "release managers" (and as pointed rather release engineers) and overestimated their responsibilities. I assumed that since "release engineers" have rights to revert breaking changes they might also be the ones to quarantine configurations in some cases, instead of reverting. However, I see your point, that revert is safer than quarantine. I also understand better now that deciding to quarantine something steps into the maintainer's area of responsibility. 

@nashif Yes, quarantining means leaving a bug in the tree, and yes it should be a temporary solution. I think I start getting a better understanding of your point. In my view both a bug in the tested code and a bug in a test or test's target are bugs in the tree that need fixing. However, I understand that there is a difference in the risk of leaving those in the tree. So it's all about risk vs benefits management. I see from your comments, that the benefits (unlocking development on main) might not be worth of the risk. Or that even there might not be benefits ("btw, I do not agree that anyone will be blocked").

Thank you for pointing to the release engineering meeting. They (you) are the main audience for the proposal and discussing this with the team is the key. You have way more experience with the releases and better understanding of what is and what is not a real problem. The idea is to help the team and the project and not to put more burden and unjustified risk.

nashif, 2025-03-28:
@PerMac One significant and major difference between the issues you list above and https://github.com/zephyrproject-rtos/zephyr/issues/87079 is the fact that this "failiure of the day" is a build failure, meaning that keeping it in tree and quarantining the test that exposes the build issue will mean that anyone who tries to use and build this feature will have a build failure, so it is not a test failing to execute, it is code that does not build that will be kept in tree instead of being reverted to "unblock development", yet blocking everyone who tries to do anything with the code.

keith-zephyr, 2025-04-30:
Process WG agreed to move forward to minimal implementation.  Please re-add the Process tag if more discussion is needed.

closed 2026-09-26 by github-actions[bot]