zephyrproject-rtos/zephyr #87698: drivers:stepper Step-Dir Pin toggle Frequency to High

jbehrensnx, 2025-03-26:
**Describe the bug**

The current step-dir implementation in `step_dir_stepper_common` toggles in single_edge mode the step pin twice as fast as it can. For some board and stepper controller combinations (for example Teensy 4.0 board with a TI drv8424 stepper controller) this causes the pin to be toggled to fast and thus erroneous behavior - e.g. the motor not stepping.

**To Reproduce**

1. Connect a `drv8424` stepper controller with a `Teensy 4.0` board and use the counter timing source (didn't test work queue timing)
2. Test the driver via the shell, using microstep resolution of `1` and a microstep interval of `5000000` - i.e. 200 steps/s
3. Observe the motor

**Expected behavior**

The motor should step correctly.

**Possible solutions**

One possible solution is to use a 50% duty cycle of the step signal - like the drv8424 driver before the step-dir generalization. This requires an internal doubling of the step signal frequency and step counting only on every second interrupt. In addition, the step signal needs to be explicitly set to low when stopping, possibly after a short delay after stopping the timing source. 
These changes are only needed for single_edge mode. dual_step mode can remain like it is.

That said, this is only one possible solution and I am interested if there are other ideas on how to solve this.

jilaypandya, 2025-03-26:
Ping @faxe1008

faxe1008, 2025-03-28:
@jbehrensnx So the issue is that your Teensy toggles too fast and the drv8424 is not fast enough to detect the edges?
Datasheet:
[image]

Thinking about it - maybe it would make sense to change the step-dir impl generally so that for each timing source event we toggle the pin once, but depending on `dual_edge` we change the timing source frequency? This should space out the toggeling more and resolve the issue.

>  In addition, the step signal needs to be explicitly set to low when stopping, possibly after a short delay after stopping the timing source.

Can you explain why?

jbehrensnx, 2025-03-28:
> So the issue is that your Teensy toggles too fast and the drv8424 is not fast enough to detect the edges?

Yes. The NO. 2 timing of of 970ns is the issue. On the Nucleo F767ZI the _pulse duration step high_ is ca 800ns, which while below 970 seems to be enough. On the Teensy however, _pulse duration step high_ is only about 60ns - way to short.

> Thinking about it - maybe it would make sense to change the step-dir impl generally so that for each timing source event we toggle the pin once, but depending on `dual_edge` we change the timing source frequency? This should space out the toggeling more and resolve the issue.

That would be my solution as well. We will also have to adjust step counting as well.

> > In addition, the step signal needs to be explicitly set to low when stopping, possibly after a short delay after stopping the timing source.

> Can you explain why?

We run the risk of loosing steps otherwise. Without this after stopping the step signal might remain on high. The first toggle of a new movement would then set the pin to low, not doing a step despite us expecting to and adjusting the step count. Deliberately setting the pin to low would alleviate this. Note that this is only needed for `stop` and `disable`, regular movement should stop correctly on low.
Possibly delaying this is because at extremely high frequencies an additional timing source event might trigger before the timing source stop takes effect and with a delay. We would guarantee, that the step signal would always be low. That said, I dont know how realistic the case requiring the delay is. If we do incorporate the delay, it would most likely need to be implemented via workqueue, as the stop and disable events should be ISR safe to allow stopping from interrupt context.

dolence, 2025-04-04:
More robust generic drivers like the ones used to drive bigger motors need wider pulse length. The driver I'm using requires 2 us wide pulses to work. Also, enable need to be active for at least 5 us before start generating pulses. Perhaps a way to adjust the pulse length would be a nice fit for both issues.

[image]

jbehrensnx, 2025-06-30:
@jilaypandya I disagree. All step-dir drivers that I know of expect a 50% duty cycle, which is why that should be the default.

jilaypandya, 2025-06-30:
> I disagree.

I don't know what are you disagreeing about :). 

I mean atleast the the snippets of data sheets posted here mention pulse duration and not 50% duty cycle, which is how it should be. Anyhow do you have a patch, if you think it's a bug?

jbehrensnx, 2025-06-30:
The various datasheets that I have read often do mention minimum timings and max frequencies, but they also are written from a perspective of having 50% duty cycle (if that makes sense). Also, other step-dir implementations outside Zephyr also use 50% duty cycle.

My disagreement was about this not being a bug. It is. The gpio pins are toggled to fast for the stepper driver.

jbehrensnx, 2025-06-30:
I currently have no patch. faxe1008 wanted to patch it, but closed it because of time issues. I played around with a workaround, but that was nothing serious, and it is currently a low priority for me.
The followup pr to #91979 might be a good place to attempt to fix this.

The solution is, in theory, relatively simple. For non-dual edge mode the frequency of the timing source is doubled and the position is only updated on every second call (through it should be on the low->high transition). The second pin toggle is also removed. However, there was some issue with improper step counting.

jilaypandya, 2025-06-30:
> I currently have no patch. faxe1008 wanted to patch it, but closed it because of time issues. I played around with a workaround, but that was nothing serious, and it is currently a low priority for [...]

Fair enough, Since it's not a high priority for you anyhow, we will tackle this after #91979 :)  There is a lot happening in that PR already :)

jbehrensnx, 2025-08-06:
I currently do not have the time to test your proposal myself, so I can not comment on anything one could notice when running/testing this change. I can only comment on what I noticed in the source code.

It should work most of the time (with one exception, see below), but there are some issues resulting from edge cases:

`zephyr_stepper_motion_controller.c/velocity_mode_task` is missing a reset of `actual_pulse_per_step`

`zephyr_stepper_motion_controller.c/z_stepper_motion_control_stop` should, when **not** using `dual-edge` set the step pin to low to ensure the system is in a consistent state. Admittedly this could mean loosing a step in extreme edge cases (stopping so quickly after transitioning the step pin to high that we are below the minimum step-high time of the stepper-driver).
One solution  for this could be to change how `stop` works: instead of stopping immediately, it sets a flag that causes the stepper to stop once the current step is completed, at the cost of potentially performing an additional step after calling `stop`.

`zephyr_stepper_motion_controller.c/z_stepper_motion_control_set_step_interval` has a similar issue but here the solution would be to simply buffer the step interval, i.e. while one can set a new step-interval at any time, while the stepper is moving a new step interval takes only effect once the current step is complete. Stm timers work in a similar way by default for example.

I don't like the name `pulse_per_step` and have trouble understanding what the name is meant to say. If we want to be more accurate, then dual-edge has one transition per step while non-dual-edge has two transitions per step (with position/step increment update occurring after the first). That said, because there are only these two possibilities, I would be fine with continuing to use the dual-edge-step boolean variable and flipping a boolean in the `_task` functions.

A lot of this comes from realizations I had when conceptualizing an acceleration stepper controller using the stm timers, which is why it did not come up until now.

jilaypandya, 2025-08-14:
> Thinking about it - maybe it would make sense to change the step-dir impl generally so that for each timing source event we toggle the pin once, but depending on `dual_edge` we change the timing source frequency? This should space out the toggeling more and resolve the issue.
 
Datasheet TMC2209:
[image]

Yep, we will need to create timing_source_events.

```
enum stepper_timing_source_events {
    STARTED,                    (toggle step pin)
    MIN_HIGH_TIME_ELAPSED,      (count a step, toggle step pin)
    MIN_LOW_TIME_ELAPSED,       (count a step if dual-edge)
    MICROSTEP_INTERVAL_ELAPSED, (toggle step pin)
    STOPPED,                    (toggle step pin if not dual edge)
}
```
On top of that we would need to add these timings `tsl, tsh` in device tree and check that the user is not entering microstep_interval less than `200ns for tmc2209` or `1940ns for drv8424`. Ping @dolence

closed 2025-09-30 by jhedberg