zephyrproject-rtos/zephyr #87705: Assembler warnings not properly promoted to errors when compiling source files with CONFIG_COMPILER_WARNINGS_AS_ERRORS

palchak-google, 2025-03-26:
**Describe the bug**
When building with `CONFIG_COMPILER_WARNINGS_AS_ERRORS=y`, warnings generated during a build should be treated as errors. However, when compiling source (C or C++) files with a gcc toolchain, warnings generated by the (GNU) assembler are not properly promoted to warnings. 

**Logs and console output**
Here is the output when compiling a C++ source file that produces assembler warnings. This operation should fail, but it incorrectly succeeds. You can see in the compiler flags that the `-Werror` flag is being passed as evidence that the `CONFIG_COMPILER_WARNINGS_AS_ERRORS=y` was set.

```
❯❯❯ ninja -v modules/boosty/CMakeFiles/libboosty.dir/platform/concurrency_test.cc.obj   
[1/1] /usr/bin/g++ -DCATCH_CONFIG_PREFIX_ALL -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DLOG_LEVEL_MAX=LOG_LEVEL_INFO -DLZ4_HEAPMODE=0 -DLZ4_MEMORY_USAGE=11 -DMCUBOOT_HAVE_ASSERT_H -DPB_BUFFER_ONLY -D__ZEPHYR__=1 -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/src -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/modules/boosty -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/zephyr/include/generated/zephyr -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/zephyr/include/generated -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/soc/native/inf_clock -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/boards/native/native_sim -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/scripts/native_simulator/common/src/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/scripts/native_simulator/native/src/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/nrf/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/nrf/tests/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/bootloader/mcuboot/boot/zephyr/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/bootloader/mcuboot/boot/bootutil/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/catch2/src/catch2/.. -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/modules/boosty/third_party/catch2/generated-includes -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nanopb/.. -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/lz4/build/cmake/../../lib -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nanopb -fno-strict-aliasing -Werror -O2 -fcheck-new -std=c++2b -Wno-register -Wno-volatile -fno-exceptions -fno-rtti -imacros /usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/zephyr/include/generated/zephyr/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests=CMAKE_SOURCE_DIR -fmacro-prefix-map=/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/usr/local/google/home/palchak/projects/pointy/teagarden=WEST_TOPDIR -ffunction-sections -fdata-sections -m32 -msse2 -mfpmath=sse -fvisibility=hidden -fno-freestanding -fmax-errors=20 -fdiagnostics-color -fdiagnostics-show-option -fno-strict-overflow -fmacro-prefix-map=../src/= -fmacro-prefix-map=/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/= -fmacro-prefix-map=/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/= -fconcepts-diagnostics-depth=3 -ftemplate-depth=64 -Wextra -Wundef -Wshadow -Wconversion -Wsign-conversion -Wimplicit-fallthrough -Wno-comment -Wsuggest-override -Wplacement-new=2 -Wformat=2 -Wswitch-enum -Wwrite-strings -Winit-self -Wlogical-op -Wno-sign-conversion -Wno-psabi -Wno-undef -Wno-conversion -MD -MT modules/boosty/CMakeFiles/libboosty.dir/platform/concurrency_test.cc.obj -MF modules/boosty/CMakeFiles/libboosty.dir/platform/concurrency_test.cc.obj.d -o modules/boosty/CMakeFiles/libboosty.dir/platform/concurrency_test.cc.obj -c /usr/local/google/home/palchak/projects/pointy/teagarden/firmware/platform/concurrency_test.cc
/tmp/cc5xlf9T.s: Assembler messages:
/tmp/cc5xlf9T.s:13599: Warning: setting incorrect section type for .noinit."platform/common/zephyr/work_queue.hh".0
/tmp/cc5xlf9T.s:13636: Warning: ignoring incorrect section type for .noinit."platform/common/zephyr/work_queue.hh".0
/tmp/cc5xlf9T.s:13636: Warning: ignoring changed section type for .noinit."platform/common/zephyr/work_queue.hh".0
/tmp/cc5xlf9T.s:13636: Warning: ignoring changed section attributes for .noinit."platform/common/zephyr/work_queue.hh".0
```

**Expected behavior**
This is what the output for the same compilation step should look like:
```
❯❯❯ ninja -v modules/boosty/CMakeFiles/libboosty.dir/platform/concurrency_test.cc.obj
[1/1] /usr/bin/g++ -DCATCH_CONFIG_PREFIX_ALL -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DLOG_LEVEL_MAX=LOG_LEVEL_INFO -DLZ4_HEAPMODE=0 -DLZ4_MEMORY_USAGE=11 -DMCUBOOT_HAVE_ASSERT_H -DPB_BUFFER_ONLY -D__ZEPHYR__=1 -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/src -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/modules/boosty -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/zephyr/include/generated/zephyr -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/zephyr/include/generated -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/soc/native/inf_clock -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/boards/native/native_sim -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/scripts/native_simulator/common/src/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/zephyr/scripts/native_simulator/native/src/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/nrf/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/nrf/tests/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/bootloader/mcuboot/boot/zephyr/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nordic/connect_sdk/bootloader/mcuboot/boot/bootutil/include -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/catch2/src/catch2/.. -I/usr/local/google/home/palchak/projects/pointy/teagarden/firmware/app/libboosty_tests/BUILD.west/libboosty_tests/modules/boosty/third_party/catch2/generated-includes -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nanopb/.. -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/lz4/build/cmake/../../lib -I/usr/local/google/home/palchak/projects/pointy/teagarden/third_party/nanopb -fno-strict-aliasing -Werror -Wa,--fatal-warnings -O2 -fcheck-new -std=c++2b -Wno-register -Wno-volatile -fno-exceptions -fno-rtti -imacros /usr/local/google/home/palchak [... cut ...]
**Impact**

**Environment (please complete the following information):**
This a build targeting the `native_sim` platform on Linux using gcc v14.2.0

**Additional context**
The fix for this issue is to pass the compile flag `-Wa,--fatal-warnings` when building C or C++ source files (with gcc, other compilers will vary). I think the change needs to be made here:
https://github.com/zephyrproject-rtos/zephyr/blob/d02bc50375aea4108f7747f28b851d82158c02da/cmake/compiler/gcc/compiler_flags.cmake#L153

palchak-google, 2025-03-26:
It's not missing. It was omitted. Reproduction requires a specific set of conditions in the source file that in turn depend on other files in the build. It's non-trivial.

Also, it's not necessary. The logs provided explicitly demonstrates the undesirable behavior as well the compiler flags change necessary to correct it.

palchak-google, 2025-03-27:
For the curious, the assembler warnings are caused by a bug in GCC as reported here:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=119487

But, the issue with Zephyr is *not* that the assembler is generating warnings. Rather, the Zephyr bug is that such warnings are not properly promoted to errors when `CONFIG_COMPILER_WARNINGS_AS_ERRORS` is selected.

nashif, 2025-03-30:
it would be really helpful if you could provide a way for this to be reproduced in the tree.

tejlmand, 2025-03-31:
Fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/87915