zephyrproject-rtos/zephyr #87748: doc: Documentation issue in 'connectivity/networking/api/sockets', broken links

clamattia, 2025-03-27:
**Describe the bug**
Broken links. The text references many functions that have broken links. They should probably link to the respective doxygen documentation.

**Environment**

* Page: `connectivity/networking/api/sockets`
* Version: main
* SHA-1: a61287e8766fe14d0055fba796c56948c483eaaf

clamattia, 2025-03-27:
> Do you mean that https://docs.zephyrproject.org/latest/connectivity/networking/api/sockets.html has broken links or some PR documentation has? I tried to go through various link in the public page but did not see broken pages. Could you add more information here?

Sorry, I have not put enough information in the description.
My expectation is/was, that things like:

[image]
would link to the doxygen documentation.

I remember it was like this at some point and created this issue to ask, if the links could be re-established?

If this is intended let me know and I will close the issue.

jukkar, 2025-03-27:
I do not remember that those were links before, and git blame tells that those lines were added around 2019 and they were not links. Those are normal BSD socket calls and we do not provide documentation for them atm, see https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/posix/sys/socket.h
I do not think we should provide documentation for them either, it is easy to search for example Linux manual pages from network. One such site is https://www.man7.org/linux/man-pages/
We have documentation and links for Zephyr variants like `zsock_socket` etc and in the docs we provide links that point to Posix documentation for actual function documentation.

clamattia, 2025-03-28:
Ok